tttapa/Control-Surface #221: Display Interfaces/Getting display elements into variables

felipegaspari, 2020-06-17:
Pieter, 

I've been trying to figure out how the display interfaces work. I find it very complex, and I understand that it is too much to explain here, . I'd be happy to understand how to use at least one display element, like TimeDisplay. How would insert that into a variable or something I can print using my own methods? 
 I'm using an ILI9341 and I'd like to create a display interface for it , but I'm very far from that at the moment. Any ideas would be greatly appreciated.
Thank you so much for this great library!

Felipe

tttapa, 2020-06-18:
The ILI9341 is non-buffered, and has a lot of pixels. While implementing the display interface for it should be trivial, and almost equivalent to the SSD1306 interface if you're using Adafruit's ILI library, it probably won't perform very well, mainly because Control Surface redraws the entire display each time.  
I have a better approach in mind, but I don't know if it's possible to implement it without too many compromises.

For now, I think it might be easier to drive the displays yourself directly.

You can still use the `TimeDisplay` class to handle MIDI input and printing, you just write the code to update the displays yourself.

For example (incomplete, I don't have an ILI9341 to test it):

```cpp
USBMIDI_Interface midi;
MCU::TimeDisplay timedisplay = {};

Adafruit_ILI9341 tft = {...};

void setup() {
  Control_Surface.begin();
  // Initialize display.
}

void loop() {
  Control_Surface.loop();

  if (/* you want to redraw the display now*/) {
    // Clear the region of the display where the TimeDisplay is to be drawn,
    // set the display's cursor, font size, font color, etc.
    timedisplay.printTo(tft);
    // handle more display stuff ...
  }
}
```

See the [`TimeDisplay`](https://tttapa.github.io/Control-Surface-doc/Doxygen/df/d69/classMCU_1_1TimeDisplay.html) documentation for more info about the `printTo` method and the other available methods.  
If you're unhappy with the way `printTo` prints the text, you can access the text data directly using the `getCharacterAt` and `getDecimalPointAt` functions.

For now, there is no way to check whether the text has changed, so you don't really know when you have to redraw the display. I'm reworking the MIDI input system to include a `dirty` flag to resolve this, but it's not ready for release yet.

jlmarest, 2020-08-27:
Hi Pieter, 
Congratulations for this library.

I'm trying to display midi outputs values. 

I'm thinking in a project about a control surface with as much faders, potentiometers and rotary encoders as arduino could manage in order to control a Waldorf Blofeld synthetizer. It's a digital synth with a lot of parameters but only six knobs. I would like to control the synth completely with the controller surface with out the need of looking at the synth display.

I would like to show all the values in a screen.

I think there would be a problem if, into every loop, the program rewrite all the params even if they haven't changed. So I'm studying the posibility to display the value at the same moment that the value is sended via Midi (USB o serial). 

What do you think about it?

Thanks, in advance.

Regards.

tttapa, 2020-09-05:
I'm currently working on making the MIDI input system more extensible, and this includes adding `getDirty()` and  `cleanDirty()` methods to most of the MIDI input elements that can be used for exactly this purpose.  

Development for this feature happens on the `new-input` branch. I haven't had time to publish the documentation yet, I'll probably do that later today or tomorrow.