tttapa/Control-Surface #223: sending absolute cc as 14 bit

Kmanggaard, 2020-06-24:
Hello,

I pieced this bit of code together for a high resolution incremental encoder, 600 ppr.
When I change the parameter to be controlled in my daw, I get a jump to the stored value. Is it possible to update the stored value with feedback from daw?

```
#include <Encoder.h>
#include <Control_Surface.h>

USBMIDI_Interface midi;

class CC14AbsoluteEncoder : public MIDIAbsoluteEncoder<ContinuousCCSender14<10>> {
  public:

    CC14AbsoluteEncoder(const EncoderPinList &pins, const MIDIAddress &address,
                        int16_t multiplier = 1, uint8_t pulsesPerStep = 4)
      : MIDIAbsoluteEncoder(pins, address, multiplier, pulsesPerStep, {}) {}
};

CC14AbsoluteEncoder enc = {
  {2, 3},       // pins
  56, // MIDI address (CC number + optional channel)
  1,
  1,
};

void setup() {
  RelativeCCSender::setMode(relativeCCmode::MACKIE_CONTROL_RELATIVE);
  Control_Surface.begin(); // Initialize Control Surface
}

void loop() {
  Control_Surface.loop();

}
```

tttapa, 2020-06-24:
That's only possible if the DAW sends the value when it changes. 

What DAW are you using? Do you have a dump of the MIDI messages sent by the DAW?

An easier solution would be to use relative CC messages if your DAW supports it.

Kmanggaard, 2020-07-24:
Just to wrap this up:

My daw (ableton) does indeed send feedback values, but not 14 bit. For those interested Live only supports 14 bit midi when directly mapping to a parameter (cmd+m / ctrl+m) NOT through a control surface script. this is true for both input and output. Took me a while to figure out.

To take advantage of a high resolution encoder with a Live remote script, use normal relative CC and change "parameter sensitivity" in your script. Look at the "SL MKIII" script for reference. Feedback is still 7 bit.

closed 2020-07-24 by Kmanggaard