tttapa/Control-Surface #229: Midi via serial

alomdj00, 2020-06-30:
Hola, estoy tratando de usar la libreria en un clon mega, por lo que no puedo usar hiduino si no el serial. Bien, resulta que no es posible detectar el envio de ninguna señal, no se como se debe de usar la interfaz, si la serie o la hairless, podrias ayudarme con un ejemplo basico

tttapa, 2020-06-30:
I don't really understand what you're asking, I hardly speak any Spanish, and Google Translate doesn't make much sense out of it either.

Anyway, if you want to use the [`HairlessMIDI_Interface`](https://tttapa.github.io/Control-Surface-doc/Doxygen/dc/dff/classHairlessMIDI__Interface.html), just replace the `USBMIDI_Interface` at the top of your sketch by this interface. For example:
```cpp
#include <Control_Surface.h> // Include the Control Surface library
 
// Instantiate the Hairless MIDI interface (uses default baud rate of 115200)
HairlessMIDI_Interface midi;
 
// Instantiate a CCPotentiometer object
CCPotentiometer potentiometer = {A0, 7};
 
void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}
 
void loop() {
  Control_Surface.loop(); // Update the Control Surface
}
```

The `HairlessMIDI_Interface` uses the serial port that's connected to the USB interface (i.e. `Serial`). If you want to use a different serial port, you can use [`SerialMIDI_Interface<T>`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d9/db8/classSerialMIDI__Interface.html), for example, to use `Serial1`:

```cpp
SerialMIDI_Interface<decltype(Serial1)> midi = {Serial1, MIDI_BAUD};
```

alomdj00, 2020-06-30:
He subido tu ejemplo exacto y no funciona, abro el hairless y se cierra o no envia  ni recibe datos.
Estoy usando el loopmidi, el convertidor serie es el cp2101, ya instale los drivers.

tttapa, 2020-06-30:
That sounds like more of a problem with Hairless than with the Control Surface library. You'll probably get better help on an online forum, https://forum.arduino.cc has a Spanish section, for example. And you'll have to provide much more information. Simply stating "it closes" or "it doesn't send or receive anything" is way too vague to be able to help you.