tttapa/Control-Surface #231: DualAddresses / Actions on a single button

zeysh, 2020-07-06:
Hi Pieter,

I hope your doing well.

I need your expertise gain...

In the documentation, I found the [DualAddresses](https://tttapa.github.io/Control-Surface-doc/Doxygen/dc/d06/classBankable_1_1DualAddresses.html) class and I wonder if there is a way to do it without a `Bank`.
Maybe a workaround can be to use only 1 `Bank`

Something like that:

```cpp
// Use button on PIN 5 to move on the LEFT_TRACK and SELECT it
// Use button on PIN 6 to move on the RIGHT_TRACK and SELECT it
// Use button on PIN 7 to enable/disable MUTE on the SELECTED track
// LED_BUILT_IN is on when the SELECTED track is MUTED
// Board: Teensy 4.0

#include <Control_Surface.h> // Include the Control Surface library

USBMIDI_Interface midi;

// Instantiate the buttons for switch between tracks and select it
DualAddresses::NoteButton track_control[] = {
  {
    5,
    {
      MCU::CHANNEL_LEFT,
      MCU::SELECT_1,
    },
  }
  {
    6,
    {
      MCU::CHANNEL_RIGHT,
      MCU::SELECT_1,
    },
  }
};

// trigger the LED if MUTE is enable on the track
NoteValueLED mute_led[] = {
  {LED_BUILTIN, MCU::MUTE_1},
};

NoteButton mute = {
  7, MCU::MUTE_1,
};

// --------------------------------- Setup ---------------------------------- //

void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}

// ---------------------------------- Loop ---------------------------------- //

void loop() {
  Control_Surface.loop(); // Refresh all elements
}
```
Many thanks

tttapa, 2020-07-06:
`DualAddresses` does not send on two addresses at the same time. It's a bankable address type for controls that require two distinct addresses (I think it's only used for `Bankable::CCIncrementDecrementButtons`, where the first address increments or decrements the position, and the second address resets the position).

If you want a single button to send messages for two different addresses, you can instantiate two NoteButtons on the same pin. A cleaner solution might be to implement your own custom MIDI output element: https://tttapa.github.io/Control-Surface-doc/Doxygen/d4/d3b/Custom-MIDI-Output-Element_8ino-example.html

zeysh, 2020-07-06:
Your first solution worked as expected: "instantiate two NoteButtons on the same pin"

```cc
// Use button on PIN 5 to move on the LEFT_TRACK and SELECT it
// Use button on PIN 6 to move on the RIGHT_TRACK and SELECT it
// Use button on PIN 7 to enable/disable MUTE on the SELECTED track
// LED_BUILT_IN is on when the SELECTED track is MUTED
// Board: Teensy 4.0

#include <Control_Surface.h> // Include the Control Surface library

USBMIDI_Interface midi;

// Instantiate the buttons for switch between tracks and select it
NoteButton track_left[] = {
  {5, MCU::CHANNEL_LEFT},
  {5, MCU::SELECT_1},
};

NoteButtonLatched track_right[] = {
  {6, MCU::CHANNEL_RIGHT},
  {6, MCU::SELECT_1},
};

// trigger the LED if MUTE is enable on the track
NoteValueLED mute_led[] = {
  {LED_BUILTIN, MCU::MUTE_1},
};

NoteButton mute = {
  7, MCU::MUTE_1,
};

// --------------------------------- Setup ---------------------------------- //

void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}

// ---------------------------------- Loop ---------------------------------- //

void loop() {
  Control_Surface.loop(); // Refresh all elements
}
```

I'll give a try with the `custom` MIDI element, but that will takes me more time.
Many thanks!

closed 2020-07-06 by zeysh

zeysh, 2020-07-06:
Samething with a custom MIDI output:

:+1:  @tttapa Thanks a lot!

```cc

#include <Control_Surface.h>
 
BEGIN_CS_NAMESPACE
 
/*
 * @brief   A class for momentary buttons and switches that send MIDI events.
 *
 * The button is debounced, and the internal pull-up resistor is enabled.
 *
 * @see     NoteButton
 * @see     Button
 */
class DualActionButton : public MIDIOutputElement {
 public:
  /*
   * @brief   Create a new MyNoteButton object on the given pin, with the 
   *          given address and velocity.
   * 
   * @param   pin
   *          The digital input pin to read from.  
   *          The internal pull-up resistor will be enabled.
   * @param   address1
   *          The MIDI address1 to send to.
   * @param   address2
   *          The MIDI address2 to send to.
   */
  DualActionButton(pin_t pin, const MIDIAddress &address1, const MIDIAddress &address2)
    : button{pin}, address1{address1}, address2{address2} {}
 
 public:
  // Initialize: enable the pull-up resistor for the button
  // This method is called once by `Control_Surface.begin()`.
  void begin() final override { button.begin(); }
 
  // Update: read the button and send MIDI messages when appropriate.
  // This method is called continuously by `Control_Surface.loop()`.
  void update() final override {
    AH::Button::State state = button.update();           // Read the button
    if (state == AH::Button::Falling) {                  // if pressed
      Control_Surface.sendNoteOn(address1, 0x7F);        // → note on
      Control_Surface.sendNoteOn(address2, 0x7F);        //
    } else if (state == AH::Button::Rising) {            // if released
      Control_Surface.sendNoteOff(address1, 0x7F);       //
      Control_Surface.sendNoteOff(address2, 0x7F);       // → note off
    }
  }
 
 private:
  AH::Button button;
  const MIDIAddress address1;
  const MIDIAddress address2;
};
 
END_CS_NAMESPACE
 
// :::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::: //
 
// Instantiate a MIDI over USB interface.
USBMIDI_Interface midi;
 
using namespace MIDI_Notes;
 
// Instantiate a MyNoteButton object
DualActionButton left = {
  5,                     // Push button on pin 5
  {MCU::CHANNEL_LEFT},   // Switch to left channel
  {MCU::SELECT_1},       // Select the current channel
};

DualActionButton right = {
  6,                     // Push button on pin 5
  {MCU::CHANNEL_RIGHT},  // Switch to right channel
  {MCU::SELECT_1},       // Select the current channel
};
 
void setup() {
  Control_Surface.begin(); // Initialize Control Surface (calls button.begin())
}
 
void loop() {
  Control_Surface.loop(); // Update the Control Surface (calls button.update())
}
```