tttapa/Control-Surface #232: Master Host for Slave USB Controllers

DylanBrianBenton, 2020-07-06:
Hey, 

I was wondering if theres a way to "daisy chain" other teensy's to one that is already connected to a PC?

Just thinking in the future I might like to expand with further surfaces and I was wondering if it was possible? Using T4 & 4.1's

**Image**
[https://i.imgur.com/RPnICJn.png](url)

tttapa, 2020-07-06:
Why USB? It sounds like complete overkill for a simple single-purpose link between Teensies. I'm sure it could be done, but I don't see the value in it.

My approach would be to set up a UART link instead. You can use the [MIDI routing example](https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/dc4/MIDI_Pipes-Routing_8ino-example.html) to merge the MIDI data from multiple slaves.

DylanBrianBenton, 2020-07-06:
Thanks Mate,

Hadn't read into UART in all honesty. Will do reading now :)

DylanBrianBenton, 2020-07-07:
> Why USB? It sounds like complete overkill for a simple single-purpose link between Teensies. I'm sure it could be done, but I don't see the value in it.
>
> My approach would be to set up a UART link [...]

Hey Mate, 

Thats for the direction. Just to clarify.

With this M/S UART setup the slave would be used to detect any presses, turns or inputs on that MCU and then relays it back to the Master which would process the USB Midi?

Sorry I know this isnt exactly library specific.

tttapa, 2020-07-07:
> With this M/S UART setup the slave would be used to detect any presses, turns or inputs on that MCU and then relays it back to the Master which would process the USB Midi?

Yes.

Both microcontrollers operate independently, the "slave" registers inputs from switches etc, and sends the appropriate MIDI messages over the UART (there are many possible transports for MIDI). 
The "master" does the same thing for its switches and other inputs, and sends the appropriate MIDI messages over USB. Additionally, the "master" has a UART MIDI input to receive the MIDI messages from the "slave". It merges the messages it receives on that UART input with its own messages, and sends everything over USB.

For MIDI input (from your computer to the microcontrollers), the "master" simply splits the MIDI stream, sending the messages to both its internal Control Surface input logic, and also sending it over the UART to the "slave".

DylanBrianBenton, 2020-07-07:
Awesome Thanks mate.

Will look more!

radiservi, 2020-09-27:
hello i'm new to the coding world i don't really understand how to make the connection by uart do you have an example?

radiservi, 2020-09-27:
if i want to do uart with rs485 it's possible?

pacolop, 2022-10-24:
Hi everyone,
as I understood, this would work with two atmega32u4 connected by I2C???

Trickster-git, 2022-10-24:
I use SERIAL from this example
https://tttapa.github.io/Control-Surface-doc/Doxygen/dd/ddd/Serial-Interface_8ino-example.html
and pipes
https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/dc4/MIDI_Pipes-Routing_8ino-example.html

master avr <-> usb
master avr <-> ser

slv avr <-> ser

pacolop, 2022-10-28:
thanks for your support.  I have to put the code to start the i2c communication both in the master and in the slave????

Trickster-git, 2022-10-31:
I can`t help with i2c, but if you want i can share my code 2 mega slave and pro micro master by serial.

pacolop, 2022-11-02:
@Trickster-git  Thank you very much, it would be helpful if apart from the code you could help me with the connections because I don't know what you mean by connecting by serial.
If you can pass me both things and it solves the problem, I will change my pcb

Trickster-git, 2022-11-02:
[addPush.zip](https://github.com/tttapa/Control-Surface/files/9924586/addPush.zip)
there is 3 files 
1 is for arduino mega used for 16 encoders and 32 buttons by mux (buttons stop work, i don`t know how)
2 is for arduino mega used for 16 encoders
3 is for pro micro use for usb midi and one pot
all boards connect by serial in loop. 1 tr->2rx 2tx->3rx 3tx->1rx
fill free with your qestions.

pacolop, 2022-11-03:
THANKS A LOT @Trickster-git 
I FINISHED SEEING THE FILE YOU SENT ME, THERE IS ONLY THE MEGA WITH 16 ENC. YOU CAN SEND ME THE OTHER TWO TO SEE IF THE COMMUNICATION CODE IS THE SAME.

Trickster-git, 2022-11-03:
[addPush_1nd.zip](https://github.com/tttapa/Control-Surface/files/9927870/addPush_1nd.zip)
[addPush_2nd.zip](https://github.com/tttapa/Control-Surface/files/9927871/addPush_2nd.zip)
[addPush_3nd.zip](https://github.com/tttapa/Control-Surface/files/9927872/addPush_3nd.zip)