tttapa/Control-Surface #233: Falla potenciometros esp32

alomdj00, 2020-07-06:
Hola Peter, estoy usando el ejemplo de potentiometer BLE en mi esp32 devkitv1 y no he logrado que lea los potenciometros en ningun pin adc, he intentado programar en todas las combinaciones desde uno en uno asta todos a la vez y no lee ningunl, el esp32 esta bien ya que el ejemplo de analogreadserial funciona perfecto con todos losbpines, parece ser problema de la libreria solo con los potenciometros, ya que botones y leds funcionan perfecto. Agradeceria enormemente que le hecharas una revision a este problema con las lecturas analogicas del esp32

tttapa, 2020-07-06:
Please see https://github.com/tttapa/Control-Surface/issues/149.

alomdj00, 2020-07-09:
Ya revise, y aun asi   continua la falla, no lee ningun pin, ya probe uno a uno  todos los adc1, y no registra lecturas,

tttapa, 2020-07-09:
I am unable to reproduce this issue. The following code works fine for me using a SparkFun ESP32 Thing with ESP32 Arduino Core version 1.0.4 and a potentiometer connected to GPIO36. Tested on Ubuntu 20.04, the control change messages arrive correctly over Bluetooth.

```cpp
#include <Control_Surface.h> // Include the Control Surface library
#include <MIDI_Interfaces/BluetoothMIDI_Interface.hpp>

// Instantiate a MIDI over USB interface.
BluetoothMIDI_Interface midi;

// Instantiate a CCPotentiometer object
CCPotentiometer potentiometer = {
  36,                                   // Analog pin connected to potentiometer
  {MIDI_CC::Channel_Volume, CHANNEL_1}, // Channel volume of channel 1
};

void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface
}
```

alomdj00, 2020-07-22:
Hola Peter, luego de intentar miles de cosas el problema lo he resuelto de la siguiente manera:
He seleccionado la placa generic esp32 en lugar   de la doit devkit.
He puesto un condensador de 0.1uf entre gnd y la entrada analogica  para evitar que se vuelva loco el esp32

closed 2020-07-22 by alomdj00