tttapa/Control-Surface #234: Push button with toggle

Willame156, 2020-07-06:
I would like to request an example using a push button with toggle function in CC commands.
Example: Clicking the first time sends CC 20, channel 1, velocity 127 and when clicking the second time sends CC 20, channel 1, velocity 0.

I thank you for your attention

tttapa, 2020-07-06:
You can use the [CCButton.ino](https://tttapa.github.io/Control-Surface-doc/Doxygen/d9/d2e/CCButton_8ino-example.html) example, replacing the `CCButton` class by [`CCButtonLatched`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d5/da6/classCCButtonLatched.html#details). Everything else remains the same.

Willame156, 2020-07-06:
Muito obrigado, você me ajudou bastante.
Obrigado mesmo.

luca-art, 2020-07-07:
By the way, I am not be able to work "CCButtonLatched" within a bank:

```
Bankable::CCButtonLatched b1 = {
  {bank, BankType::CHANGE_ADDRESS},     // bank configuration
  2,                                   // digital pin
  {102, CHANNEL_11},                  // address
};
Bankable::CCButton b2 = {
  {bank, BankType::CHANGE_ADDRESS},     // bank configuration
  3,                                   // digital pin
  {103, CHANNEL_11},                  // address

...
```
With "b2" button, no problem, but with "b1" latched version, I have this error:

exit status 1
invalid use of template-name 'CS::Bankable::CCButtonLatched' without an argument list

I think it is a number of banks related issue but I have tried adding a line of code just before BankType with differents syntax, but no luck.
I would appreciated your help.
(Sorry if I had to open a new post)

tttapa, 2020-07-07:
`Bankable::CCButtonLatched` is a template class. Please see the documentation: <https://tttapa.github.io/Control-Surface-doc/Doxygen/d9/d1d/classBankable_1_1CCButtonLatched.html>

```cpp
Bankable::CCButtonLatched<4> b1 = {
  {bank, BankType::CHANGE_ADDRESS},     // bank configuration
  2,                                   // digital pin
  {102, CHANNEL_11},                  // address
};
```
Replace the 4 with the number of banks you're using.

luca-art, 2020-07-07:
Thank you very much, it works perfectly