tttapa/Control-Surface #236: Invert Potentiometer Input

GinjahWolf92, 2020-07-09:
I was a bit stupid and mounted my faders upside down and I am wondering if there is a way to implement a portion of code that could invert the slider input i.e 0 is 127 and 127 is 0

One line of code as a 'flag' that would invert values.

**Describe alternatives you've considered**
I hunted through the documentation and I was unable to find the solution I was looking for.

**Additional context**
I am unsure as to how this could be implemented but I would be very interested as to how a solution could be created.

```cpp
#include <Control_Surface.h>

HairlessMIDI_Interface midi;

CCPotentiometer potentiometer1 = { A7, {MIDI_CC::Channel_Volume, CHANNEL_1} };
CCPotentiometer potentiometer2 = { A2, {MIDI_CC::Channel_Volume, CHANNEL_2} };
CCPotentiometer potentiometer3 = { A1, {MIDI_CC::Channel_Volume, CHANNEL_3} };
CCPotentiometer potentiometer4 = { A0, {MIDI_CC::Channel_Volume, CHANNEL_4} };
CCPotentiometer potentiometer5 = { A10, {10, CHANNEL_5} };

void setup() {
    Control_Surface.begin();
}

void loop() {
    Control_Surface.loop();
}
```

GinjahWolf92, 2020-07-09:
Whoops, just noticed that there is a solution, apologies!

closed 2020-07-09 by GinjahWolf92

tttapa, 2020-07-09:
For future reference, you can use the [`CCPotentiometer::invert`](https://tttapa.github.io/Control-Surface-doc/Doxygen/db/d32/classCCPotentiometer.html#a7fa1616cc61c19a5efcc863c950f7f30) method.

To invert all potentiometers in the array:

```cpp
CCPotentiometer potentiometers[] = {
  { A7, {MIDI_CC::Channel_Volume, CHANNEL_1} },
  { A2, {MIDI_CC::Channel_Volume, CHANNEL_2} },
  { A1, {MIDI_CC::Channel_Volume, CHANNEL_3} },
  { A0, {MIDI_CC::Channel_Volume, CHANNEL_4} },
  { A10, {10, CHANNEL_5} },
};

void setup() {
  for (auto &potentiometer : potentiometers)
    potentiometer.invert();
  // ...
}
```

To invert only the first potentiometer:
```cpp
void setup() {
  potentiometers[0].invert();
  // ...
}
```

GinjahWolf92, 2020-07-09:
Thanks! I ended up doing this but I suppose both ways ultimately work!
```cpp
#include <Control_Surface.h>

HairlessMIDI_Interface midi;

CCPotentiometer potentiometer1 = { A7, {MIDI_CC::Channel_Volume, CHANNEL_1} };
CCPotentiometer potentiometer2 = { A2, {MIDI_CC::Channel_Volume, CHANNEL_2} };
CCPotentiometer potentiometer3 = { A1, {MIDI_CC::Channel_Volume, CHANNEL_3} };
CCPotentiometer potentiometer4 = { A0, {MIDI_CC::Channel_Volume, CHANNEL_4} };
CCPotentiometer potentiometer5 = { A10, {10, CHANNEL_5} };

void setup() {
    Control_Surface.begin();
    potentiometer1.invert();
    potentiometer2.invert();
    potentiometer3.invert();
    potentiometer4.invert();
    potentiometer5.invert();
}

void loop() {
    Control_Surface.loop();
}
```

tttapa, 2020-07-09:
Yes, they should be equivalent. For large numbers of potentiometers, the for loop is easier to read and maintain, but for just 5 potentiometers it doesn't really matter.