tttapa/Control-Surface #239: Roland V-1HD video switcher

Sylvan72, 2020-07-11:
450/5000
Good night.

I need to light 4 LEDs to indicate which camera input is connected to a Roland V-1HD video switcher. When a video input is selected, it transmits a bank number along with a program change:

CC Bank MSB => B0h 00h 01h
CC Bank LSB => B0h 20h 00h

PC => C0h 00h - Program (00h-03h)

I was unable to create something that read these three values. What function do I use to read the MSB, LSB and PC codes from the Rx serial input?

Thanks

tttapa, 2020-07-12:
I've not yet added 14-bit CC input to the library. I think the easiest approach is to just handle it yourself in the MIDI input callbacks, see https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/d32/MIDI-Input-Callback_8ino-example.html

```cpp
bool channelMessageCallback(ChannelMessage cm) {
  if (cm.header == (CONTROL_CHANGE | CHANNEL_1.getRaw())) {
    if (cm.data1 == 0x00) {
      // MSB is in cm.data2
      return true;
    } else if (cm.data1 == 0x20) {
      // LSB is in cm.data2
      return true;
    }
  } else if (cm.header == (PROGRAM_CHANGE | CHANNEL_1.getRaw())) {
    // Program is in cm.data1
    return true;
  }
  return false; // Return true to indicate that handling is done,
                // and Control_Surface shouldn't handle it anymore.
                // If you want Control_Surface to handle it as well,
                // return false.
}
```

For future reference: there are cleaner ways to do this in the new release of the library I'm working on, but it's not finished yet.

Sylvan72, 2020-07-12:
Thank you very much.