tttapa/Control-Surface #240: Trackpad laptop send midi values

lmeucchi, 2020-07-16:
Hello,
I have a trackpad from an old laptop, it has a Ps2 connection and uses 4 pins: 5v, gnd, data, clock
Can the pad be used in control surface to send pitch values ​​on the "x" axis and on the "y" axis modulation?
is it possible to send midi notes on the "x" axis and velocity on the "y" axis?

Connect the pad using the ps2 library, I have two variables "mx" and "my" but I don't know how to include it in control surface

Thank you

```cpp
#include <ps2.h>

/*
 * an arduino sketch to interface with a ps/2 mouse.
 * Also uses serial protocol to talk back to the host
 * and report what it finds.
 */

/*
 * Pin 5 is the mouse data pin, pin 6 is the clock pin
 * Feel free to use whatever pins are convenient.
 */
PS2 mouse(6, 5);

/*
 * initialize the mouse. Reset it, and place it into remote
 * mode, so we can get the encoder data on demand.
 */
void mouse_init()
{
  mouse.write(0xff);  // reset
  mouse.read();  // ack byte
  mouse.read();  // blank */
  mouse.read();  // blank */
  mouse.write(0xf0);  // remote mode
  mouse.read();  // ack
  delayMicroseconds(100);
}

void setup()
{
  Serial.begin(9600);
  mouse_init();
}

/*
 * get a reading from the mouse and report it back to the
 * host via the serial line.
 */
void loop()
{
  char mstat;
  char mx;
  char my;

  /* get a reading from the mouse */
  mouse.write(0xeb);  // give me data!
  mouse.read();      // ignore ack
  mstat = mouse.read();
  mx = mouse.read();
  my = mouse.read();

  /* send the data back up */
  Serial.print(mstat, BIN);
  Serial.print("\tX=");
  Serial.print(mx, DEC);
  Serial.print("\tY=");
  Serial.print(my, DEC);
  Serial.println();
//  delay(20);  /* twiddle */
}
```

tttapa, 2020-07-16:
It depends what you want to do with the data. There's obviously no PS2 support baked into the library, so you'll have to handle sending MIDI yourself.

- https://tttapa.github.io/Control-Surface-doc/Doxygen/d4/d3b/Custom-MIDI-Output-Element_8ino-example.html
- https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/d9b/MIDI-Output_8ino-example.html
- https://tttapa.github.io/Control-Surface-doc/Doxygen/d6/d72/classMIDI__Sender.html

lmeucchi, 2020-07-17:
Thanks, I will try to understand how your library works ... it is very difficult for me, and even more if I want to customize ... but I will read it 
Thank you

tttapa, 2020-07-17:
The first thing is to understand what kinds of MIDI messages exist, and what kinds of MIDI data your DAW accepts and understands.

Then you just have to call `Control_Surface.send***(data)` in your sketch, where `***` is the kind of MIDI message you want to send, and `data` is whatever data you want to send, probably the position of the touchpad, which may be scaled or summed.

The idiomatic way to initialize the touchpad and update it is to implement the abstract `MIDIOutputElement` class, as shown in the first example I linked to. It has a `begin` method where you have to initialize the touchpad, and an `update` method, where you have to read the data from the touchpad and send it over MIDI using `Control_Surface.send***`.

It might be a good idea to follow an introductory C++ tutorial first, so you understand the basics of classes, functions and objects.

lmeucchi, 2020-07-19:
Thanks, I am going to start learning more about C ++ so I can understand about classes and functions of your code. 
I think that with your library I can do many things. It can be seen that you spent a lot of time on this and now I know that there is the possibility of sending personalized messages ... it's amazing

thank you

closed 2021-12-10 by lmeucchi