tttapa/Control-Surface #241: CCButtonLatched LED without ShiftRegister

ravedya, 2020-07-20:
Hello,

I recently made a foot switch with the LED to control guitar effect plugin. I did it all by seeing the documentation and it comes up using the shift register. I did buy the 74HC595 IC and it did work great. But I wonder if I could did that without using the shift register. My problem is I don't know how to get the state of the CCButtonLatched since it's in array and how to make it into the if conditional premise. I'm sorry if this is a stupid question but your explanation will help me a lot. 

Thank you!

```cpp
#include <Control_Surface.h>  // Include the library
 
USBMIDI_Interface midi;  // Instantiate a MIDI Interface to use

// Instantiate an analog multiplexer
CD74HC4067 mux = {
  8,       // Digital input pin
  {2, 3, 4, 5} // Address pins S0, S1, S2, S3
};

// Create an array of switch that send out MIDI Control Change messages 

  CCButtonLatched footswitchControl[] = {
  { mux.pin(0), { 20, CHANNEL_1 } },
  { mux.pin(1), { 21, CHANNEL_1 } },
  { mux.pin(2), { 22, CHANNEL_1 } },
  { mux.pin(3), { 23, CHANNEL_1 } },
};

const pin_t ledPedal1 = 10;
const pin_t ledPedal2 = 16;
const pin_t ledPedal3 = 14;
const pin_t ledPedal4 = 15;

void setup() {
  Control_Surface.begin();  // Initialize the Control Surface
  pinMode(ledPedal1, OUTPUT);
  pinMode(ledPedal2, OUTPUT);
  pinMode(ledPedal3, OUTPUT);
  pinMode(ledPedal4, OUTPUT);
}

void loop() {
  Control_Surface.loop();  // Update the Control Surface

  if (footswitchControl[1].update() == Button::Pressed){ //this should be the code to fetch the state of the button
      digitalWrite(ledPedal1, HIGH);}
      else {digitalWrite(ledPedal1, LOW);}
  if (footswitchControl[2].update() == Button::Pressed){
      digitalWrite(ledPedal2, HIGH);}
      else {digitalWrite(ledPedal2, LOW);}
  if (footswitchControl[3].update() == Button::Pressed){
      digitalWrite(ledPedal3, HIGH);}
      else {digitalWrite(ledPedal3, LOW);}
  if (footswitchControl[4].update() == Button::Pressed){
      digitalWrite(ledPedal4, HIGH);}
      else {digitalWrite(ledPedal4, LOW);}
}
```

tttapa, 2020-07-20:
1. To most parts of the Control Surface library, there's no difference between an output pin of the Arduino and an output pin of a shift register. They can be used interchangeably, if your original code had `sreg.pin(0)`, you could replace it with `13` for "digital pin 13" of the Arduino, for example.

2. In C++, array indices start at 0, not 1. `footswitchControl[4]` doesn't exist.

3. `CCButtonLatched::update` doesn't return a button state, it returns nothing. In fact, you shouldn't call it from user code at all, all MIDI elements are updated automatically when calling `Control_Surface.loop()`.

4. To get the state of a `CCButtonLatched` object, you can either use `CCButtonLatched::getState()` or `CCButtonLatched::getButtonState()`. The former returns the latched state, which corresponds to the MIDI messages, the latter returns the state of the actual button. See the [`CCButtonLatched` documentation](https://tttapa.github.io/Control-Surface-doc/Doxygen/d5/da6/classCCButtonLatched.html).

ravedya, 2020-07-20:
So how should I do the loop code then? It seems like mine doesn't work and the error code is 
> no matching function for call to 'CS::CCButtonLatched::getState(CS::CCButtonLatched&)' 
Once again thank you.

`void loop() {
  Control_Surface.loop();  // Update the Control Surface

  if (CCButtonLatched::getState(footswitchControl[0])== HIGH){
      digitalWrite(ledPedal1, HIGH);}
      else {digitalWrite(ledPedal1, LOW);}
  if (CCButtonLatched::getState(footswitchControl[1]) == HIGH){
      digitalWrite(ledPedal2, HIGH);}
      else {digitalWrite(ledPedal2, LOW);}
  if (CCButtonLatched::getState(footswitchControl[2]) == HIGH){
      digitalWrite(ledPedal3, HIGH);}
      else {digitalWrite(ledPedal3, LOW);}
  if (CCButtonLatched::getState(footswitchControl[3]) == HIGH){
      digitalWrite(ledPedal4, HIGH);}
      else {digitalWrite(ledPedal4, LOW);}
}`

ravedya, 2020-07-20:
So how should I code the loop then? Mine doesn't work and give : no matching function for call to 'CS::CCButtonLatched::getState(CS::CCButtonLatched&)' in return. 
`void loop() {
  Control_Surface.loop();  // Update the Control Surface

  if (CCButtonLatched::getState(footswitchControl[0])== HIGH){
      digitalWrite(ledPedal1, HIGH);}
      else {digitalWrite(ledPedal1, LOW);}
  if (CCButtonLatched::getState(footswitchControl[1]) == HIGH){
      digitalWrite(ledPedal2, HIGH);}
      else {digitalWrite(ledPedal2, LOW);}
  if (CCButtonLatched::getState(footswitchControl[2]) == HIGH){
      digitalWrite(ledPedal3, HIGH);}
      else {digitalWrite(ledPedal3, LOW);}
  if (CCButtonLatched::getState(footswitchControl[3]) == HIGH){
      digitalWrite(ledPedal4, HIGH);}
      else {digitalWrite(ledPedal4, LOW);}
}`

ravedya, 2020-07-20:
So how should I code the loop then? Mine doesn't work and give : no matching function for call to CS::CCButtonLatched::getState(CS::CCButtonLatched&) in return.

`void loop() {
Control_Surface.loop(); // Update the Control Surface

if (CCButtonLatched::getState(footswitchControl[0])== HIGH){
digitalWrite(ledPedal1, HIGH);}
else {digitalWrite(ledPedal1, LOW);}
if (CCButtonLatched::getState(footswitchControl[1]) == HIGH){
digitalWrite(ledPedal2, HIGH);}
else {digitalWrite(ledPedal2, LOW);}
if (CCButtonLatched::getState(footswitchControl[2]) == HIGH){
digitalWrite(ledPedal3, HIGH);}
else {digitalWrite(ledPedal3, LOW);}
if (CCButtonLatched::getState(footswitchControl[3]) == HIGH){
digitalWrite(ledPedal4, HIGH);}
else {digitalWrite(ledPedal4, LOW);}
}`

tttapa, 2020-07-20:
`CCButtonLatched::getState()` is a member function. You call it using the member access operator (`.`) on an instance of the `CCButtonLatched` class: `footswitchControl[0].getState()`

`CCButtonLatched::getState()` returns a Boolean value, not `HIGH`, comparing it against `HIGH` will give unexpected results or even compilation errors in newer versions of the Arduino Core API.

You could try something like this:
```cpp
#include <Control_Surface.h>  // Include the library
 
USBMIDI_Interface midi;  // Instantiate a MIDI Interface to use

// Get the length of an array
template <class T, size_t N> constexpr size_t len(T (&)[N]) { return N; }

// Instantiate an analog multiplexer
CD74HC4067 mux = {
  8,           // Digital input pin
  {2, 3, 4, 5} // Address pins S0, S1, S2, S3
};

// Create an array of switches that send out MIDI Control Change messages 
CCButtonLatched footswitchControl[] = {
  { mux.pin(0), { 20, CHANNEL_1 } },
  { mux.pin(1), { 21, CHANNEL_1 } },
  { mux.pin(2), { 22, CHANNEL_1 } },
  { mux.pin(3), { 23, CHANNEL_1 } },
};

const pin_t ledPins[] = { 10, 16, 14, 15 };

static_assert(len(footswitchControl) == len(ledPins), 
              "Error: requires same number of switches as LEDs");

void setup() {
  Control_Surface.begin();  // Initialize the Control Surface
  for (auto pin : ledPins)
    pinMode(pin, OUTPUT);
}

void loop() {
  Control_Surface.loop();  // Update the Control Surface

  // Loop over all switches and LEDs
  for (size_t i = 0; i < len(ledPins); ++i)
    // If the button was pressed, this means the toggled state changed
    if (footswitchControl[i].getButtonState() == Button::Falling)
      // Update the LED state to reflect the toggled switch state
      digitalWrite(ledPins[i], footswitchControl[i].getState() ? HIGH : LOW);
}
```

ravedya, 2020-07-20:
Well man it's kinda more complicated than I thought but it's worth learning. Thanks for the help!

tttapa, 2020-07-20:
It might be a good idea to follow an introductory C++ tutorial first, it'll prove much more effective than copy-and-pasting code you don't understand.

closed 2020-07-20 by ravedya

tttapa, 2020-07-20:
Well, reading it back, that does sound a bit harsh, I genuinely believe that's the best approach here, and it's a mistake many people make. Looking at examples and modifying them to suit your needs is fun way to learn, and an essential part of the learning process. However, if you don't understand the underlying basics of the language, you'll eventually get stuck.  
I usually recommend only using code that you understand. So let's say you wanted to use the code I posted above: try it out, modify some parts, and then look up the documentation for parts you don't understand. Finally, convince yourself that you've understood every line, word, symbol ... before proceeding to the next part of the project.

It doesn't have to be that strict all the time of course, and you don't have to learn everything in one day.

To really understand the code above, you'll need some basic knowledge of classes and list initialization. You'll also need to know how to call member functions, and how if statements and for loops work. The for loop in the setup function is a range-based for loop, which might be new to you. The `auto` keyword let's the compiler figure out the type for you. You might be able to guess what `static_assert` does, but it could be a good idea to look up the documentation on cppreference.com anyway, to learn how to use the language reference and documentation.

The `len()` function template is quite advanced, so don't get discouraged if you don't understand it right away. It combines templates, C-style arrays, references and constexpr.