tttapa/Control-Surface #243: Getting my Footcontroller to work well with Helix Native.

ziplob123, 2020-07-21:
Hi there,
I'm trying to create a custom footpedal to control the 'Line 6 Helix Native' from within my Daw (Flstudio) using your Control surface. 

Before doing this I had zero coding knowlege at all and still don't TBH but thats now work in progress.
Anyway my problem is the Helix uses 'CC's' that can be mapped and changed fine but some are protected and cannot be changed here is where my problem lies.

An example of this is the helix has 8 Snapshots using  8 buttons to select which snapshot and also has 2 other buttons to move up and down through the snapshots in the bank.
A bank of 8 snapshots with 8 buttons - 1-8  is changed using 0x00-0x07 
2 buttons to move up and down the snapshot bank using 0x08 to Incriment and 0x09 to Decrement
Button1= Channel 1, CC69 and value of 8  (Decrement the snapshot) 
and 
Button2= Channel 1, CC69 and value of 9  (Increment the snapshot)

The way the CCButton currently works is
CCButton But10 = { 11, MIDI_CC::Hold_2 }; 
-sends Channel 1, CC69 and a on value of 127 and a off value of 0 (but I need both values to be 9)

To try and counter this in my code aT line 20 and 21 I'm using: 
CCButton But9 = { 10, MIDI_CC::Hold_2, (0x08, 0x08) };  -to force the Decrement
CCButton But10 = { 11, MIDI_CC::Hold_2, (0x09, 0x09) }; -to force the Increment

So the MIDI_CC::Hold_2 gives Channel 1 and CC69 which works fine, I used (0x07, 0x07) and (0x08, 0x08) to try and force both the 'on' and 'off' values to send 8  and  9 for both.
This worked for the on value but the off value still sends 0x00 (0) which resets the helix's snapshot increments back to snapshot1.

Is there a simple way to overide the 'On Off' value for CCButton?

Sorry if this answer is so simple I been trying to sort it out myself without bringing the issue here but I'm pulling my hair out haha.
Thanks making my project possible by simplify everything with the Control surface Libary

```cpp
// #include <Control_Surface.h>
 
HairlessMIDI_Interface stream;

CCPotentiometer Modwheel = { A0, MIDI_CC::Breath_Controller };
CCPotentiometer Volume = { A1, MIDI_CC::Channel_Volume };

CCButtonLatching But1 = { 8, MIDI_CC::General_Purpose_Controller_1 };
CCButtonLatching But2 = { 4, MIDI_CC::General_Purpose_Controller_2 };
CCButtonLatching But3 = { 5, MIDI_CC::General_Purpose_Controller_3 };
CCButtonLatching But4 = { 9, MIDI_CC::General_Purpose_Controller_4 };
CCButtonLatching But5 = { 6, MIDI_CC::General_Purpose_Controller_5 };
CCButtonLatching But6 = { 2, MIDI_CC::General_Purpose_Controller_6 };
CCButtonLatching But7 = { 3, MIDI_CC::General_Purpose_Controller_7 };
CCButtonLatching But8 = { 7, MIDI_CC::General_Purpose_Controller_8 };
CCButton But9 = { 10, MIDI_CC::Hold_2, (0x08, 0x08) };
CCButton But10 = { 11, MIDI_CC::Hold_2, (0x09, 0x09) };

void setup() 
{

 
  Control_Surface.begin(); // Initialize Control Surface
}
 
void loop() {
  Control_Surface.loop(); // Update the control surface
}
```

tttapa, 2020-07-21:
The correct syntax is:
```cpp
CCButton But9 = { 10, MIDI_CC::Hold_2, {0x08, 0x08} };
```
If you use parentheses, they just group `0x08, 0x08` together as a single argument. The compiler then evaluates the comma operator, so the it would evaluate to `{ 10, MIDI_CC::Hold_2, 0x08 }`, without setting the second argument of the DigitalCCSender.

By the way, there's no need to use the MIDI_CC constants, you can just use the CC number:
```cpp
CCButton But9 = { 10, 69, {0x08, 0x08} };
```

ziplob123, 2020-07-21:
Thanks for the help. Thats working just great now
Yeah this is my first time learning programming. So it is a big jump for me, but I was on the right track.

closed 2020-07-22 by ziplob123