tttapa/Control-Surface #245: multiplexer and matrix

m0ck0, 2020-07-25:
hi im trying to build a midi keyboard and add a bunch of pots, so far i found the example on how to set up pots through a multiplexer but i cant figure out how to add more than one multiplexer in parallel mode.

how do i set up a couple (2 or 3) multiplexers in parallel to make a matrix?

djoulz22, 2020-07-25:
Hi m0ck0,

If you use a CD74HC4067, you can put all of it in parallel. So, for example
:

For 3 multiplexer :

S0 => D5
S1 => D6
S2 => D7
S3 => D8

Mux1 COM => A0
Mux2 COM => A1
Mux3 COM => A2

Optionally the enable pin of each mux could be plugged, but optionally...

Next, declare Muxes :

// Array of mux
CD74HC4067 muxes[] = {
{
  A0,               // input pin
  {5, 6, 7, 8}, // Address pins S0, S1, S2, S3
  //3, // Optionally, specify the enable pin
},
{
  A1,               // input pin
  {5, 6, 7, 8}, // Address pins S0, S1, S2, S3
  //4, // Optionally, specify the enable pin
},
{
  A2,               // input pin
  {5, 6, 7, 8}, // Address pins S0, S1, S2, S3
  //9, // Optionally, specify the enable pin
}
};

Next, declare buttons :

// Array of button
CCButton buttons[] = {
  {
    muxes[0].pin(1),        // Analog pin connected to button : First mux,
pin 1
    , {0x03, CHANNEL_1}
  },
  {
    muxes[1].pin(2),        // Analog pin connected to button : Second mux,
pin 2
    , {0x09, CHANNEL_1}
  },
  {
    muxes[3].pin(3),        // Analog pin connected to button : Third mux,
pin 3
    , {0x10, CHANNEL_1}
  },
};

Next declare pots :

// Array of pots
CCPotentiometer potentiometers[] = {
    { muxes[0].pin(0), { 0x14, CHANNEL_1 } },
    { muxes[0].pin(1), { 0x15, CHANNEL_1 } },
    { muxes[0].pin(2), { 0x16, CHANNEL_1 } },
    { muxes[0].pin(3), { 0x17, CHANNEL_1 } },
    { muxes[0].pin(4), { 0x18, CHANNEL_1 } },
    { muxes[0].pin(5), { 0x19, CHANNEL_1 } },
    { muxes[0].pin(6), { 0x1A, CHANNEL_1 } },
    //{ muxes[0].pin(7), { 0x35, CHANNEL_1 } },
    { muxes[0].pin(8), { 0x01, CHANNEL_1 } },

    { muxes[1].pin(0), { 0x4A, CHANNEL_1 } },
    { muxes[1].pin(1), { 0x47, CHANNEL_1 } },
    { muxes[1].pin(2), { 0x48, CHANNEL_1 } },
    { muxes[1].pin(3), { 0x07, CHANNEL_1 } },
    { muxes[1].pin(4), { 0x49, CHANNEL_1 } },
    { muxes[1].pin(5), { 0x4B, CHANNEL_1 } },
    { muxes[1].pin(6), { 0x4F, CHANNEL_1 } },
    { muxes[1].pin(7), { 0x50, CHANNEL_1 } },
    { muxes[1].pin(8), { 0x51, CHANNEL_1 } },
    { muxes[1].pin(9), { 0x52, CHANNEL_1 } },
    { muxes[1].pin(10), { 0x34, CHANNEL_1 } },

    { muxes[2].pin(3), { 0x1F, CHANNEL_1 } },
    { muxes[2].pin(4), { 0x11, CHANNEL_1 } },
    { muxes[2].pin(9), { 0x12, CHANNEL_1 } },
    { muxes[2].pin(10), { 0x13, CHANNEL_1 } },
    { muxes[2].pin(11), { 0x10, CHANNEL_1 } },
    { muxes[2].pin(12), { 0x1C, CHANNEL_1 } },
    { muxes[2].pin(13), { 0x1D, CHANNEL_1 } },
    { muxes[2].pin(14), { 0x1E, CHANNEL_1 } },
    { muxes[2].pin(15), { 0x1B, CHANNEL_1 } },
};

So, good sketch! ;-)

djoulz22, 2020-07-25:
Tell us if it's okay!

m0ck0, 2020-07-27:
Yes! thanks so much!, i still have some things to figure out (like velocity and transposing) but that's a whole other battle. 

i'm not finished yet, but in case it helps someone i'm gonna live my code here. 

```#include <Arduino_Helpers.h> // Include the Arduino Helpers library
#include <AH/Hardware/ExtendedInputOutput/AnalogMultiplex.hpp>
#include <Control_Surface.h>

HairlessMIDI_Interface midi;

 CD74HC4051 mux [] = {
 {
   A0,
   {2, 3, 4},
},
 {
   A1,
   {2, 3, 4},
},
 {
   A2,
   {2, 3, 4},
}
};

const AddressMatrix<8, 5> addresses = {{
  { 1,  2,  3,  4,  5},
  { 6,  7,  8,  9, 10},
  {11, 12, 13, 14, 15},
  {16, 17, 18, 19, 20},
  {21, 22, 23, 24, 25},
  {26, 27, 28, 29, 30},
  {31, 32, 33, 34, 35},
  {36, 37, 38, 39, 40},
}};

NoteButtonMatrix<8, 5> buttonmatrix = {
  {mux[0].pin(0), mux[0].pin(1), mux[0].pin(2), mux[0].pin(3), mux[0].pin(4), mux[0].pin(5), mux[0].pin(6), mux[0].pin(7)}, // row pins
  {mux[1].pin(0), mux[1].pin(1), mux[1].pin(2), mux[1].pin(3), mux[1].pin(4)},    // column pins
  addresses,    // address matrix
  CHANNEL_1,    // channel and cable number
};

void setup() {
  Control_Surface.begin();
}

void loop() {
    Control_Surface.loop();
```

closed 2020-07-27 by m0ck0

djoulz22, 2020-07-27:
For transposing, i don't know if there IS a specific best way... I
suppose...

You Can use Banks instead, if there IS not solution.

For the velocity, buttons could be concidered as pots... L'aube

PS: sorry for m'y Bad english, i'm just a Franch... ;-)