tttapa/Control-Surface #246: Change channel of the Channel object

mhlsf, 2020-07-27:
Hi I would like to be able to change channel using a button but I don't know how to manipulate the channel class in order the change the channel ? How would I do this ?

Thanks for your work!

```cpp
#include <Encoder.h> // Include the Encoder library.
// This must be done before the Control Surface library.
#include <Control_Surface.h> // Include the Control Surface library
#include <Def/Channel.hpp>
 
// Instantiate a MIDI over USB interface.
//HairlessMIDI_Interface midi;
USBDebugMIDI_Interface midi;

const Channel channel = CHANNEL_1; 
// Instantiate a CCRotaryEncoder object
MIDIAddress address = {12, channel, 0};

CCAbsoluteEncoder enc = {
  {A1, A2},       // pins
   address, // MIDI address (CC number + optional channel)
  5,            // optional multiplier if the control isn't fast enough
};
 
void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}
 
void loop() {
  Control_Surface.loop(); // Update the Control Surface
}
```

tttapa, 2020-07-27:
Changing channels is done using banks: https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/d74/Bank_8ino-example.html

mhlsf, 2020-07-27:
How I see I thougth they were for changin program on a midi synthethizer ! I will look into it thank you !

mhlsf, 2020-07-27:
I see there is no Bankable available with the **CCAbsoluteEncoder**, and I already tried with the **CCRotaryEncoder** but it doesn't work as ableton take absolute value, (potentiometer go from 1 to 127 back and forth), is there a fix for that ? Is it possible to use Bankable on **CCAbsoluteEncoder** ?

Thanks

tttapa, 2020-07-27:
You're right, I haven't merged it into the master branch yet.

You should be able to configure Ableton to use relative messages, so I'd try that first. 

I don't have much time to merge CCAbsoluteEncoder into master right now, but you could use the `share-encoder` branch. You'll need my fork of the Encoder library.

<https://tttapa.github.io/Control-Surface-doc/share-encoder/Doxygen/d7/d2c/classBankable_1_1CCAbsoluteEncoder.html>
<https://github.com/tttapa/Encoder>

mhlsf, 2020-07-27:
Work as expected ! But you should know that rotary encoder multiplier doesn't work starting at 5 for the value, is it a known bug on this branch ?

For Information if I set the multiplier to 5 or plus, I get 0,127, 126 values ?

tttapa, 2020-07-27:
Thanks for letting me know! I'll have a look at it tomorrow. It might be an integer oveflow.

tttapa, 2020-07-28:
It turned out to be a rounding error. I pushed a fix to the shared-encoder branch.

I only tested the non-bankable version so far, but it's getting late here so that'll have to do for now.

mhlsf, 2020-08-06:
I tested with the last change you made ! Work perfectly ! Thank you ! Also, do you know if there is a way to display the "range" (CC controller value) of currently used potentiometer with a serie of led ? For example 7 leds controlled with a shift register ?

tttapa, 2020-08-10:
> Also, do you know if there is a way to display the "range" (CC controller value) of currently used potentiometer with a serie of led ? For example 7 leds controlled with a shift register ?

You could use something like this:
```cpp
#include <Control_Surface.h>

USBDebugMIDI_Interface midi;
 
// Instantiate a shift register with the SPI slave select pin as latch pin, most
// significant bit first, and a total of 8 outputs.
SPIShiftRegisterOut<8> sreg = {SS, MSBFIRST};

// A potentiometer that sends MIDI control change messages.
CCPotentiometer pot = { A0, 0x10 };

// A range of 7 LEDs on the first 7 pins (indices 0-6) of the shift register.
DotBarDisplayLEDs<7> leds = { sreg.pins().slice<0, 6>() };

void setup() {
  Control_Surface.begin();
  leds.begin();
}

void loop() {
  Control_Surface.loop();
  leds.display(pot.getValue() / 128.f);
}
```

mhlsf, 2020-08-25:
Hi ! Thanks for the example, but I don't understand how the wiring should work, I plug the RCLK (Register Clock / Latch) of my shift register on the 10 pin of the arduino, but I don't see anywhere in the code which pin is the data feeder for the shift register ? Did I miss something ?

Thank you !

tttapa, 2020-08-28:
It uses the SPI bus for data. Please see the [documentation](https://tttapa.github.io/Control-Surface-doc/Doxygen/d5/d09/classAH_1_1SPIShiftRegisterOut.html#a3732a80bf2f4ebf4d0ac32a9d393fb73) and [this example](https://tttapa.github.io/Control-Surface-doc/Doxygen/d9/d5f/1_8SPI-Blink_8ino-example.html).

closed 2020-12-01 by mhlsf