tttapa/Control-Surface #247: Problem Using Serial Monitor

mipabersatu, 2020-07-28:
So I'm trying to monitor the value of potentiometer in the serial monitor. But it happens that the monitor won't appear when I clicked it. 
Do you know what possibly can affect it?

tttapa, 2020-07-28:
This sounds like an issue with your installation of the Arduino IDE, not with the Control Surface library.

mipabersatu, 2020-07-28:
But I tried a regular code just to show serials on monitor and it did work.

tttapa, 2020-07-28:
Does the monitor not appear, or does it not print to the monitor?

mipabersatu, 2020-07-29:
The monitor appear once and it doesn't print.

tttapa, 2020-07-29:
Then you'll have to post the code you're using.

mipabersatu, 2020-07-30:
Oh I found the problem and you're right it's because of the outdated IDE I guess. I updated the program and it fixed. Thanks.

closed 2020-07-30 by mipabersatu