tttapa/Control-Surface #248: First steps with PRO MICRO and MUX on REAPER. Please help.

MiOiM, 2020-07-28:
Hi guys.
I am trying to create my own MIDI controller for REAPER, based on a PRO MICRO (atmega32U4) (Leonardo clone) and this library, which looks really amazing!

To start with a simple design, I have connected a Rotary Encoder to the Pro Micro via a CD74HC4067 mux. The design is like this:

[image]

The code is basically the Rotary Encoder example:

```cpp
#include <Encoder.h> // Include the Encoder library.
// This must be done before the Control Surface library.
#include <Control_Surface.h> // Include the Control Surface library
 
// Instantiate a MIDI over USB interface.
USBMIDI_Interface midi;
 
// Instantiate an analog multiplexer
CD74HC4067 mux = {
  1,       // Analog input pin
  {4, 3, 2, 0} // Address pins S0, S1, S2
};

// Instantiate a CCRotaryEncoder object
CCRotaryEncoder enc = {
  {mux.pin(15), mux.pin(14)},       // pins
  MCU::V_POT_1, // MIDI address (CC number + optional channel)
  1,            // optional multiplier if the control isn't fast enough
};
 
void setup() {
  // Select the correct relative MIDI CC mode.
  // Options:
  //   - TWOS_COMPLEMENT (default)
  //   - BINARY_OFFSET
  //   - SIGN_MAGNITUDE
  //   - NEXT_ADDRESS
  // Aliases:
  //   - REAPER_RELATIVE_1
  //   - REAPER_RELATIVE_2
  //   - REAPER_RELATIVE_3
  //   - TRACKTION_RELATIVE
  //   - MACKIE_CONTROL_RELATIVE
  //   - KORG_KONTROL_INC_DEC_1
  RelativeCCSender::setMode(relativeCCmode::REAPER_RELATIVE_1);
  Control_Surface.begin(); // Initialize Control Surface
}
 
void loop() {
  Control_Surface.loop(); // Update the Control Surface
}
```

And the issue cannot be other that Reaper not detecting the changes when using the encoder.
The device is recognized in Reaper's MIDI Hardware settings, and set as Enabled + Control, but when trying to select it for an Action or using ReaLearn, nothing happens.

I understand that the issue could be:
- Wrong Pro Micro pins for the mux (I'm looking at you, pins 0 and 1)
- Wrong code (so much to learn about Control-Surface and MIDI!)
- Wrong Reaper setup (again...)

I appreciate your help if you see something wrong (or right) in all this. Cheers.

tttapa, 2020-07-28:
You cannot use encoders through a mux, see https://github.com/tttapa/Control-Surface/issues/191.

MiOiM, 2020-07-28:
Oh, I see. Pity.

I was trying to have 7 of them to control the pots of a guitar amp sim vst plugin in Reaper. Made it work with regular pots and the MIDIUSB library, but now I am trying to add a pedalboard with 10 foot switches and 10 leds to it, and I didn't like when changing patches the 'digital' pots jump to the position of the physical ones...

I guess I will have to try with 2 Pro Micro.

Do you know if 2 devices running Control-Surface simultaneously play nice together with DAWs?

Also, amazing job with Control-Surface and all the info you are sharing. Thank you!

tttapa, 2020-07-29:
I wouldn't recommend using two microcontrollers. Getting one microcontroller that supports 7 of them (preferably using interrupts) is a better solution in my opinion. Most Teensy 3.x and 4.x boards can do that.  

Before you buy one though, check that the VST actually supports relative CC messages, many of them don't.

MiOiM, 2020-07-30:
Thank you for your reply, @tttapa 

Following your recommendation, I am doing some tests before getting a Teensy. I have successfully connected button which is recognized in Reaper and can be assigned to Actions.

I'm still struggling with the encoders, though. I have connected one directly to the Pro Micro, to two interrupt pins (pins 0 & 4) and the push button too. This is the code (removed the comments):

```
#include <Encoder.h>

#include <Control_Surface.h>
 
USBMIDI_Interface midi;
 
CCRotaryEncoder enc = {
  {0, 4}, 
  MCU::V_POT_1, 
  1,
};

CCButton button = {
  9,
  {MIDI_CC::General_Purpose_Controller_1, CHANNEL_1},
};

void setup() {
  RelativeCCSender::setMode(relativeCCmode::REAPER_RELATIVE_1);
  Control_Surface.begin(); 
}
 
void loop() {
  Control_Surface.loop();
}
```

Probably the issue is in Reaper settings. I have added it as a MIDI device, with Input and Control enabled:
[image]

When mapping the button, Reaper detects it:
[image]

However, the encoder is never detected. Any idea of what I am doing wrong?

tttapa, 2020-07-30:
I think the problem is that both `MCU::V_POT_1` and `General_Purpose_Controller_1` use the same controller number (0x10). Try a different controller numbers for the button and the encoder.

MiOiM, 2020-07-30:
Tried with General_Purpose_Controller_2 but no luck. Then commented out the whole CCButton instance, leaving just the encoder and still nothing...

Do you know MIDI-OX? There I get the info from the button, but nothing from the encoder. Should I get it too?

I will also try with a different encoder, in case it's a defective one...

tttapa, 2020-07-30:
How did you connect the encoder to the Arduino? Have you tried the examples that come with the Encoder library? That's probably the best way to start debugging.

MiOiM, 2020-07-30:
I have been stupidly thinking that the first pin in the row of the Pro Micro is pin 0 and the next one is pin 1. Obviously is the other way around... Damn, I'm feeling so dumb...

@tttapa, really appreciate your help. Thanks a lot!

closed 2020-07-30 by MiOiM

MiOiM, 2020-07-30:
Wow! It works extremely well... 
The VST plugin is TH-U. I'm mapping the encoders and the button to Reaper via ReaLearn...
Now I want bag full of encoders, a mountain of buttons and to find out how to use leds... Soon...
This is amazing :D

tttapa, 2020-07-30:
Glad to hear, enjoy!