tttapa/Control-Surface #249: Burton fast and long click

djoulz22, 2020-07-28:
Hi Peter,

Is it possible to have 2 actions on buttons?

I see in the code the "long press" term but don't really understand the goal.

Si, for exemple :
If I click the button, i send CC to select track
If I click long time or latch it, i change the bank, like in my last question.

Thanks in advance ans long life to you and your library

tttapa, 2020-07-30:
You could try something like this:

```cpp
#include <Control_Surface.h>

USBDebugMIDI_Interface midi;

IncrementButton button = {2}; // pin

void setup() {
  Control_Surface.begin();
  button.begin();
  Serial.begin(115200);
}

MIDIAddress address = {0x10, CHANNEL_1};

void loop() {
  Control_Surface.loop();

  static uint8_t addressOffset = 0;
  const uint8_t maxOffset = 10;
  switch (button.update()) {
    case IncrementButton::IncrementShort:
      Control_Surface.sendCC(address + addressOffset, 127);
      break;
    case IncrementButton::IncrementLong: // fallthrough
    case IncrementButton::IncrementHold:
      ++addressOffset;
      if (addressOffset >= maxOffset)
        addressOffset = 0;   
      Serial.println(addressOffset);
      break;
    default: break;
  }
}
```
It requires some tuning though, and you need the latest update (ae7aa01da639b6c7e991a739a6f5751fd91bd1e0). You can edit `LONG_PRESS_DELAY` and `LONG_PRESS_REPEAT_DELAY` in [AH/Settings/Settings.hpp](https://tttapa.github.io/Control-Surface-doc/Doxygen/d5/d4f/AH_2Settings_2Settings_8hpp_source.html).

djoulz22, 2020-08-01:
Thanks, Peter, I'll try it.