tttapa/Control-Surface #250: Multiple virtual cables for MIDI Input

Soberat, 2020-07-30:
I'm having an issue with Traktor where outputting data from a single deck works fine, but when I add the second deck to the output they refuse to work properly. So the solution would seem to be to use 2 virtual cables, where theoretically the outputs wouldn't influence one another. 

Now I know that you can use virtual cables for output using the MIDI_Address class with the cableNumber parameter, however unless the Teensy 3.2 shows up as 2 different devices in Traktor I can't route it differently (All Ports sends out to all devices on the list)
[image]

So I'm confused - how would I declare multiple cables so they could show up as multiple devices?
(The output from Traktor is not standard configurable MIDI output. It's obtained by pretending to be a Numark Mixdeck - it's the only way to obtain some track data, hence the workarounds)

tttapa, 2020-07-30:
The Arduino Leonardo only supports a single virtual cable. Teensy 3.x and 4.x support either 1, 4 or 16 virtual cables, by using the USB Type option in the Tools menu of the Arduino IDE (`MIDI`, `MIDIx4` or `MIDIx16`).

To use it in your code, you simply add the cable number as a third argument to the `MIDIAddress` constructor: https://tttapa.github.io/Control-Surface-doc/Doxygen/d7/d0f/classMIDIAddress.html#abdecb57eff7fb838cb3ba7d6d86cbcfa

For example `CCButton button = {2, {0x10, CHANNEL_1, 5}};` uses virtual cable 5 (i.e. the sixth cable). Future releases will use constants just like the `CHANNEL_n` constants, so it will be `CCButton button = {2, {0x10, CHANNEL_1, CABLE_6}};` but these changes haven't been merged into master yet.

Soberat, 2020-08-08:
To confirm the behavior, choosing the MIDIx4 option using a Teensy will make 4 MIDI devices show up, as shown here in Traktor:
[image].
So the issue is resolved!

Also, thank you for the library. It's an absolute game changer!

closed 2020-08-08 by Soberat