tttapa/Control-Surface #253: Don't get Bankable::CCRotaryEncoder to work, error does not name type

bobtorxx, 2020-08-05:
Hallo Pieter,

First of all, this library and the work you put in to it is incredible! Its super awesome and its works fabulous! Through your examples and documentation I learned a lot about coding in Arduino. I had very little experience in coding en through this project of creating a midi controller I learned a lot so also thanks for that.
At the moment I'm working on a controller. The idea is to have a bankable rotary encoder with a increment switch which scrolls through 3 banks. The 3 banks are connected with 3 leds (RGB) so that all the settings together get a colour.
 
I made it work with a potentiometer and a switch. However no matter what I try with example of the Encoder I don't seem to get it work. I get the error: 

exit status 1
'CCRotaryEncoder' in namespace 'CS::Bankable' does not name a type"

What am I doing wrong or is it not possible?

I hope this is the right place to ask for such a question, it is the first time I post something on GitHub so excuse me if its the wrong formatting or place. 

Thanks again! and greetings from Gijs 

```cpp

#include <Control_Surface.h> // Include the Control Surface library

// Instantiate a MIDI over serial interface.
auto &serial = Serial;
SerialMIDI_Interface<decltype(serial)> midi = {serial, MIDI_BAUD};

const pin_t redPin = 5;
const pin_t greenPin = 6;
const pin_t bluePin = 7;

Bank<3> bank;
// Selector to change banks
IncrementSelector<3> selector = {
  bank,       // Bank to manage
  2,          // push button pin
};

/*
  // Instantiate a CCPotentiometer object
  Bankable::CCPotentiometer potentiometer = {
  {bank, BankType::CHANGE_ADDRESS},     // Bank configuration
  A0,                                   // Analog pin connected to potentiometer
  {0x01}, // midi cc adress
  };
*/

Bankable::CCRotaryEncoder enc = {
  {bank, BankType::CHANGE_ADDRESS},     // Bank configuration
  {2, 3}, //pins of encoder
  {0x01}, // midi cc adress
  1,     //multiplier
  4,     //4pulses per step
};

using namespace MIDI_Notes;

CCRangeLEDsPWM<3> RGB = { .      // RGB LED
  { redPin, greenPin, bluePin }, // LED pin list
  0x01,                          // start address of range
};

void setup() {
  Control_Surface.begin(); // Initialize Control Surface
  Serial.begin(115200);
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface
}
```

tttapa, 2020-08-05:
You have to include the Encoder library first, please see the [documentation](https://tttapa.github.io/Control-Surface-doc/Doxygen/da/d40/classBankable_1_1CCRotaryEncoder.html#details) and [this example](https://tttapa.github.io/Control-Surface-doc/Doxygen/d2/d1f/RotaryEncoder_8ino-example.html).

There's also a stray period on this line:
```cpp
CCRangeLEDsPWM<3> RGB = { .      // RGB LED
                          ^~~~

bobtorxx, 2020-08-05:
Thanks for your fast response. Got it working :) So stupid of me I should knew that but I guess I have been staring at the screen for to long to see it. Now I got it working I see that wen I turn the encoder it doesn't increment. It jumps from a value of 1 and then immediate to 127. Instead of incrementing in steps (like the example with te potmeter). Thanks again

tttapa, 2020-08-05:
Yes, that's the intended behavior. CCRotaryEncoder sends relative messages. 127 represents a value of -1 in two's complement. You have to select a relative mode in your DAW, or use CCAbsoluteEncoder.

luca-art, 2020-08-18:
By the way, and hello, can CCAbsoluteEncoder be used within a bank?  I did not see it in bankable list. And with a multiplexer
CD74HC4067?
Thank you

tttapa, 2020-08-18:
At the moment, this is not possible with the `master` version, but you can use the `share-encoder` branch: https://tttapa.github.io/Control-Surface-doc/share-encoder/Doxygen/d7/d2c/classBankable_1_1CCAbsoluteEncoder.html

luca-art, 2020-08-18:
Ok thanks