tttapa/Control-Surface #254: Multiple USB - Control-Surface - Naming?

CyborgGarfunkel, 2020-08-10:
Hi,

I have not some nicely working USB Interfaces. How can I name them individually. I found how to set differend MIDI devices within one USB device. But how can I add more 'Control-Surface's? The name.c file is not seem to fit here. At least if gave me some stange 'if then' errors. How can I have individually named USB interfaces?

tttapa, 2020-08-10:
I'm not sure what you mean. Do you want to use multiple Arduinos or Teensies on a single computer, or do want multiple virtual MIDI over USB cables on a single Teensy?

If you're using a Teensy, you can edit the names in https://github.com/tttapa/Control-Surface/blob/master/src/MIDI_Interfaces/Teensy-USB-MIDI-name.py, then run it using Python, and it'll update https://github.com/tttapa/Control-Surface/blob/master/src/MIDI_Interfaces/Teensy-USB-MIDI-name.c.

When you get errors, please post the exact errors.

CyborgGarfunkel, 2020-08-10:
Thanks for your quick reply and your work that makis fantastic controllers possible!  Yes. I want to use multiple Arduinos or Teensies on a single computer. I was looking for a solution to leave the device name within the main xyz.ino file. If I change the Teensy-USB-MIDI-name.xyz then I have to edit the Teensy-USB-MIDI-name.xyz on every time.

tttapa, 2020-08-10:
That's not possible, except maybe by making the Teensy-USB-MIDI-name.c declarations weak. I don't know if that'll work either.

CyborgGarfunkel, 2020-08-11:
Ok. So I edited the Teensy-USB-MIDI-name.py but it did not change the device name. 
`midi_name = 'My Control-Surface 01'
midi_port_names = [
    'My Control-Surface 01',`

tttapa, 2020-08-11:
Did you run the Python script?

If you just delete Teensy-USB-MIDI-name.c entirely, you should be able to set the names in your sketch.

CyborgGarfunkel, 2020-08-11:
Ok. Cool. That worked. In the Teensy-USB-MIDI-name.c is a comment that said:
`// Edit Teensy-USB-MIDI-name.py instead.`
I assumed that would be enough. `// Edit and run Teensy-USB-MIDI-name.py instead.` would have cleared that. Now it's perfectly named and I can do 20 more interfaces :-)

CyborgGarfunkel, 2020-08-13:
I have now 2 devices constantly connected. They are named differendly. Every time I restart Windows the Name is set 2x sometimes to one - sometimes to the other device. But the 2 devices are not listed both. In my DAW they are numbered if I open the controller window. I errased the USB key from the registry. I named them without spaces in the filename... any ideas?

tttapa, 2020-08-13:
I don't know. Last time I used multiple audio interfaces in Windows, it messed up all the names, everything had the name of the first name. This is either a problem with Windows or the Teensy Core, so it would be better suited on the PJRC forums.

CyborgGarfunkel, 2020-08-13:
Thanks I'll check there and hopefully come back with a solution.

CyborgGarfunkel, 2020-08-14:
Alternatively can you give a recommendation for a board that you would use for multiple devices. It's probably more productive to change the board rather than looking of a solution on the Teensy 3.2.

tttapa, 2020-08-14:
I'd use a Teensy as well. I don't have multiple Teensy 3.2s to test this, but given my previous experience with Windows, I suspect that this is a Windows issue, not necessarily a Teensy issue.

CyborgGarfunkel, 2020-08-17:
I tried to errase all registry and uninstalled the interfaces but that did not help. I tried to manually set serial numbers etc. no result. 

The **only way around it** was to define the first interface as 'MIDI + SERIAL' and the 2nd 'MIDI'. So at least that works if you only have 2 interfaces running sumultaniously.

closed 2020-08-17 by CyborgGarfunkel

benwadub, 2020-08-23:
hi could you explain exactly how you did it to change the name please?

CyborgGarfunkel, 2020-08-24:
Sure.  Pieter prepared all tis nicely!

1. Go to: C:\Program Files (x86)\Arduino\libraries\Control-Surface-master\src\MIDI_Interfaces
2. Edit the 'Teensy-USB-MIDI-name.py' to your liking.
3. Execute the Teensy-USB-MIDI-name.py' 
4. Compile your script again and upload it to your arduino.

Be sure the you have Python installed first. After it's installed the easiest way to execute the 'Teensy-USB-MIDI-name.py' is by simply drag & drop it on the python.exe

Trickster-git, 2022-12-28:
> Sure. Pieter prepared all this nicely!
> 
> 1. Go to: C:\Program Files (x86)\Arduino\libraries\Control-Surface-master\src\MIDI_Interfaces
> 2. Edit the 'Teensy-USB-MIDI-name.py' to your liking.
> 3. Execute the Teensy-USB-MIDI-name.py'
> 4. Compile your script again and upload it to your arduino.
> 
> Be sure the you have Python installed first. After it's installed the easiest way to execute the 'Teensy-USB-MIDI-name.py' is by simply drag & drop it on the python.exe

Hi, This work for TEENSY boards, how change name Pi pico board?