tttapa/Control-Surface #255: CC switch toggles LED... I know BUT...

CyborgGarfunkel, 2020-08-10:
What did I overlook? It should be pretty simple. I have a CCButton or a CCButtonLatched connected to a pin. That PIN get's current what the CC is send from my DAW, right? So the LED and the Switch are at the same PIN. Just repeating basics here to avoid having misunderstood some basic concepts here. If I need to define a PIN for the LED the 'Custom-Note-LED-Input-Element-Callback' example did not make sense. Could anyone bring some light into this please?!?

tttapa, 2020-08-10:
I don't understand what you mean. A CCButton(Latched) only reads the state of an input pin, and sends a MIDI message to the DAW, not the other way around. CCValueLED only receives messages from the DAW, and sets the state of an output pin.

You cannot have a CCValueLED object and a CCButton object on the same pin.

CyborgGarfunkel, 2020-08-11:
Ok. So at the end it's pretty easy:

`// First instantiate the botton object
CCButton buttons[] = {
  {0, {0x00, CHANNEL_1}},
  {1, {0x01, CHANNEL_1}},
};

// instantiate the LED object
CCValueLED leds[] = {
  {5, {0x00, CHANNEL_1}}, 
  {6, {0x01, CHANNEL_1}},
};`

Nice. It works!

CyborgGarfunkel, 2020-08-11:
Can I use any Pin or does it have to be PWM Pins?

tttapa, 2020-08-11:
It doesn't have to be a PWM pin.

closed 2020-08-13 by CyborgGarfunkel