tttapa/Control-Surface #256: How to display current Bank/how to get selected bank

AnnoyedSandwich, 2020-08-12:
Lets say i have four banks with an IncrementDecrementSelector. 
I know that you can use different colored LEDs to display the current selecte bank but displaying the current bank on an lcd display would be sleker for a MIDI controller. 
I didnt found anything in the examples or the documentation so I woudl appreciate some help.

tttapa, 2020-08-12:
You can use [`Bank::getSelection()`](https://tttapa.github.io/Control-Surface-doc/Doxygen/db/dbd/classBank.html#a883bbbd61bff47fdc24105ba663d622b).

AnnoyedSandwich, 2020-08-12:
can you post a small example, please? When i try it i get the error message "Argument list for class template "CS::Bank" is missing

tttapa, 2020-08-12:
Please post your code. It's a member function, you apply it to an instance of the `Bank<>` class.

```cpp
Bank<4> bank;

// ...

void loop() {
  // ...
  auto currBank = bank.getSelection();
  // do something with currBank
}
```

AnnoyedSandwich, 2020-08-12:
yess, thank you so much. Now it works.

closed 2020-08-12 by AnnoyedSandwich

benwadub, 2020-10-21:
hi @AnnoyedSandwich could you explain how you displayed the bank on the screen after getting the bankselection please?

tttapa, 2020-10-21:
If you want to display it on a graphical display, you could use the [`BankDisplay`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d6/dc4/classBankDisplay.html) class.

benwadub, 2020-10-22:
if my bank always refer to track 1 to 8 but only change the midi address can I use it?
i like that my display always show 12345678 that are the numbers of my tracks and just show something like A and B to know if my pots are on bank A or bank B, with the exemple I found when I press the bank button the display show me track 9 to 16 but I don t want 16 tracks is it doable with this class or do I misunderstand something?
[image]
[image]
```cpp
// Bank seting
BankDisplay bankDisp[] = {
  // first track of the bank (1), position (0, 50), font size (2)
  {display, bank, 1, {3, 10}, 2, WHITE},
  {display, bank, 2, {35, 10}, 2, WHITE},
  {display, bank, 3, {67, 10}, 2, WHITE},
  {display, bank, 4, {99, 10}, 2, WHITE},
  {display, bank, 5, {3, 35}, 2, WHITE},
  {display, bank, 6, {35, 35}, 2, WHITE},
  {display, bank, 7, {67, 35}, 2, WHITE},
  {display, bank, 8, {99, 35}, 2, WHITE},```

tttapa, 2020-10-22:
The images aren't working, you have to wait for them to upload completely before clicking "comment".

The BankDisplay class takes the offset between tracks into account, if you want different behavior, it's easy enough to implement your own class. The implementation of BankDisplay is really simple:
https://github.com/tttapa/Control-Surface/blob/700cfb580ca30c6456f0c9b8369b64844c4578c8/src/Display/SelectorDisplay.hpp#L34-L56

You could use `bank.getSelection()` instead of `bank.getOffset()`, for example.

benwadub, 2020-10-22:
i understood that just after sending the message, I found an easy way to counter my problem, I created a bank 2 with no offset then assigned the bank display to this bank2 that finally do nothing certainly not the best coding tricks ever but that do the job ;-)