tttapa/Control-Surface #257: Issues with using Hairless midi

StutteA, 2020-08-13:
I'm using Hairless midi together with loopMidi to be able to hook it up to fl studio. I'm having no issues when using the USBDebugMIDI_Interface, it debugs everything properly in the serial monitor, although when I use HairlessMIDI_Interface together with the Hairless software something is messed up as hell. I mostly get a lot of "Warning: got a status byte when we were expecting 2 more data bytes, sending possibly incomplete MIDI message 0xc0". Sometimes the number of data bytes it was expecting is 1, and the incomplete midi message seems to be more or less random.
The midi messages themselves are absolute chaos, you can see it for yourself in the image. 

If I start up loopMidi and set Hairless's output to be loopMidi the program closes as soon as one of the warning messages gets recieved.

This is the code I used:
```cpp
#include <Control_Surface.h>

HairlessMIDI_Interface midi;
CCPotentiometer potentiometer = { A0, {0, CHANNEL_1} };

void setup() {
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
}
```

This is the debug from Hairless:
[image]
[image]

tttapa, 2020-08-13:
This sounds like a baud rate issue. `HairlessMIDI_Interface` expects the default baud rate in Hairless, 115200. If you select a different rate in the Hairless settings, it won't work.

StutteA, 2020-08-14:
That was exactly it! Thank you so much!!!

closed 2020-08-14 by StutteA