tttapa/Control-Surface #259: Is it possible to use multiplexer / demultiplexers together with button matrix?

StutteA, 2020-08-16:
Is it possible to lower the amount of pins a button matrix needs, without too much delay and hassle?
I'm not completely sure on how a button matrix works, and I read about multiplexers and demultiplexers, and from what I know about button matrixes I should be able to use a multiplexer to read the input. But I'm not really sure, so any help is super appreciated! 

Small extra question: how many multiplexers can you use before the delay becomes noticeable?

m0ck0, 2020-08-16:
i know this one! well, some of it. here is a basic example of the code im using for it (i have the second multiplexer assigned to pots, but i chopped that part for clarity, also sorry about the messed up order in the pins)

for this to work the key are wired in 8 gorups of 7 from one side they're wired to the multiplexer, and from the other side to the arduino with a diode in series 

here is a little diagram of how its wired (hope it makes sense to you)

multiplexer
   l
   l____/  ____->l- ___(arduino input)
   l____/  ____->l-____
   l____/  ____->l-____
   l____/  ____->l-____
   l____/  ____->l-____
   l____/  ____->l-____
   l____/  ____->l-____
        (key)   (diode)

as for latency i really don't know but i think as long as you aren't cascading the multiplexers it won't be noticeable. 

```
#include <Arduino_Helpers.h> // Include the Arduino Helpers library
#include <AH/Hardware/ExtendedInputOutput/AnalogMultiplex.hpp>
#include <Control_Surface.h>

HairlessMIDI_Interface midi;

 CD74HC4051 mux[] = { 
  { A4, {2, 3, 4}},
  { A5, {2, 3, 4}}
 };

const AddressMatrix<7, 8> addresses = {{

  {36,  42,  48,  54,  60,  66,  72,  78},
  {37,  43,  49,  55,  61,  67,  73,  79},
  {38,  44,  50,  56,  62,  68,  74,  80},
  {39,  45,  51,  57,  63,  69,  75,  81},
  {40,  46,  52,  58,  64,  70,  76,  82},
  {41,  47,  53,  59,  65,  71,  77,  83},
  { 0,   0,  84,   4,   5,   8,   7,   6}

  }};
  

 NoteButtonMatrix<7, 8> buttonmatrix = {
  { 15, 14,  8, 6, 7, 13, 16},  {mux[0].pin(4), mux[0].pin(1), mux[0].pin(2), mux[0].pin(3), mux[0].pin(0), mux[0].pin(5), mux[0].pin(7), mux[0].pin(6),},
  addresses,    // address matrix
  CHANNEL_1,    // channel and cable number
};
   
void setup() {

  Control_Surface.begin();
  
    mux[0].begin();                 
    mux[0].pinMode(0, INPUT_PULLUP);
}      

void loop() {
    Control_Surface.loop();
}
```

StutteA, 2020-08-16:
This is really helpful! Although there's a lot going on, so I have a few questions about some things that are a bit unclear.

> i know this one! well, some of it. here is a basic example of the code im using for it (i have the second multiplexer assigned to pots, but i chopped that part for clarity, also sorry about the messed up order in the pins)
Does this mean that it is possible to use a multiplexer for the pots? 

>#include <Arduino_Helpers.h> // Include the Arduino Helpers library
>#include <AH/Hardware/ExtendedInputOutput/AnalogMultiplex.hpp>What are these doing, and do I need to think about it?

StutteA, 2020-08-16:
This is really helpful! Although there's a lot going on, so I have a few questions about some things that are a bit unclear.

>i know this one! well, some of it. here is a basic example of the code im using for it (i have the second multiplexer assigned to >pots, but i chopped that part for clarity, also sorry about the messed up order in the pins)

Does this mean that it is possible to use a multiplexer for the pots?

>#include <Arduino_Helpers.h> // Include the Arduino Helpers library
>#include <AH/Hardware/ExtendedInputOutput/AnalogMultiplex.hpp>

What are these doing, and do I need to think about it?

> CD74HC4051 mux[] = { 
>  { A4,                           // Analog input pin
>    {2, 3, 4}},                  // Address pins S0, S1, S2 
>  { A5,                          // Analog input pin
>    {2, 3, 4}}                  // Address pins S0, S1, S2
> };

Can you really set the adress pins to be the same? Won't that mess something up? Because if you can do that then holy hell, I just got more pins to play with.

tttapa, 2020-08-18:
> Does this mean that it is possible to use a multiplexer for the pots?

Yes, you can use multiplexers for potentiometers, see https://tttapa.github.io/Control-Surface-doc/Doxygen/d5/d7d/md_pages_Getting-Started.html.

> > #include <Arduino_Helpers.h> // Include the Arduino Helpers library
> > #include <AH/Hardware/ExtendedInputOutput/AnalogMultiplex.hpp>
> 
> What are these doing, and do I need to think about it?

You don't need those, just including `Control_Surface.h` should be enough.

> Can you really set the adress pins to be the same? Won't that mess something up? Because if you can do that then holy hell, I just got more pins to play with.

Yes, if you use multiplexers as inputs, you can use the same address pins for all multiplexers, because you just set them right before reading the pin in question.   
If you're using a demultiplexer, that's not possible, of course.

> (Edit)
> This is more of a electronics question that anything, but are all multiplexers analog? Or are some analog and the rest digital?

Not all multiplexers are analog, but many popular multiplexers used by hobbyist are analog.

StutteA, 2020-08-20:
Super sorry for late reply, but huge thanks for the answers!

closed 2020-08-20 by StutteA