tttapa/Control-Surface #261: Trouble with Multiplexer

AnnoyedSandwich, 2020-08-18:
So im currently trying out the library to build a midi controller in the future. But i having some troubles with multiplexing. The connected buttons dont work and the potentiometer is constantly making control change events. This happens without truning the knob and even whne i disconnect the 
Im 95% sure that the wiring is correct so I think there is a problem with my code. I you have the time could you take a look at it and maybe find my mistake? And if my code is correct do you have any idea why nothing related to the multiplexer works?

```cpp
#include <Control_Surface.h> // Include the Control Surface library
#include <Wire.h>
#include <LiquidCrystal_I2C.h>

// Instantiate a MIDI over USB interface.
HairlessMIDI_Interface midi;
using namespace MIDI_Notes;

//------------------Bank (here everything works)------------------------------------------------
Bank<2> bank(1); 
// Instantiate a Bank selector to control which one of the four Banks is active.
IncrementDecrementSelector<2> selector = {
    bank,       // Bank to manage
    {2, 3},     // push button pins (increment, decrement)
    Wrap::Wrap, // Wrap around
};
// Instantiate a NoteButton object
Bankable::NoteButton button1 = {
  {bank, BankType::CHANGE_CHANNEL},
  5,                       // Push button on pin 5
  {note(C, 4)}, // Note C4 on MIDI channel 1
};
//------------------Mux------------------------------------------------
CD74HC4067 mux = {
  A0,
  {8, 9, 10, 11}
  };

NoteButton buttons[] = {
  {mux.pin(0), {note(C, 2), CHANNEL_1}},
  {mux.pin(1), {note(Db, 2), CHANNEL_1}}
};

CCPotentiometer pottesst[] = {
  {mux.pin(2), {MIDI_CC::Channel_Volume, CHANNEL_1}},
};

LiquidCrystal_I2C lcd = LiquidCrystal_I2C(0x27, 16, 2);

void setup() {
  Control_Surface.begin(); 
  lcd.init();              
  lcd.backlight();

}
 
void loop() {
  Control_Surface.loop(); // Update the Control Surface
  int currBank = bank.getSelection() + 1;
  lcd.print(String(" Bank ") + String(currBank)); //to display the current bank
  lcd.setCursor(1,0);
}
```

AnnoyedSandwich, 2020-08-18:
when I disconnect the multiplexer, the nonsensical messages are still showing up, so I don't know what's going on.

tttapa, 2020-08-18:
The code looks alright, and it's very unlikely that a problem in your code would cause random messages like that. This is almost certainly a hardware problem. Please post a wiring diagram and a picture of your actual connections.

AnnoyedSandwich, 2020-08-18:
[image]

AnnoyedSandwich, 2020-08-18:
I disconnected everything except the potentiometer, still, nothing on the Hairless MIDI debug window.

tttapa, 2020-08-18:
The header pins aren't soldered to the PCB, it doesn't make proper contact like that.

tttapa, 2020-08-18:
Getting nonsensical values when you disconnect everything is normal, as it leaves the input pin floating, so it's just an antenna and picks up noise or you measure capacitive coupling with the surroundings. 

You can test your code by just connecting the potentiometer to pin A0 directly, without using the multiplexer. It should then affect all buttons and the CCPotentiometer.

AnnoyedSandwich, 2020-08-18:
ok thanks, soldering the header worked (although the seller told me that wouldn't be necessary)
Other question, can you bank button or Potentiometers which are connected to a multiplexer?

tttapa, 2020-08-19:
> Other question, can you bank button or Potentiometers which are connected to a multiplexer?

Yes, just initialize it with the right pin (`mux.pin(#)`).