tttapa/Control-Surface #262: CC Value to LED Brightness

CyborgGarfunkel, 2020-08-19:
I would like to use a LED or later a VU meter to display a CC value. So would need to set the brightness to the CC. I'm not sure it's the analogwrite or NoteCCFastLEDCallback function. The LED's are connected to PWM Pins.

My current simple function looks like this. 

```
CCValueLED leds[] = {
  {3, {0x00, CHANNEL_1}}, // LED pin, address (note number, channel, cable)
  {4, {0x01, CHANNEL_1}}, // LED pin, address (note number, channel, cable)
  {5, {0x02, CHANNEL_1}}, // LED pin, address (note number, channel, cable)
  {6, {0x03, CHANNEL_1}}, // LED pin, address (note number, channel, cable)
 
};
```

tttapa, 2020-08-19:
You can use the `CCValueLEDPWM` class. Please see the documentation (the "modules" page in particular) for an overview of the available classes: https://tttapa.github.io/Control-Surface-doc/Doxygen/d7/de4/group__midi-input-elements-leds.html

CyborgGarfunkel, 2020-08-19:
Woooo. That's so so so cool. You have no idea how happy I am with all the tools I'm building these day's thanks to your work. Boah. Ok. So I simply changed it to this and now the LED's shine according to its submitted value... Insane!

```
CCValueLEDPWM leds[] = {
  {3, {0x00, CHANNEL_1}}, // LED pin, address (note number, channel, cable)
  {4, {0x01, CHANNEL_1}}, // LED pin, address (note number, channel, cable)
  {5, {0x02, CHANNEL_1}}, // LED pin, address (note number, channel, cable)
  {6, {0x03, CHANNEL_1}}, // LED pin, address (note number, channel, cable)
 };
```

CyborgGarfunkel, 2020-08-19:
I added a **VU Meter** instead of the LED to show Total Level and the L/R phase differences. So you can use a inexpensive VU Meter like this one: `shorturl.at/rzG89` and just plug it at any PWM pin. To show the right values it's neccesary to insert a 4K7 (47K) Resistor. But I works just fine.

closed 2020-08-19 by CyborgGarfunkel

Jhenesson, 2022-11-15:
I couldn't run this function on Arduino uno (With Hiduino), do you have any restrictions?