tttapa/Control-Surface #264: Advice needed - 10 encoders, 12 buttons, 5 LEDS, TeensyLC

SwooshMihu, 2020-08-20:
So i have been messing with Control Surface for a while now and i have made a midi controller with 6 encoders, a 3x3 button matrix and 2 button used for controlling the banks. All works well but now i would like to create a bigger one to increase it's functionality. 

Reading the questions posted on github and consulting countless tutorials i can finally say i have no idea how to do the following project and so i am asking you for some advice. I would like to create a midi controller with 10 encoders ( 2 of which must be able to be pressed like a button), 10 buttons(eventually in a matrix), 2 buttons used for navigating the banks and 5 LEDs to signal what bank is selected, all this with a Teensy LC. It is clear that i must use multiplexers for this project but i don't have a clue what a good pick would be and i would very much appreciate your input on this.  Thanks

tttapa, 2020-08-20:
You cannot use multiplexers for encoders, so that's going to be the hardest part. Ideally, you'd have two interrupt pins for each encoder. If you have one interrupt pin per encoder, that'll work too. If an encoder is connected to two non-interrupt pins, the library falls back to polling, which might result in missed pulses, depending on the overall load on the microcontroller. The Teensy LC has only 18 interrupt pins (see https://forum.pjrc.com/threads/27624-Coming-Soon-Teensy-LC-%28low-cost-Teensy%29). If you connect 10 encoders (20 pins) there will barely be any pins left for other things. You could use I2C port expanders for the other buttons and LEDs. These are not directly supported by Control Surface (yet), simply because I haven't had the time yet. I have some port expanders lying around, but I won't be able to start working on it until early september.

I'm not sure what you want help with specifically, most of it sounds similar to your first controller, just with more buttons and controllers. Do you have specific questions?

This issue section is primarily for questions about features of the Control Surface library and bug reports, for general help planning out your project, you'll get much more help on an online forum, such as the Arduino forum.

SwooshMihu, 2020-08-21:
I am looking forward to the implementation of I2C port expanders. Thank you for your answer, it was really helpful!

closed 2020-08-21 by SwooshMihu