tttapa/Control-Surface #269: Mackie control, valor de los niveles de volumen

alomdj00, 2020-08-25:
Hola Peter,  estuve probando el ejemplo de ingenieria inversa para mackie control y al mover el volumen 1 en fl studio y ableton he obtenido valores que no comprendo muy bien, revise el documento oficial de mackie control y los valores que recibi en el monitor serie son iguales a los expuestos en el manual. Mi pregunta es como se puede convertir estos valores en una variable de tipo int o cualquier otra, por ejemplo este valor E0, 40, 55  = Fader Ch. 1, position (55 << 7) + 40   como puedo convertirlo a una variable, me gustaria poder convertirlo para mostrarlo en un lcd. Muchas gracias por tu tiempo y trabajo en esta gran libreria.

tttapa, 2020-08-28:
E0h is a pitch bend message. Right now, there are no specific "MIDI Input Elements" for pitch bend, but you can use the MIDI callbacks directly, see [this example](https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/d32/MIDI-Input-Callback_8ino-example.html). I'm working on a better MIDI input system, which will include pitch bend, but unfortunately I have no time to finish it right now.

alomdj00, 2020-08-28:
He estado haciendo pruebas con el ejemplo que propones y en dispositivos usb nativos va perfecto, sin embargo con el esp32 no he logrado que me muestre en el monitor serie los datos recibidos del daw

tttapa, 2020-09-05:
That's probably because the ESP32 doesn't support MIDI over USB. This means that the `USBMIDI_Interface` uses the same Serial port as the one connected to the Serial monitor.

alomdj00, 2020-09-05:
Puedo redireccionar los mensajes  midi de ble a el monitor serie

tttapa, 2020-09-05:
Yes, simply replace the line `USBMIDI_Interface` by `BluetoothMIDI_Interface` in the example.

alomdj00, 2020-09-07:
En el ejemplo de mackie control ingenieria inversa reemplace el USBMIDI_Interface por BluetoothMIDI_Interface pero no sale nada en el monitor serie, en mi daw lo marque como mackie control pero no sale nada en el monitor, uso esp32 devkit, leds y demas va perfecto en modo mackie, pero el ejemplo de ingenieria inversa o trabaja con el esp en modo ble

tttapa, 2020-09-07:
The following works for me on a SparkFun ESP32 Thing on Ubuntu 20.04 and ESP32 Core version 1.0.4 and Control Surface `master`:

```cpp

#include <Control_Surface.h>

BluetoothMIDI_Interface midi;

bool channelMessageCallback(ChannelMessage cm) {
  if ((cm.header & 0xF0) == 0x90)
    Serial << hex << cm.header << ' ' << cm.data1 << ' ' << cm.data2 << dec 
           << "\t(" <<  MCU::getMCUNameFromNoteNumber(cm.data1)  << ")" << endl;
  else 
    Serial << hex << cm.header << ' ' << cm.data1 << ' ' << cm.data2 << dec << endl;
  return false; // Return true to indicate that handling is done,
                // and Control_Surface shouldn't handle it anymore.
                // If you want Control_Surface to handle it as well,
                // return false.
}
 
bool sysExMessageCallback(SysExMessage se) {
  Serial << F("SysEx: ") << hex;
  for (size_t i = 0; i < se.length; ++i)
    Serial << se.data[i] << ' ';
  Serial << dec << F("on cable ") << se.CN << endl;
  return false; // Return true to indicate that handling is done,
                // and Control_Surface shouldn't handle it anymore.
                // If you want Control_Surface to handle it as well,
                // return false.
}
 
bool realTimeMessageCallback(RealTimeMessage rt) {
  Serial << F("Real-Time: ") << hex << rt.message << dec
         << F(" on cable ") << rt.CN << endl;
  return false; // Return true to indicate that handling is done,
                // and Control_Surface shouldn't handle it anymore.
                // If you want Control_Surface to handle it as well,
                // return false.
}

void setup() {
  // Make sure that the Serial interface is fast enough, so it doesn't stall the MIDI 
  // application
  Serial.begin(1000000);
  Control_Surface.begin();
  Control_Surface.setMIDIInputCallbacks(channelMessageCallback,   //
                                        sysExMessageCallback,     //
                                        realTimeMessageCallback); //
}

void loop() {
  Control_Surface.loop();
}
```

tttapa, 2020-09-08:
> Right now, there are no specific "MIDI Input Elements" for pitch bend, but you can use the MIDI callbacks directly, see [this [...]

I added Pitch Bend input support on the `new-input` branch: https://tttapa.github.io/Control-Surface-doc/new-input/Doxygen/d0/d6c/Pitch-Bend-Value_8ino-example.html

alomdj00, 2020-09-09:
> I added Pitch Bend input support on the `new-input` branch: https://tttapa.github.io/Control-Surface-doc/new-input/Doxygen/d0/d6c/Pitch-Bend-Value_8ino-example.html

He copiado el ejemplo y me tira error, PBValue no es un tipo de nombre, acabo de descargar la libreria, por lo que no se a que se deba el error

tttapa, 2020-09-09:
Did you check out the `new-input` branch?

alomdj00, 2020-09-10:
Borre la anterior libreria y descargue e instale la control surface new input, si embargo no compila ningun ejemplo

tttapa, 2020-09-11:
I assume the problem is that you have to download the submodules, they are not included when you click the "Download ZIP" button on GitHub.

If you use Git, you have to run:
```sh
cd "path-to-Arduino/libraries"
git clone https://github.com/tttapa/Control-Surface.git --branch new-input
cd Control-Surface/
git submodule update --init
```