tttapa/Control-Surface #270: Help implementing latching buttons.

GinjahWolf92, 2020-08-26:
Hey! So I had a query regarding some of my latching buttons I own and I was wondering how I would be able to make them send a MIDI signal to my DAW. The buttons connect to a digital pin on the Ard. Board but my DAW doesn't register the button presses. Any help would be appreicated greatly!

```cpp
#include <Control_Surface.h>

USBMIDI_Interface midi;

CCPotentiometer potentiometer1 = { A3, {MIDI_CC::Channel_Volume, CHANNEL_1} };
CCPotentiometer potentiometer2 = { A2, {MIDI_CC::Channel_Volume, CHANNEL_2} };
CCPotentiometer potentiometer3 = { A1, {MIDI_CC::Channel_Volume, CHANNEL_3} };
CCPotentiometer potentiometer4 = { A0, {MIDI_CC::Channel_Volume, CHANNEL_4} };
CCPotentiometer potentiometer5 = { A10, {MIDI_CC::Channel_Volume, CHANNEL_5} };
CCPotentiometer potentiometer6 = { A9, {MIDI_CC::Channel_Volume, CHANNEL_6} };

// Instantiate a CCButton object
CCButton button1 = {1, {MIDI_CC::General_Purpose_Controller_1, CHANNEL_7}};
CCButton button2 = {3, {MIDI_CC::General_Purpose_Controller_1, CHANNEL_8}};
CCButton button3 = {0, {MIDI_CC::General_Purpose_Controller_2, CHANNEL_9}};
CCButton button4 = {2, {MIDI_CC::General_Purpose_Controller_1, CHANNEL_10}};
CCButton button5 = {6, {MIDI_CC::General_Purpose_Controller_1, CHANNEL_11}};
CCButton button6 = {4, {MIDI_CC::General_Purpose_Controller_1, CHANNEL_12}};
CCButton button7 = {5, {MIDI_CC::General_Purpose_Controller_1, CHANNEL_13}};

void setup() {
    Control_Surface.begin();

// Uncomment the following to invert the potentiometers.
   
   // potentiometer1.invert();
   // potentiometer2.invert();
   // potentiometer3.invert();
   // potentiometer4.invert();
   // potentiometer5.invert();
   // potentiometer6.invert();
}

void loop() {
    Control_Surface.loop();
}
```

GinjahWolf92, 2020-08-26:
Okay, since leaving the above issue, I have manged to find out that I can get code to work however I have noticed I would need to flip my 5v and GND - is there a way to flip the expected input to get the desired behaviour to not need to redo all of my wiring?

GinjahWolf92, 2020-08-26:
The above comment is refrencing the `CCButton` and NOT the `CCButtonLatching`

tttapa, 2020-08-26:
You shouldn't have 5V connected to your switches. The internal pull-up resistor is used, so switches are connected between ground and the Arduino pin. If you have normally closed switches, you can use `CCButton::invert()` in your setup.

GinjahWolf92, 2020-08-27:
Okay, thanks so much for your help! I have another query but I will post in new question as it unrelated to this. :)

closed 2020-08-27 by GinjahWolf92