tttapa/Control-Surface #272: midi interface with display and menu

benwadub, 2020-08-29:
hi Pieter, i d like to make a kind of multitool midi interface,
i d like it to be a bridge between usb midi and serial midi, I know now how to set the midi interface but I d like to add a screen, encoder and push button to define the routing between the midi interfaces, so lots of question and remember I m a beginner at all this!
1 is it doable to switch on/off midi pipes with rotary encoder and push button?
2 what display should I buy to start my work that would be certain to work with your library?
3 can you help me setting a menu navigation in the library?
or maybe do you have a link that could help me learning all that?
thanks

tttapa, 2020-09-05:
1. Yes. You can connect and disconnect pipes at runtime, see the `connectSourcePipe()`, `disconnectSourcePipes()`, and `disconnnect()` methods of the `MIDI_Sink` class. Similarly for the `MIDI_Source` class. `connectSourcePipe()` connects a `MIDI_Pipe` to the sink (you can connect multiple pipes to the same sink or source), `disconnectSourcePipes()` disconnects all pipes connected to the sink, and `disconnect()` disconnects only the pipe that connects the sink to the given source.  
See the documentation in the "Connecting and disconnecting MIDI Pipes" section of the [`MIDI_Sink`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d9/db6/classMIDI__Sink.html) and [`MIDI_Source`](https://tttapa.github.io/Control-Surface-doc/Doxygen/df/d5b/classMIDI__Source.html) classes.  
You can also disconnect a `MIDI_Pipe` by calling the `disconnect()` method. You can also specify a specific source or sink to disconnect from the pipe. If you do that, keep in mind that the other end of the pipe remains connected (which may or may not be what you want). If you call `disconnect()` without any arguments, it disconnects the pipe from all sinks and sources it's connected to, and you can then reuse the pipe somewhere else.
2. It depends on what you want to do with it. The library includes support for small OLED displays such as the SSD1306, because that's what I used for my controllers. If you want larger displays, you can no longer use this method and you'll have to roll your own code anyway.
3. Menu navigation is far beyond the scope of the Control Surface library.

benwadub, 2020-09-05:
thanks I ll try to find an ssd1306 with spirit, all the ones I find are i2c

tttapa, 2020-09-05:
You can use the I2C ones as well, but they are just much slower than SPI, so I always prefer SPI. I got mine off eBay.

There are other much more interesting displays out there, the SSD1306 displays are really small and are hard to synchronize, so they might not be suitable for your application. I just mentioned them because they're the only ones I really have experience with.

benwadub, 2020-09-06:
which one do you recommend instead of the ssd1306?

tttapa, 2020-09-06:
That would be a question for the Arduino or PJRC forum, I don't have enough experience with other displays to recommend anything.