tttapa/Control-Surface #273: 1 Arduino = multiple Midi Interfaces

CyborgGarfunkel, 2020-09-04:
First a question if I understood it correctly. Since I'll seen that the 'Teensy-USB-MIDI-name.py' contains various XT names I understood that an individual Arduino can act as various separate interfaces. I found the pipes concept but couldn't wrap my head around it. If so could you please explain how to declare and address a CCButton to a specific interface. That would be great.

tttapa, 2020-09-05:
The different names are not related to the MIDI Pipes. 

MIDI Pipes are simply a concept to route MIDI traffic between different MIDI interfaces. If you have multiple MIDI interfaces (e.g. a Serial interface and a MIDI over USB interface), you can connect them using pipes. Pipes are virtual connections that transfer MIDI data. This concept is specific for this library.

The different names in `Teensy-USB-MIDI-name.py` are the names for the MIDI USB virtual cables. Each MIDI over USB interface supports up to 16 virtual cables. These show up on the computer as different devices. MIDI USB virtual cables are defined by the MIDI over USB standard, they are not specific to this library.

Because each virtual cable requires its own buffer, they cannot be used on some Arduino boards like the Leonardo, because it doesn't have enough memory. In theory, it should be possible on boards like the Arduino Due, but the MIDIUSB library simply doesn't support it (because it was initially written for the Leonardo). The only boards I know of that support multiple virtual cables are Teensy boards. You can select the `MIDIx4` or `MIDIx16` modes in the `Tools > USB Type` menu for 4 or 16 virtual cables respectively. In your code, you specify the cable number as a third parameter to the `MIDIAddress` constructor, see [MIDI addresses](https://tttapa.github.io/Control-Surface-doc/Doxygen/d6/dc6/md_pages_Basics.html).

For example, a `CCButton` on Controller 111 on channel 2 and cable 5:
```cpp
CCButton button = {
    2,                   // pin number
    {111, CHANNEL_2, 5}, // MIDI address
};
```
For now (in Control Surface v1.x.x), cable numbers are zero-based. This will probably change in version 2.0.0 to be more consistent with the channel numbers.

CyborgGarfunkel, 2020-09-05:
Cool. Thank you. After I created the `MIDIx4` successfully 4 devices show up in my List. Just still the data is send to the main cable.
```
CCButton button[] = {
  {10, {0xA, CHANNEL_1},2},
  {11, {0xB, CHANNEL_1},3},
};
```
Should work, right?

CyborgGarfunkel, 2020-09-05:
Ok. Sorry... Like this it works just fine!

```
CCButton button[] = {
  {10, {0xA, CHANNEL_1, 2}},
  {11, {0xB, CHANNEL_1, 3}},
};
```