tttapa/Control-Surface #274: MCU::VOLUME does not work in Mackie Control surface

luca-art, 2020-09-05:
Hi
This one works ok in Ableton and Cubase like a Mackie surface:

```
NoteButton mute1 = { 2, {MCU::MUTE_1}};
NoteButton mute2 = { 3, {MCU::MUTE_2}};
NoteButton mute3 = { 4, {MCU::MUTE_3}};
NoteButton mute4 = { 5, {MCU::MUTE_4}};
NoteButton mute5 = { 7, {MCU::MUTE_5}};
NoteButton mute6 = { 6, {MCU::MUTE_6}};
NoteButton mute7 = { 8, {MCU::MUTE_7}};
NoteButton mute8 = { 9, {MCU::MUTE_8}};
```

but not this one:

```
CCPotentiometer vol1 = { A15, {MCU::VOLUME_1}};
CCPotentiometer vol2 = { A14, {MCU::VOLUME_2}};
CCPotentiometer vol3 = { A13, {MCU::VOLUME_3}};
CCPotentiometer vol4 = { A12, {MCU::VOLUME_4}};
CCPotentiometer vol5 = { A11, {MCU::VOLUME_5}};
CCPotentiometer vol6 = { A10, {MCU::VOLUME_6}};
CCPotentiometer vol7 = { A9, {MCU::VOLUME_7}};
CCPotentiometer vol8 = { A8, {MCU::VOLUME_8}};
```
nothing happens

If you know it, Could you say me someting about that?
Thank you very much

tttapa, 2020-09-05:
`MCU::VOLUME_n` are meant to be used with Pitch Bend, see https://tttapa.github.io/Control-Surface-doc/Doxygen/da/dc1/group__MCU__PB.html.

You could use `PBPotentiometer` instead of `CCPotentiometer`.

luca-art, 2020-09-05:
Thank you for your quick answer, as always, it works perfectly
bye

closed 2020-09-05 by tttapa