tttapa/Control-Surface #277: Change octave

helderazinheirinha, 2020-09-07:
Hello! Congratulations on the excellent work with this library!
I am trying to convert an old keyboard to USB MIDI. The keyboard has 2 octaves (25 keys) and I wanted to add buttons to change the octave up and down. Is this code okay? (I don't know how to code, I just mixed 3 examples from this library).
```cpp
/**
#include <Control_Surface.h>

USBMIDI_Interface midi;

using namespace MIDI_Notes;

Transposer<-2, +2> transposer (12);

// Instantiate a Selector to change the transposition
IncrementDecrementSelector<transposer.getNumberOfBanks()> selector = {
  transposer,
  {22, 23},
  Wrap::Wrap,
};

// The note numbers corresponding to the buttons in the matrix
 AddressMatrix<4, 7> buttons = {{
  {60, 61, 62, 63, 64, 65, 66},
  {67, 68, 69, 70, 71},
  {72, 73, 74, 75, 76, 77, 78},
  {79, 80, 81, 82, 83, 84},
}};

Bankable::NoteButtonMatrix<4, 7> buttonmatrix = {
  transposer,
  {9, 10, 11, 12}, // row pins
  {2, 3, 4, 5, 6, 7, 8},    // column pins
  buttons,    // address matrix
  CHANNEL_5,    // channel and cable number
};

void setup() {
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
}
```

tttapa, 2020-09-08:
At first sight, it looks alright. (Apart from the fact that your entire program is commented out because of the `/**` on the first line.)

Did you try it out? Did it not work?

helderazinheirinha, 2020-09-30:
Yes it works! :) Thank you for your good work!!
Now I have another issue, I am trying to make a MCU controller with the functions Play, Stop and Rec to use with FL Studio. And the problem is with the Stop button, when I press Stop in the controller the Stop button, in the software, stays "down" (like if i was pressing it down continuously). So, the first time time that I press it down it stops but, stays pressed and will not stop again, unless I press it, in the software, with the mouse.
Can you help me?
I used this code:

`#include <Control_Surface.h>

// Add a MIDI interface to use:
USBMIDI_Interface midi;

// Add transport control buttons:
NoteButton transportButtons[] = {
  {7, MCU::PLAY},   
  {8, MCU::STOP},
  {9, MCU::RECORD},
};

void setup() {
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
}`

tttapa, 2020-09-30:
That sounds like a problem with FL studio, Control Surface just sends the state of the buttons (it sends an "On" event when the button is pressed, and an "Off" event when the button is released). How these events are interpreted are up to FL studio.

You can try the `USBDebugMIDI_Interface` instead of `USBMIDI_Interface` to look at the MIDI messages in the serial monitor, and see if they are what you expect.

helderazinheirinha, 2020-09-30:
Thank you for your reply! I used the _**USBMIDI_Interface**_ and this is the result:
`15:41:10.778 -> Note On         	Channel: 1	Data 1: 0x5d	Data 2: 0x7f	Cable: 1
15:41:11.019 -> Note Off        	Channel: 1	Data 1: 0x5d	Data 2: 0x7f	Cable: 1`

I don't know if this is what is suposed to be...

I also used the  _**Mackie Control Universal Reverse Engineering**_ sketch and pressed the Stop buttun on FL Studio and this is the result:
  -> b0 7b 00
  -> b1 7b 00
  -> b2 7b 00
  -> b3 7b 00
  -> b4 7b 00
  -> b5 7b 00
  -> b6 7b 00
  -> b7 7b 00
  -> b8 7b 00
  -> b9 7b 00
  -> ba 7b 00
  -> bb 7b 00
  -> bc 7b 00
  -> bd 7b 00
  -> be 7b 00
  -> bf 7b 00

This information is, somehow, hepful?

tttapa, 2020-09-30:
`Bx 7B 00` is “all notes off” (for channel `x`), it's not related to the stop button itself.

tttapa, 2020-09-30:
> Thank you for your reply! I used the USBMIDI_Interface and this is the result:
-> Note On Channel: 1 Data 1: 0x5d Data 2: 0x7f Cable: 1
-> Note Off Channel: 1 Data 1: 0x5d Data 2: 0x7f Cable: 1
>
> I don't know if this is what is suposed to be...

Yes, that looks alright: note on when pressed, note off when released.

helderazinheirinha, 2020-09-30:
> `Bx 7B 00` is “all notes off” (for channel `x`), it's not related to the stop button itself.
Can I send that message with your library?

tttapa, 2020-09-30:
Sure, but I don't think it's going to help you:

```cpp
Control_Surface.sendCC({MIDI_CC::All_Notes_Off, CHANNEL_1}, 0x00);
```

helderazinheirinha, 2020-09-30:
Yes it didn't work...

Thank you for all your replies!! I'll keep trying to get it to work :)