tttapa/Control-Surface #278: Logic Pro X - issues

pajczur, 2020-09-08:
Hello,
I am so sorry for littering that forum, but I need to ask for help.

I tried to test RotaryEncoder.ino and also AbsoluteRotaryEncoder.ino
And they don't work for me in Logic Pro X.

I have no idea how to configure it, but Logic Pro X doesn't see any move of my encoder.

I suppose there should be some special settings here:
```
CCRotaryEncoder enc = {
  {2, 3},       // pins
  MCU::V_POT_1, // MIDI address (CC number + optional channel)
  1,            // optional multiplier if the control isn't fast enough
};
```

Or here:
```
RelativeCCSender::setMode(relativeCCmode::MACKIE_CONTROL_RELATIVE);
```

But have no idea what to chose.

Or maybe I do smething wrong in Logic Pro. I try to open Control Assignment and use option "Learn". But as I told before, Logic Pro doesn't recognise my encoder when I move it.

I am sure my encoder circuit on Arduino UNO is ok. Because before I was using other library for MIDI. And it was working fine.
But I needed to use also some strange application "Hairless-midiserial" to transpose data from my encoder to midi protocol.

Now with Control-Surface library I also tried Hairless-midiserial, and it shows some input messages, but it says "Error: got unexpected data byte 0x0."

For any help great thanks in advance.
Best Regards

**Post your code**
Please post the full code required to answer your question. Post the code even if it doesn't work. In that case, also include a description of what you want the code it to do.

```cpp
// Your code here
```

tttapa, 2020-09-08:
If you want to use Hairless MIDI, you have to use the `HairlessMIDI_Interface` instead of the `USBMIDI_Interface` in your sketch. Make sure the baud rate of Hairless is set to `115200`. Please see the [MIDI over USB](https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/d4a/md_pages_MIDI-over-USB.html) page.

pajczur, 2020-09-08:
Hello, great thanks for your answer.
But the point is that I DON’T want to use Hairless.

But I am totally greenhorn in Arduino and in Midi protocol. That’s why I was testing everything what came to my head. It did’t work without hairless that’s only reason why I try with hairless. 

But the poin is, how to modify default code in RotaryEncoder.ino and/or how to prepare Logic Pro to cooperate with my encoder. 

I know it’s probably not question for that forum. But I am just desperate. I’ve spent about 5 hours today to figure it out how to make it work, and with no result. That’s why I ask for help. 

Great thanks in advance.

pajczur, 2020-09-08:
I don’t even understand what does mean that comment:
`MCU::V_POT_1, // MIDI address (CC number + optional channel) `

If my CC number is 2 and my channel number is 1, then should I pass to function 3? 

It seems to have no sense. Or am I wrong?

tttapa, 2020-09-08:
> But the point is that I DON’T want to use Hairless.

The Arduino UNO doesn't support MIDI over USB, so you have to use Hairless. The alternative is flashing custom firmware to the ATmega16U2, but I wouldn't recommend it. The MIDI over USB page I linked to earlier has more information.

> I don’t even understand what does mean that comment:
`MCU::V_POT_1, // MIDI address (CC number + optional channel)`

`MCU::V_POT_1` is a constant defined by the library. It's the controller number of the first "V-Pot" in the Mackie Control Universal protocol.

> If my CC number is 2 and my channel number is 1, then should I pass to function 3?

This is explained on the [MIDI Addresses page](https://tttapa.github.io/Control-Surface-doc/Doxygen/d6/dc6/md_pages_Basics.html):

```cpp
CCRotaryEncoder enc = {
  {2, 3},       // pins
  {2, CHANNEL_1}, // MIDI address (CC number + optional channel)
  1,            // optional multiplier if the control isn't fast enough
};
```

pajczur, 2020-09-09:
Great thanks for your support. So I will try with hairless, and in the future maybe I will buy SparkFun ESP32. It looks like it supports bluetooth Midi which I like. Thanks a lot for all your explanation.