tttapa/Control-Surface #279: Como soluciono el error

salmantino, 2020-09-10:
Hola, he realizado la compilación de uno de los ejemplos de la librería, y me da el siguiente error

Arduino:1.8.13 (Windows 7), Tarjeta:"Arduino Micro"

In file included from f:\Documentos WINDOWS7\Mis documentos\Arduino\libraries\Control-Surface-master\src\MIDI_Interfaces\USBMIDI\USBMIDI.cpp:32:0:

f:\Documentos WINDOWS7\Mis documentos\Arduino\libraries\Control-Surface-master\src\MIDI_Interfaces\USBMIDI\USBMIDI_MIDIUSB.ipp:2:10: fatal error: MIDIUSB.h: No such file or directory

 #include <MIDIUSB.h>

          ^~~~~~~~~~~

compilation terminated.

exit status 1

Error compilando para la tarjeta Arduino Micro.

Existe alguna solución?

tttapa, 2020-09-10:
You forgot to install the MIDIUSB library, please see the [installation instructions](https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/da8/md_pages_Installation.html).

salmantino, 2020-09-10:
Thank you very much for your reply, I assumed it and as soon as I installed it it worked.

closed 2020-09-10 by salmantino