tttapa/Control-Surface #280: Stanton C.500 control unit

yo2ldk, 2020-09-14:
Hi,
 I'm novice in MIDI things, and I see you have a lot of experience in this field
 for my project, I need from a MIDI device, to decode functions of some buttons and rotary encoders and send signals to PC,
 like a key or combination key pressed on a keyboard. Not sure if your library can help me with that..
 I have now a control unit from Stanton C.500, it can be perfect for me if it work as I need
but this is just a controller, no interface with PC, but maybe informations send it to CD unit is in MIDI language, not find any information about that , maybe you know better
 Thank you in advance for your time !

tttapa, 2020-09-14:
I'm not sure what you're asking:
- If you want to press keys on a (computer) keyboard, you'll need the Keyboard library. Control Surface only sends MIDI.
- The [manual](http://www.stantondj.com/pdf/products/cd-players/c500_manual.pdf) for the Stanton C.500 doesn't seem to mention MIDI anywhere, and there's no control output or input, except for fader start inputs. The link between the main unit and the control unit probably uses a proprietary protocol. If you can't find any information on the internet, you'll have to reverse-engineer it yourself.

yo2ldk, 2020-09-14:
Thank you for your time, nice reply and info,
 you have right, that it is; I see now is cannot be a MIDI communications between controller and unit
my best regards !

closed 2020-09-14 by yo2ldk