tttapa/Control-Surface #281: Pantalla LCD 16x2 en lugar de una oled, en Mackie Control.

alomdj00, 2020-09-21:
Hola Peter.

Tengo un par de pantallas LCD 16x2 y me gustaria darles uso, lamentablemente no se si sean compatibles con tu libreria, ya que solo hay ejemplos para las oled.  Se podran utilizar estas pantallas ya que en la Mackie original hacen uso de este estilo de pantallas pero de 55x2 si no conte mal. Podrias orientarme un poco, si tienes un ejemplo me serviria bastante. 

Muchas gracias.

tttapa, 2020-09-21:
Control Surface itself doesn't include any code to drive 16x2 displays specifically, but you can easily drive it yourself, using data from the Control Surface MIDI Input Elements.

alomdj00, 2020-09-21:
Es 100% funcional el uso de pantallas,  es decir,  se muestran tal cual en la mackie original. O hay alhunas limitaciones en la escucha de mensajes o caracteres para mostrar en cualquier pantalla. Muchas gracias

tttapa, 2020-09-21:
The [LCD](https://tttapa.github.io/Control-Surface-doc/Doxygen/d5/d2e/classMCU_1_1LCD.html) class keeps track of the text of the main LCD of the Mackie. You also have the [AssignmentDisplay](https://tttapa.github.io/Control-Surface-doc/Doxygen/d1/d08/classMCU_1_1AssignmentDisplay.html) and [TimeDisplay](https://tttapa.github.io/Control-Surface-doc/Doxygen/df/d69/classMCU_1_1TimeDisplay.html) that receive the contents of the 7-segment displays.

alomdj00, 2020-09-26:
Hola, Peter.
Estoy probando la escucha de la pantalla lcd principal de mackie y me he tropezado con un inconveniente, y es el no poder mostrar en la lcd solo lo referente al primer canal, me muestra los 8 canales, he tratado de una y otra forma pero no he logrado dividir los canales, estos me aparecen en una sola pantalla. Hay manera de hacerlo o es un inconveniente de mackie.
Nuevamente muchas gracias.

tttapa, 2020-09-26:
The MCU protocol uses one large text buffer for the display. The `LCD::getText()` method returns a pointer to the beginning of this buffer. You can then extract a single "track" from this buffer like this:

```cpp
const char *text = lcd.getText() + 7 * track + 56 * line;
char buffer[8] = "";
strncpy(buffer, text, 7);
// print buffer to the display
```
Where `track` is a number in [0, 7] and `line` is a number in [0, 1].

alomdj00, 2020-10-04:
Hola Peter, luego de estar intentando he logrado separar los tracks, pero me he encontradp con un pequeño problema, el lcddisplay hace llamado en cada bucle por lo que se lleva imprime e imprime el lcd mackie, me gustaria saber si hay alguna forma de comparar lo que se obtiene constantemente con valores anteriores y si cambia que imprima en la pantalla el nuevo mensaje de lo contrario no haga nada. Sera esto posible. Muchas gracias

tttapa, 2020-10-04:
This is possible using the `new-input` branch, it's not yet merged into `master`. Using the new branch, all MIDI input elements have a pair of `getDirty()` and `clearDirty()` methods. The `dirty` flag is set whenever a MIDI message is received that changed the value.

See https://tttapa.github.io/Control-Surface-doc/new-input/Doxygen/d5/d2e/classMCU_1_1LCD.html#a67459e7b1758df918954df57cfec7384

To install this new branch, it's easiest to use git:
```sh
git clone https://github.com/tttapa/Control-Surface --branch new-input --recursive
```

Vahen1981, 2021-03-24:
@alomdj00  I'm trying to do the same, but is really difficult to me, could you please share your code for the lcd screen? It could be very useful for me... Thanks!