tttapa/Control-Surface #282: LED Toggle Via CC

DylanBrianBenton, 2020-09-21:
Sorry If I'm completely blind, but is there a class to toggle the state of a LED by received CC messages?

Example, My teensy receives either of these 2 CC's back from my software and the below happens

90 20 7f  - Turns LED on
90 20 00 - Turns LED off

Or is this a DIY job?

tttapa, 2020-09-21:
You can find all MIDI Input Elements by going to the [Modules](https://tttapa.github.io/Control-Surface-doc/Doxygen/modules.html) page, and then [MIDI Input Elements](https://tttapa.github.io/Control-Surface-doc/Doxygen/df/d8b/group__MIDIInputElements.html). You'll find the [CCValueLED](https://tttapa.github.io/Control-Surface-doc/Doxygen/da/d34/classCCValueLED.html) class in the [MIDI Input Elements LEDs](https://tttapa.github.io/Control-Surface-doc/Doxygen/d7/de4/group__midi-input-elements-leds.html) module.

DylanBrianBenton, 2020-09-21:
Hey mate, i did see that but i thought that was only held on/high while the message was present?

tttapa, 2020-09-21:
What do you mean "while the message is present"? Messages are instantaneous: when a message is received, it triggers an action, such as turning on/off an LED or updating a value in memory. The message itself doesn't live very long, it's processed in less than a millisecond.

DylanBrianBenton, 2020-09-21:
> 
> 
> What do you mean "while the message is present"? Messages are instantaneous: when a message is received, it triggers an action, such as turning on/off an LED or updating a value in memory. The message itself doesn't live very long, it's processed in less than a millisecond.

i must be missing a parameter. im using the below,where would i add an address to turn it off? Sorry if its a silly question

`CCValueLED track = {REC_LED,{0,CHANNEL_2}};`

tttapa, 2020-09-21:
There is no address to turn it off. Whether the LED is on or off depends on the value of the Control Change message.

E.g. the message `B4 20 7F` consists of the following parts:
- `B` header for MIDI Control Change
- `4` MIDI Channel 5
- `20` Controller number 0x20
- `7F` Controller value 0x7F

tttapa, 2020-09-21:
> 90 20 7f - Turns LED on
> 90 20 00 - Turns LED off

These are not Control Change messages. These are Note On messages. If you want an LED that reacts to Note messages, you need `NoteValueLED`, not `CCValueLED`. In your case, you'd use

```cpp
NoteValueLED myLed = {
  13, // LED pin
  0x20, // note number
}; 
```

DylanBrianBenton, 2020-09-22:
Sorry, No it was a copy pasted example so thats incorrect. It's definitely a Midi CC. 

So just to make it clear (using your cc example) B4 20 7F received by CCValueLED will toggle it from one state to another?

My confusion might have stemmed as im using the CSI interface for translation between reaper and my teensy. The feedback in that software has an on/off state

`FB_TwoState  B4 20 7F B4 20 00`

When B4 20 7F is send it turns the LED on and when B4 20 00 is received it turns it off. Does that make more sense?

[This might explain it better than I can :) ](https://github.com/malcolmgroves/reaper_csi/wiki/Feedback-Processors)

tttapa, 2020-09-22:
> So just to make it clear (using your cc example) B4 20 7F received by CCValueLED will toggle it from one state to another?

It doesn't "toggle" the LED, that would be impossible to keep in sync. A value of 7F turns the LED on.

> FB_TwoState B4 20 7F B4 20 00
>
> When B4 20 7F is send it turns the LED on and when B4 20 00 is received it turns it off. Does that make more sense?

Yes, that's what `CCValueLED` does.

DylanBrianBenton, 2020-09-23:
Hey Mate, 

Sorry about that. It was due to me not understanding the MIDI address values correctly.

closed 2020-09-23 by DylanBrianBenton