tttapa/Control-Surface #284: One RGB led VU. Different intensity for each segment.

MarcosGogo, 2020-09-25:
Hello.

First of all, sorry for any errors in English, my native language is Portuguese.

I really wanted to congratulate the creator of this incredible library. The Library is very incredible. Keep it up.

This is my first post on github. I am very amateur and I am still starting in the world of programming.

I am wanting to understand where I can start, to implement two things with this library. And to know if it's possible to do what I'm thinking.

1) I would like to direct the 12 segments of the Mackie VU to only one RGB LED, with differences in intensity and color for each of the 12 segment values ​​of the Mackie protocol. Below is a gif with a demonstration of the behavior I hope to achieve.

I would like to know where to start developing this idea. I don't know if I can have a brightness and color value for each segment of the Mackie Vu? I imagine that it may be possible to achieve this by configuring something by PWM pins or by intensity within the HSV color control. But I still don't know where to start.

[image]

2) This same RGB led that would read the VU signal from DAW, I wish it could be used for SOLO and MUTE. In this case when the channel is on solo or mute, the behavior of VU disappears to enter another color for SOLO and another for MUTE, while mute and solo are active. After the channel is no longer in SOLO or MUTE, the VU is active again.

All of these programming will be used with parameters of the MCU: MACKIE protocol.

I hope I haven't been too confused in my explanation. Any initial help is going to be really cool.

Thanks for all and continue with this incredible project.

tttapa, 2020-09-27:
With the latest stable release, it's probably best to parse the MIDI data yourself, see https://tttapa.github.io/Control-Surface-doc/Doxygen/d1/d5d/MIDI-Input_8ino-example.html.

I've recently started reworking the MIDI input system to allow for things like this. The changes are on the `new-input` branch. With that version of the library, you can do something like this:

```cpp
#include <Control_Surface.h>

USBMIDI_Interface midi;

struct MuteSoloVU_RGB : Updatable<> {
  MuteSoloVU_RGB(uint8_t track, 
                 PinList<3> rgbPins,
                 unsigned int decayTime = MCU::VUDecay::Default)
    : mute(MCU::MUTE_1 + track - 1),
      solo(MCU::SOLO_1 + track - 1), 
      vu(track, decayTime), 
      rgbPins(rgbPins) {}

  // Called periodically
  void update() override {
    // If any of the listeners received a new MIDI message:
    if (mute.getDirty() || solo.getDirty() || vu.getDirty()) {
      updateLEDs();
      mute.clearDirty(), solo.clearDirty(), vu.clearDirty();
    }
  }

  // Called once upon initialization
  void begin() override {
    /* nothing */
  }

  void updateLEDs() {
    if (mute.getValue())
      showLEDs(0x00, 0x00, 0xFF);
    else if (solo.getValue())
      showLEDs(0xFF, 0x00, 0xFF);
    else
      showLEDs(0x00, VU_PWM_lut[vu.getValue()], 0x00);
  }

  void showLEDs(uint8_t r, uint8_t g, uint8_t b) {
    analogWrite(rgbPins[0], r);
    analogWrite(rgbPins[1], g);
    analogWrite(rgbPins[2], b);
  }

  NoteValue mute, solo;
  MCU::VU vu;

  PinList<3> rgbPins;

  static const uint8_t VU_PWM_lut[13];
};

const uint8_t MuteSoloVU_RGB::VU_PWM_lut[] = {0,  1,  3,  5,   8,   14,  21,
                                              33, 50, 75, 113, 170, 255};

MuteSoloVU_RGB led = {
  1,                    // Track
  {4, 5, 6},            // RGB LED pins
  MCU::VUDecay::Default // VU decay time
};

void setup() {
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
}
```

You'll have to clone the library using git:
```sh
git clone https://github.com/tttapa/Control-Surface --branch new-input --recursive
```
The VU meter only shows different intensities of green, not orange or red, but that shouldn't be too hard to change yourself.

MarcosGogo, 2020-10-01:
Hi Pieter,

I really have no words to thank you for that quick answer, and for this amazing code. You really rocks, thanks so much Mr. Robot.

I am very happy to have been able to code, on my own, the colors orange and red for the last segments of the VU. It was a great step up for me, taking my condition as a true beginner.

The code its almost perfect, I just need to adjust some strange behavior i notice, and clarify some doubts that I still have.

Follow the working code below with the updates I made to obtain the colors orange and red, in addition to the green, which was already coded.

```cpp
//
#include <Control_Surface.h>

USBMIDI_Interface midi;

struct MuteSoloVU_RGB : Updatable<> {
  MuteSoloVU_RGB(uint8_t track, 
                 PinList<3> rgbPins,
                 unsigned int decayTime = MCU::VUDecay::Default)
    : mute(MCU::MUTE_1 + track - 1),
      solo(MCU::SOLO_1 + track - 1), 
      vu(track, decayTime), 
      rgbPins(rgbPins) {}

  // Called periodically
  void update() override {
    // If any of the listeners received a new MIDI message:
    if (mute.getDirty() || solo.getDirty() || vu.getDirty()) {
      updateLEDs();
      mute.clearDirty(), solo.clearDirty(), vu.clearDirty();
    }
  }

  // Called once upon initialization
  void begin() override {
    /* nothing */
  }

  void updateLEDs() {
    if (mute.getValue())
      showLEDs(0x00, 0x00, 0xFF);
    else if (solo.getValue())
      showLEDs(0xFF, 0xFF, 0x00);
    else
      showLEDs(VU_PWM_lut_RED [vu.getValue()], VU_PWM_lut_GREEN [vu.getValue()], 0x00); 
  }

  void showLEDs(uint8_t r, uint8_t g, uint8_t b) {
    analogWrite(rgbPins[0], r);
    analogWrite(rgbPins[1], g);
    analogWrite(rgbPins[2], b);
  }

  NoteValue mute, solo;
  MCU::VU vu;

  PinList<3> rgbPins;

  static const uint8_t VU_PWM_lut_RED [12];
  static const uint8_t VU_PWM_lut_GREEN [12];
  static const uint8_t VU_PWM_lut_BLUE [12];
};

const uint8_t MuteSoloVU_RGB::VU_PWM_lut_RED [] = {0, 0,  0,  0,   0,   0,  0,
                                              0, 0, 0, 255, 255}; 
const uint8_t MuteSoloVU_RGB::VU_PWM_lut_GREEN [] = {0,  1,  3,  5,   8,   14,  50,
                                              75, 150, 200, 66, 0};
const uint8_t MuteSoloVU_RGB::VU_PWM_lut_BLUE [] = {0, 0,  0,  0,  0,  0,  0,
                                              0, 0, 0, 0, 0}; 
[... 90 lines cut ...]
  // Called once upon initialization
  void begin() override {
    /* nothing */
  }

  void updateLEDs() {
    if (rec.getValue())
      showLEDs(VU_PWM_lut_RED [vu.getValue()], VU_PWM_lut_GREEN [vu.getValue()], 0x00);
    else if (mute.getValue())
      showLEDs(0x00, 0x00, 0xFF);
    else if (solo.getValue())
      showLEDs(0xFF, 0xFF, 0x00);
    else
      showLEDs(0x00, 0x00, 0x00);  // Or another code that disable the led.
  }
```
I ended up having errors in setting the REC to "REC_RDY_1". But I don't even know if this logic is correct to maintain the perfect hierarchy that was before with the addition of working only if REC_RDY is true.

Sorry for the giant message and the lack of clarity. Too many non-native languages for me.
I hope I haven't confused it yet. I promise to improve my programming syntax to communicate better.

Thank you so much for everything. This library was the great gift and good news of 2020.

Best,
Marcos

tttapa, 2020-10-01:
1. The maximum value of the VU meter is 12, so it has 13 states (you have to count 0). If you only have 12 elements in the lookup tables you'll go out of bounds.

2. The decay rate used by the MCU protocol is 1.8s for all 12 segments, which is 150 ms per segment. This is rather slow, and not all DAWs support this, some expect the controller to not decay at all. 

The flickering you're seeing could be caused by the DAW expecting the VU meters to hold instead of decaying automatically. You could try using [`MCU::VUDecay::Hold`](https://tttapa.github.io/Control-Surface-doc/new-input/Doxygen/d6/d98/namespaceMCU_1_1VUDecay.html#ac4cd1576514f126eb68fd54327c2a4da) for the decay time. You can also use any integer value in milliseconds instead of these constants. It's the decay time for a single level.

I've never noticed the flickering, I'll try again later today and report back.

The MCU VU meter also has an "overload" flag, you can use [`vu.getOverload()`](https://tttapa.github.io/Control-Surface-doc/new-input/Doxygen/d8/d4c/classMCU_1_1VU.html#ac7697fc99dabb393feae0d4e6bbbf923) in your code check it. Some DAWs don't seem to set this flag, though.

 Finally, the code you posted is incomplete. Did you add a `rec` member variable? You have to check if it's dirty or not, and also clear the dirty flag afterwards, like with the other inputs (`mute`, `solo`, `vu`).  
If you want to disable all LEDs if Rec/Rdy is off, you could use
```cpp
  void updateLEDs() {
    if (!rec.getValue())
      showLEDs(0x00, 0x00, 0x00);
    else if (mute.getValue())
      showLEDs(0x00, 0x00, 0xFF);
    else if (solo.getValue())
      showLEDs(0xFF, 0x00, 0xFF);
    else
      showLEDs(0x00, VU_PWM_lut[vu.getValue()], 0x00);
  }
```

MarcosGogo, 2020-10-01:
Hi Pieter,

Thank you very much for responding so quickly.

Thanks for the explanation about the number of states in Vu. It's more clear to me now.

Good to know about `vu.getOverload()` flag. I will try to implement this in some way. But only after solving the other problems first. 

The good news is that I think I was able to implement Rec_Rdy's behavior. Maybe it's not perfect yet, I still don't know if it's necessary to call him everywhere I put it in the code. But either way, it seems to be working.

About Decay Time. It seems that each DAW sends its own behavior. I think this has to do with whether DAW sends the update of the VU segments via PEAK VU or directly via the VU. On the Ableton and Reaper, it appears that the VU follows the dynamics of the PEAK VU, which has a slow decay time. In Ardor, it appears that the VU is updated by the normal VU and not by the PEAK VU, even though the Ardor visually has the PEAK VU with its decay time slower than the VU itself. I would like to obtain this faster and more precise behavior of Ardor, because this way the LED Vu becomes more responsive, generating a more accurate sensation of the current intensity of the input signal. But it seems that this is a setting that can only be done within the DAW when it is possible to change the behavior of the VU.

The real problem that remains is with the flick, which is still a little strange.
When I tested `MCU :: VUDecay :: Hold` before implementing `REC_RDY`. Everything worked, the flick disappeared. And the Mute, Solo and VU leds were updating correctly.

Unfortunately, when I implemented Rec_Ready's behavior, the flick also disappeared, but it started to generate some updating problems between the functions. Usually saving an LED value when connecting and enabling some parameters, making the LED VU behavior no longer work, after some changes.

On the other hand, I did another test with the REC_RDY implemented, but going backwards with `MCU :: VUDecay :: Hold `to `MCU :: VUDecay :: Default`. With this code, which I will paste below, everything worked, except for the flick problem that came back. But the LED updates were correct.

One thing I found strange, is that in the `MCU :: VUDecay :: `function, the 'Hold' parameter is highlighted in blue, and the "Default" parameter is not highlighted. I believe that this must be normal, and it must be something of the language syntax that I don't understand yet.

What I would really like to have is this code below with 'MCU :: VUDecay :: Default' without the flick.

```cpp
#include <Control_Surface.h>

USBMIDI_Interface midi;

struct MuteSoloVU_RGB : Updatable<> {
  MuteSoloVU_RGB(uint8_t track, 
                 PinList<3> rgbPins,
                 unsigned int decayTime = MCU::VUDecay::Default)
    : rec(MCU::REC_RDY_1 + track - 1),
      mute(MCU::MUTE_1 + track - 1),
      solo(MCU::SOLO_1 + track - 1), 
      vu(track, decayTime), 
      rgbPins(rgbPins) {}

  // Called periodically
  void update() override {
    // If any of the listeners received a new MIDI message:
    if ( rec.getDirty() || mute.getDirty() || solo.getDirty() || vu.getDirty()) {
      updateLEDs();
       rec.clearDirty(), mute.clearDirty(), solo.clearDirty(), vu.clearDirty();
    }
  }

  // Called once upon initialization
  void begin() override {
    /* nothing */
  }

void updateLEDs() {
    if (!rec.getValue())
      showLEDs(0x00, 0x00, 0x00);
    else if (mute.getValue())
      showLEDs(0x00, 0x00, 0x80);
    else if (solo.getValue())
      showLEDs(0x90, 0x90, 0x00);
    else
      showLEDs(VU_PWM_lut_RED [vu.getValue()], VU_PWM_lut_GREEN [vu.getValue()], VU_PWM_lut_BLUE [vu.getValue()]);
  }

  void showLEDs(uint8_t r, uint8_t g, uint8_t b) {
    analogWrite(rgbPins[0], r);
    analogWrite(rgbPins[1], g);
    analogWrite(rgbPins[2], b);
  }

  NoteValue rec, mute, solo;
  MCU::VU vu;

  PinList<3> rgbPins;

  static const uint8_t VU_PWM_lut_RED [13];
  static const uint8_t VU_PWM_lut_GREEN [13];
  static const uint8_t VU_PWM_lut_BLUE [13];
  
};

const uint8_t MuteSoloVU_RGB::VU_PWM_lut_RED [] = {1, 0,  0,  0,  0,   0,   0,  0,
                                              0, 0, 0, 255, 180}; 
const uint8_t MuteSoloVU_RGB::VU_PWM_lut_GREEN [] = {0, 1, 3,  4,  8,   14,   50,  75,
                                              100, 150, 200, 66, 0};
const uint8_t MuteSoloVU_RGB::VU_PWM_lut_BLUE [] = {1,  0,  0,  0,  0,  0,
                                              0, 0, 0, 0, 0, 0, 0};                                                                                            
                                            

MuteSoloVU_RGB led [] = {
  {1,                    // Track
  {4, 9, 10},            // RGB LED pins
  MCU::VUDecay::Default}, // VU decay time
  {2,                    // Track
  {12, 14, 15},            // RGB LED pins
  MCU::VUDecay::Default},
};

void setup() {
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
}
```

I still don't know how to find out and make sure the REC should be dirty or not.

Thank you very much for everything Pieter. You are helping me a lot.

MarcosGogo, 2020-10-03:
**UPDATE**

Hi Pieter,

I've been rethinking here, and I'm going to abandon this idea of ​​REC_RDY's behavior. So the implementation will be simpler.

I want to make it work with `MCU::VUDecay::Default`, and solve the Flick problem.

I strongly believe that if I can implement something like `dotMode ()` that exists in the `VULEDs class`, maybe it would solve this problem, since apparently my problem with flick seems to be an interleaving of the previous state with the current state of VU segments. So if the VU is in `dotMode ()` instead of `barMode ()`, the previous State will not exist, and the flick will probably disappear.
I'm just not sure if it can implement something like `dotMode ()` in this code with just one RGB LED.

You can disregard the code from the previous post, I will remove the behavior of `REC_RDY`. If you can help me to implement, if possible, `dotMode ()` in this new code, I can test here to see if it works. And so I don't need to use `MCU::VUDecay::Hold`, which has some problems here. The problem I had with `MCU::VUDecay::Hold`, was that it works well only while the VU is active, receiving some signal. When I press stop on the DAW or the VU signal does not receive any signal, the LED has the last state frozen and does not return to the initial state. I believe that this is the expected behavior of `MCU::VUDecay::Hold`, and that is why I would like to implement it with the `MCU::VUDecay::Default`.

Thank you so much for everything.
Below is the code I intend to use, lacking only the implementation of `dotMode ()`

```cpp
#include <Control_Surface.h>

USBMIDI_Interface midi;

struct MuteSoloVU_RGB : Updatable<> {
  MuteSoloVU_RGB(uint8_t track, 
                 PinList<3> rgbPins,
                 unsigned int decayTime = MCU::VUDecay::Default)
    : mute(MCU::MUTE_1 + track - 1),
      solo(MCU::SOLO_1 + track - 1), 
      vu(track, decayTime), 
      rgbPins(rgbPins) {}

  // Called periodically
  void update() override {
    // If any of the listeners received a new MIDI message:
    if ( mute.getDirty() || solo.getDirty() || vu.getDirty()) {
      updateLEDs();
       mute.clearDirty(), solo.clearDirty(), vu.clearDirty();
    }
  }

  // Called once upon initialization
  void begin() override {
    /* nothing */
  }

void updateLEDs() {
    if (mute.getValue())
      showLEDs(0x00, 0x00, 0x80);
    else if (solo.getValue())
      showLEDs(0x90, 0x90, 0x00);
    else
      showLEDs(VU_PWM_lut_RED [vu.getValue()], VU_PWM_lut_GREEN [vu.getValue()], VU_PWM_lut_BLUE [vu.getValue()]);
  }

  void showLEDs(uint8_t r, uint8_t g, uint8_t b) {
    analogWrite(rgbPins[0], r);
    analogWrite(rgbPins[1], g);
    analogWrite(rgbPins[2], b);
  }

  NoteValue mute, solo;
  MCU::VU vu;

  PinList<3> rgbPins;

  static const uint8_t VU_PWM_lut_RED [13];
  static const uint8_t VU_PWM_lut_GREEN [13];
  static const uint8_t VU_PWM_lut_BLUE [13];
  
};

const uint8_t MuteSoloVU_RGB::VU_PWM_lut_RED [] = {1, 0,  0,  0,  0,   0,   0,  0,
                                              0, 0, 0, 255, 180}; 
const uint8_t MuteSoloVU_RGB::VU_PWM_lut_GREEN [] = {0, 1, 3,  4,  8,   14,   50,  75,
                                              100, 150, 200, 66, 0};
const uint8_t MuteSoloVU_RGB::VU_PWM_lut_BLUE [] = {1,  0,  0,  0,  0,  0,
                                              0, 0, 0, 0, 0, 0, 0};                                                                                            
                                            

MuteSoloVU_RGB led [] = {
  {1,                    // Track
  {4, 9, 10},            // RGB LED pins
  MCU::VUDecay::Default}, // VU decay time
  {2,                    // Track
  {12, 14, 15},            // RGB LED pins
  MCU::VUDecay::Default},
};

void setup() {
  Control_Surface.begin();
}

void loop() {
  Control_Surface.loop();
}
```

tttapa, 2020-10-03:
> One thing I found strange, is that in the MCU :: VUDecay :: function, the 'Hold' parameter is highlighted in blue, and the "Default" parameter is not highlighted. I believe that this must be normal, and it must be something of the language syntax that I don't understand yet.

The syntax highlighting of the Arduino IDE is really primitive, it doesn't analyze your actual code, it just recognizes some words, regardless of how they're used. Most of the time, you should completely ignore the colors in the IDE, it has nothing to do with how the compiler actually sees your code.

I'm unable to reproduce the flickering problem with the Tracktion DAW I'm using. The only solution I can think of is to use a longer decay time (e.g. 300 instead of `MCU::VUDecay::Default`). 

To explain this more clearly, I think this is what's going on:  
The DAW sends its meter values every 300ms, for example, and the decay time is
150ms. One division on the time axis below is 150ms.
Then the flickering looks something like this:

```
level:
       3     ┌───────┐       ┌───────┐
       2     │       └───────┘       └───────┐
       1     │                               └───────┐
       0   ──┘                                       └───────
time:      ──┴───────┴───────┴───────┴───────┴───────┴───────
            (1)     (2)     (3)     (4)     (5)     (6)
```
1. DAW: Set level 3
2. Decay
3. DAW: Maintain level 3
4. Decay
5. Decay
6. Decay

The DAW doesn't expect the Arduino to decay at (2), but it does, so it decays
to the level below it, then the DAW sends its next update at (3) and it goes back to the correct level.
This repeats, resulting in a flicker.

> I still don't know how to find out and make sure the REC should be dirty or not.

I'm not sure what you mean here. The `NoteValue` and `VU` classes receive values over MIDI and save them for you. "Dirty" means that they received a new value, so the stored value has changed, and you, the user of that value should read the value, and take some action, like updating an LED. To notify the `NoteValue` and `VU` classes that you have read this new value, you clear the "dirty" flag again.

Dot mode and bar mode are just display modes, they don't save any states. The mode only determines whether all LEDs before the active LED are turned on (bar mode) or off (dot mode). This is the code used by the `VULEDs` class (indirectly):

https://github.com/tttapa/Control-Surface/blob/0395a0a99e333b7aaa35a7edcf8630c3a0da35a7/src/AH/Hardware/LEDs/DotBarDisplayLEDs.hpp#L40-L47

`displayRange(0, value)` turns on all LEDs from the first one up to the LED with index `value`.  
`displayDot(value-1)` turns on the LED with index `value-1`. 

For example, if there are 12 LEDs, and `value = 6`, `displayRange(0, 6)` will turn on the first six LEDs: `◉◉◉◉◉◉◎◎◎◎◎◎`  
and `displayDot(5)` turns on the sixth LED: `◎◎◎◎◎◉◎◎◎◎◎◎`   
(LED indices are zero-based)

Jhenesson, 2022-11-14:
I've been waiting for years for a solution like this (considering my low level of programming knowledge)
All I need to finish my project is to use a simple LED to match the VU of the respective channel with the lighting intensity. I'm trying to adapt this code to the solution.