tttapa/Control-Surface #286: ssd1306 and teensy 4.0

benwadub, 2020-09-27:
hi, again me, I d like to add the ssd 1306 display I just received to my controller but can t find how to adapt the exemple code to my teensy 4.0, no problem to wire and and vcc but how to wire the scl sda  res and dc pins of the screen and how to write it in the code, I can t find in your exemple code the screen pins, I just see dc pin reset and cs that I don t have on my screen
```cpp
/**
 /**
 * An example demonstrating the use of DisplayElement%s to display information
 * from the DAW on a small OLED display.
 *
 * @boards  Teensy 3.x
 * 
 * Connections
 * -----------
 * 
 * - 5:  Push button (to ground)
 * - 6:  Push button (to ground)
 * - 7:  OLED Data/D1 (SPI MOSI)
 * - 13: OLED Clock/D0 (SPI SCK)
 * - 17: OLED Data/Command
 * - 10: OLED Cable Select
 * 
 * Add a capacitor between the reset pin of the display and ground, and a 
 * resistor from reset to 3.3V. The values are not critical, 0.1µF and 10kΩ 
 * work fine.  
 * You do need some way to reset the display, without it, it won't work.  
 * Alternatively, you could use an IO pin from the Teensy to reset the 
 * display, but this just "wastes" a pin.
 * 
 * Behavior
 * --------
 * 
 * - The time (bars, beats, fraction), play and record status are shown at the 
 *   top of the display.
 * - For each of the 8 first tracks, a VU level meter with peak indicator and
 *   a V-Pot ring showing the pan are displayed, as well as the the mute, solo 
 *   and record arm status.
 * - Two tracks are displayed at once. By pressing the push buttons connected
 *   to pins 5 and 6, you can cycle through four banks to display all 8 tracks.
 * 
 * Mapping
 * -------
 * 
 * Map "Control Surface" as a Mackie Control Universal unit in your DAW.
 * 
 * @note    There seem to be some differences in the way some applications 
 *          handle VU meters: some expect the hardware to decay automatically,
 *          some don't.  
 *          If you notice that the meters behave strangely, try both 
 *          MCU::VUDecay::Hold and MCU::VUDecay::Default, or try a different 
 *          decay time.
 * 
 * Written by PieterP, 2019-11-12  
 * https://github.com/tttapa/Control-Surface
 */

#include <Encoder.h> // Include the Encoder library.
// This must be done before the Control Surface library.
#include <Control_Surface.h> // Include the Control Surface library
// Include the display interface you'd like to use
#include <Display/DisplayInterfaces/DisplayInterfaceSSD1306.hpp>

// ----------------------------- MIDI Interface ----------------------------- //
// ========================================================================== //

/*
   Instantiate a MIDI interface to use for the Control Surface.
*/

USBMIDI_Interface midi;
// USBDebugMIDI_Interface midi(115200);

// ----------------------------- Display setup ------------------------------ //
// ========================================================================== //

/*
   Instantiate and initialize the SSD1306 OLED display
*/

constexpr uint8_t SCREEN_WIDTH = 128;
constexpr uint8_t SCREEN_HEIGHT = 64;

constexpr int8_t OLED_DC = 17;    // Data/Command pin of the display
constexpr int8_t OLED_reset = -1; // Use the external RC circuit for reset
constexpr int8_t OLED_CS = 10;    // Chip Select pin of the display

constexpr uint32_t SPI_Frequency = SPI_MAX_SPEED;

// Instantiate the displays
Adafruit_SSD1306 ssd1306Display = {
  SCREEN_WIDTH, SCREEN_HEIGHT, &SPI,          OLED_DC,
  OLED_reset,   OLED_CS,       SPI_Frequency,
};

// --------------------------- Display interface ---------------------------- //
// ========================================================================== //

// Implement the display interface, specifically, the begin and drawBackground
// methods.
class MySSD1306_DisplayInterface : public SSD1306_DisplayInterface {
 public:
  MySSD1306_DisplayInterface(Adafruit_SSD1306 &display)
    : SSD1306_DisplayInterface(display) {}

  void begin() override {
    // Initialize the Adafruit_SSD1306 display
    if (!disp.begin())
      FATAL_ERROR(F("SSD1306 allocation failed."), 0x1306);

    // If you override the begin method, remember to call the super class method
    SSD1306_DisplayInterface::begin();
  }

  void drawBackground() override { disp.drawLine(1, 8, 126, 8, WHITE); }

} display = ssd1306Display;

// ------------------------------- Bank setup ------------------------------- //
// ========================================================================== //

/*
   Create a bank and a bank selector to change its setting.
*/

Bank<4> bank(2); // Create a new bank with two tracks per bank

// Create a new bank selector to control the bank using two push buttons
IncrementDecrementSelector<4> bankselector = {bank, {5, 6}, Wrap::Wrap};

// -------------------------- MIDI Input Elements --------------------------- //
// ========================================================================== //

/*
   Define all elements that listen for MIDI messages.
*/

// Time display keeps track of the bar counter
MCU::TimeDisplay timedisplay = {};

// Play / Record
NoteValue play = {MCU::PLAY};
NoteValue record = {MCU::RECORD};

// Mute
Bankable::NoteValue<4> mute[] = {
  {bank, MCU::MUTE_1},
  {bank, MCU::MUTE_2},
};

// Solo
Bankable::NoteValue<4> solo[] = {
  {bank, MCU::SOLO_1},
  {bank, MCU::SOLO_2},
};

NoteValue rudeSolo = {MCU::RUDE_SOLO};

// Record arm / ready
Bankable::NoteValue<4> recrdy[] = {
  {bank, MCU::REC_RDY_1},
  {bank, MCU::REC_RDY_2},
};

// VU meters
MCU::Bankable::VU<4> vu[] = {
  {bank, 1, MCU::VUDecay::Hold},
  {bank, 2, MCU::VUDecay::Hold},
};

// VPot rings
MCU::Bankable::VPotRing<4> vpot[] = {
  {bank, 1},
  {bank, 2},
};

// ---------------------------- Display Elements ---------------------------- //
// ========================================================================== //

/*
   Define all display elements that display the state of the input elements.
*/

// Time display
MCU::TimeDisplayDisplay timedisplaydisplay = {
  // position (0, 0), font size (1)
  display, timedisplay, {0, 0}, 1, WHITE,
};

// Play / Record
NoteBitmapDisplay playDisp = {
  display, play, XBM::play_7, {16 + 64, 0}, WHITE,
};
NoteBitmapDisplay recordDisp = {
  display, record, XBM::record_7, {26 + 64, 0}, WHITE,
};

// Mute
NoteBitmapDisplay muteDisp[] = {
  {display, mute[0], XBM::mute_10B, {14, 50}, WHITE},
  {display, mute[1], XBM::mute_10B, {14 + 64, 50}, WHITE},
};

// Solo
NoteBitmapDisplay soloDisp[] = {
  {display, solo[0], XBM::solo_10B, {14, 50}, WHITE},
  {display, solo[1], XBM::solo_10B, {14 + 64, 50}, WHITE},
};

NoteBitmapDisplay rudeSoloDisp = {
  display, rudeSolo, XBM::solo_7, {36 + 64, 0}, WHITE};

// Record arm / ready
NoteBitmapDisplay recrdyDisp[] = {
  {display, recrdy[0], XBM::rec_rdy_10B, {14 + 14, 50}, WHITE},
  {display, recrdy[1], XBM::rec_rdy_10B, {14 + 14 + 64, 50}, WHITE},
};

// VU meters
MCU::VUDisplay vuDisp[] = {
  // position (32+11, 60), width (16), bar height (3) px, bar spacing (1) px
  {display, vu[0], {32 + 11, 60}, 16, 3, 1, WHITE},
  {display, vu[1], {32 + 11 + 64, 60}, 16, 3, 1, WHITE},
};

// VPot rings
MCU::VPotDisplay vpotDisp[] = {
  // position (0, 10), outer radius (16) px, inner radius (13) px
  {display, vpot[0], {0, 10}, 16, 13, WHITE},
  {display, vpot[1], {64, 10}, 16, 13, WHITE},
};

// Bank seting
BankDisplay bankDisp[] = {
  // first track of the bank (1), position (0, 50), font size (2)
  {display, bank, 1, {0, 50}, 2, WHITE},
  {display, bank, 2, {64, 50}, 2, WHITE},
};

// --------------------------------- Setup ---------------------------------- //
// ========================================================================== //

void setup() {
  // The default SPI MOSI pin (11) is used for I²S, so we need to use the
  // alternative MOSI pin (7)
  SPI.setMOSI(7);
  // Correct relative mode for MCU rotary encoders
  RelativeCCSender::setMode(MACKIE_CONTROL_RELATIVE);
  Control_Surface.begin(); // Initialize Control Surface
}

// ---------------------------------- Loop ---------------------------------- //
// ========================================================================== //

void loop() {
  Control_Surface.loop(); // Refresh all elements
}
```

tttapa, 2020-09-27:
These are questions about the Adafruit_SSD1306 library. Please see the examples that come with that library, as well as the pinout diagram of the Teensy board you're using.

benwadub, 2020-09-27:
I already tried this but as a total noob I don t understand everything don t know where to put data connection by exemple :-( and the reset

tttapa, 2020-09-27:
Did you get the examples of the Adafruit_SSD1306 library to work? If that doesn't work, you won't get this example to work either.

benwadub, 2020-09-27:
I m totally confusing between all these things!! 😅
I ll have to find clear documentation on spi protocol and find what correspond to what sck dc cs... all made a big head hake to my brain this evening 😅
I don’t know at all where my screens pins have to go on the teensy and don t understand for the moment how the code work for spi!

benwadub, 2020-09-28:
It seems that the labeling on my screen is not like the standard spi did you already saw that? I don t have pin labeled mosi or midi or dc, 
[image]

benwadub, 2020-09-28:
[image]

tttapa, 2020-09-28:
The images you posted don't work. You have to wait for them to upload before posting.

benwadub, 2020-09-28:
Sorry! 
[image]
[image]

tttapa, 2020-09-28:
I've never come across an SSD1306 display with 6 pins. The DC pin suggests that it's an SPI variant, but the lack of a CS pin and the SCL/SDA labels suggest I²C.

Since this is not strictly related to the Control Surface library, I'd suggest using the Arduino forum instead.

benwadub, 2020-09-28:
I asked on the teensy Facebook group, hope I ll have an answer!
It s a display that can be used with i2c and spi, it could be better for me to find a nirmal spi display

PARISOT Benoit
06 83 11 12 11
12 chemin de châtillon 39570 Pannessières
TRAITEUR /CHEF A DOMICILE/ FOOD TRUCK
PRODUITS LOCAUX CUISINE 100% FAITE MAISON

Facebook

Site internet

contact@leatinerant.fr

Instagram

benwadub, 2020-09-28:
ok so I got it working! with the Adafruit library now! I ll be able to start with your library!

benwadub, 2020-09-28:
ok so my display work with these pins assignment in the Adafruit exemple
```cpp
#define SCREEN_WIDTH 128 // OLED display width, in pixels
#define SCREEN_HEIGHT 64 // OLED display height, in pixels

// Declaration for SSD1306 display connected using software SPI (default case):
#define OLED_MOSI   11//sda pin on my display
#define OLED_CLK   13//scl pin on my display
#define OLED_DC    18//dc pin on my display
#define OLED_CS    10//not wired
#define OLED_RESET -1
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT,
  OLED_MOSI, OLED_CLK, OLED_DC, OLED_RESET, OLED_CS);
```
but can t make it work with your exemple: mcu OLED display I can t find where you use the clk pin in your code?

tttapa, 2020-09-28:
The code you posted uses software SPI, which is much slower than hardware SPI. Also try to avoid preprocessor macros, use constants instead.

Try this
```cpp
constexpr uint8_t SCREEN_WIDTH = 128; // OLED display width, in pixels
constexpr uint8_t SCREEN_HEIGHT = 64; // OLED display height, in pixels

// Declaration for SSD1306 display connected using software SPI (default case):
constexpr uint8_t OLED_DC = 18; // DC pin on my display
constexpr uint8_t OLED_CS = 10; // not wired
constexpr uint8_t OLED_RESET = -1;

Adafruit_SSD1306 ssd1306Display = {
  SCREEN_WIDTH, SCREEN_HEIGHT,
  &SPI,
  OLED_DC, OLED_RESET, OLED_CS,
};
```

benwadub, 2020-09-28:
I ll try tomorrow thanks!
I think i didn’t show you the finished controler here it is! Thanks again for this I m glad to achieved this with your help!
[image]

tttapa, 2020-09-28:
Looks great! Thanks for sharing.

benwadub, 2020-09-29:
hi, that doesn't work, is it normal that you don t declare the clk pin that is on pin 13 ?

benwadub, 2020-09-29:
hi, I found that spa has to be in mode 3 for ssd1306 and teensy 4.0
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
class Adafruit_SSD1309_SPI : public Adafruit_SSD1306 {
 public:
  Adafruit_SSD1309_SPI(uint8_t w, uint8_t h, SPIClass *spi,
    int8_t dc_pin, int8_t rst_pin, int8_t cs_pin, uint32_t bitrate=8000000UL) :
    Adafruit_SSD1306(w, h, spi, dc_pin, rst_pin, cs_pin, bitrate) {
#ifdef SPI_HAS_TRANSACTION
  spiSettings = SPISettings(bitrate, MSBFIRST, SPI_MODE3);
#endif

I add this lines at the top of the code like in the Adafruit exemple and that works fine now!
thanks for the help!

tttapa, 2020-09-30:
I didn't know that, the SSD1306 SPI displays I have laying around all use the default SPI mode. Are you using an SSD1309 or SSD1306?

benwadub, 2020-09-30:
Have a look at this page it s where I found the solution! https://github.com/adafruit/Adafruit_SSD1306/issues/167

benwadub, 2020-09-30:
It s a 1306 here

benwadub, 2020-10-01:
hi i m relie not enough experienced to use theses screens I got, do you have link where you buy yours to be sure they will work with the library please?

tttapa, 2020-10-03:
I got mine on eBay long ago, I doubt the same listing will still be available. Just make sure it has 7 pins: Gnd, Vcc, D0, D1, Res, DC, CS

If you made your display work with the Adafruit library, why can't you get it to work with Control Surface? It uses the same library, you just have to wrap your display in the `MySSD1306_DisplayInterface` class of the example.

```cpp
// You should already have this from the Adafruit library, it is identical.
Adafruit_SSD1309_SPI ssd1306Display( /* options */);

// This is a wrapper to pass your display to the Control Surface library
class MySSD1306_DisplayInterface : public SSD1306_DisplayInterface {
 public:
  MySSD1306_DisplayInterface(Adafruit_SSD1306 &display)
    : SSD1306_DisplayInterface(display) {}

  void begin() override {
    // Initialize the Adafruit_SSD1306 display
    if (!disp.begin())
      FATAL_ERROR(F("SSD1306 allocation failed."), 0x1306);

    // If you override the begin method, remember to call the super class method
    SSD1306_DisplayInterface::begin();
  }

  void drawBackground() override { disp.drawLine(1, 8, 126, 8, WHITE); }

} display = ssd1306Display; // Pass the Adafruit display to the wrapper.
```
You'll have to post your code and explain exactly what the problem is, otherwise I can't help you.

benwadub, 2020-10-03:
Wouah thanks, I didn’t want to be insistant with my problem that is not really related to your library! 
I forgot to save the code when that worked and can t make it working again! I ll try again with that!

benwadub, 2020-10-03:
I got it but with my code the screen is scrolling, see the vidéo any idea of what can do it?
[image]

benwadub, 2020-10-03:
```cpp
#include <Encoder.h> // Include the Encoder library.
#include "Control_Surface.h"
#include <Display/DisplayInterfaces/DisplayInterfaceSSD1306.hpp>

//#include <name.c>
// celà doit etre écrit avant la library control surface
 //USBDebugMIDI_Interface midi = 115200; // enlever les // en début de ligne pour entrer en mode debug usb et voir dans le panneau de control si vos controler envoient bien les infos

//auto &serial = Serial1;// Selectionne le port série à utiliser remplacer par serial pour une arduino
//SerialMIDI_Interface<decltype(serial)> midi = {serial, MIDI_BAUD};// démarre une interface midi serial au midi baud rate par defaut
USBMIDI_Interface usbmidi;// enlever les / en debut de ligne pour activer l'interface usb, penser à désactiver l'interface série(din)
HardwareSerialMIDI_Interface midiser = Serial1;
 
 
MIDI_PipeFactory<8> pipes;
// ----------------------------- Display setup ------------------------------ //
// ========================================================================== //

/*
   Instantiate and initialize the SSD1306 OLED display
*/

constexpr uint8_t SCREEN_WIDTH = 128;
constexpr uint8_t SCREEN_HEIGHT = 64;

constexpr int8_t OLED_DC = 19;    // Data/Command pin of the display
constexpr int8_t OLED_reset = -1; // Use the external RC circuit for reset
constexpr int8_t OLED_CS = 10;   //wired on 13

constexpr uint32_t SPI_Frequency = SPI_MAX_SPEED;

// Instantiate the displays
Adafruit_SSD1306 ssd1306Display = {
  SCREEN_WIDTH, SCREEN_HEIGHT, &SPI,          OLED_DC,
  OLED_reset,   OLED_CS,       SPI_Frequency,
};

// --------------------------- Display interface ---------------------------- //
// ========================================================================== //

// Implement the display interface, specifically, the begin and drawBackground
// methods.
class MySSD1306_DisplayInterface : public SSD1306_DisplayInterface {
 public:
  MySSD1306_DisplayInterface(Adafruit_SSD1306 &display)
    : SSD1306_DisplayInterface(display) {}

  void begin() override {
    // Initialize the Adafruit_SSD1306 display
    if (!disp.begin())
      FATAL_ERROR(F("SSD1306 allocation failed."), 0x1306);

    // If you override the begin method, remember to call the super class method
    SSD1306_DisplayInterface::begin();
  }

  void drawBackground() override { disp.drawLine(1, 8, 126, 8, WHITE); }

} display = ssd1306Display;

CD74HC4067 mux1 = {
  14,       // numéro de broche de l'arduino
  {9, 3, 4, 5} // numéro de pins de l'arduino sur lesquels sont branchés tous les multiplexeurs apellés mux S0, S1, S2
};

CD74HC4067 mux2 = {
  15,              
  {9, 3, 4, 5}, 
};
[... 242 lines cut ...]
  usbmidi >> pipes >> midiser; // all incoming midi from USB is sent to serial
  usbmidi << pipes << midiser; // all incoming midi from Serial is sent to USB
  usbmidi >> pipes >> usbmidi; // all incoming midi from USB is looped back
  midiser << pipes << midiser;
  Control_Surface >> pipes >> usbmidi;
  Control_Surface << pipes << usbmidi;
  Control_Surface >> pipes >> midiser;
  Control_Surface << pipes << midiser;
  usbmidi.begin();
  midiser.begin();
 // The default SPI MOSI pin (11) is used for I²S, so we need to use the
  // alternative MOSI pin (7)
  SPI.setMOSI(11);
  // Correct relative mode for MCU rotary encoders
  RelativeCCSender::setMode(MACKIE_CONTROL_RELATIVE);
  Control_Surface.begin(); // Initialize Control Surface

}                           
void loop() {
  

  Control_Surface.loop(); // Update the Control Surface
  usbmidi.update();
  midiser.update();
}```

tttapa, 2020-10-03:
Your code is way too complicated, start with a shorter example when debugging. Remove 90% of it, leaving a single MIDI interface, a single MIDI input element, and a single display element, without all the mux stuff or MIDI routing. Don't start integrating until your separate components work correctly.

```cpp
constexpr int8_t OLED_CS = 10;   //wired on 13
```
What do you mean "wired on 13"? Pin 13 is the SCK pin, you shouldn't be using it for other purposes. I can't see whether pin 13 is connected from your image. Pin 13 (SCK) should be connected to the clock pin of the display. Please see the Teensy pinout:
[image]

Why are you calling `Control_Surface.begin()` multiple times? Please see https://tttapa.github.io/Control-Surface-doc/Doxygen/d5/da5/Dual-MIDI-Interface_8ino-example.html, you have to connect the pipes before initializing `Control_Surface`.

Do you really want to loop back all MIDI data to the same interface? I just tried it, and it simply crashed my DAW, because the DAW also loops back some messages, creating an infinite loop.

If you are using `Control_Surface` to handle the MIDI interfaces, you don't have to call `usbmidi.begin()` or `update()` yourself.

`SPI.setMOSI(11);` is redundant, pin 11 is the default SPI MOSI pin.

I cannot see any videos, so I'm not sure what you mean by flickering. If the problem persists, try shorter cables or a lower SPI clock speed.

tttapa, 2020-10-03:
Also, what's the value of the capacitor on your reset line? It looks way too small. I'm using 0.1 µF. If it's too small, the display won't reset properly. Also try disconnecting the Teensy entirely to reset the display.

benwadub, 2020-10-03:
So my code is working fine without the display settings, I loop the midi to be able to use my contrôler over usb for a part of it and midi din for the other part and this work nicely!
For the control surface begin it s a mistake I just done by copy/paste from your exemple sorry!
I removed the spi pin didn’t t know it was not useful!
By screen scrolling I mean my vpots and vu meter... are turning from left to right infinitely, and my capacitor is a 104z that is normaly 0,1 µF i m right?

benwadub, 2020-10-03:
I tried removing the pots and all I got on mux and this work fine! I think the problem is because my mux are not connected while I try the display screen!
Could you just write me a line of code for mute button on cc number 83 channel1 display the mute icon on the screen? I d like to not use the mux protocol cause I made a template for my automap on ableton so I won t be able to select my controler as a Mackie control

tttapa, 2020-10-04:
> Could you just write me a line of code for this line

What do you mean?

benwadub, 2020-10-04:
when I push my cc button that send the cc94 on channel that is mapped to my daw mute, I d like that my daw send back this information to my controller screen this way I could know just looking my controller what track is muted or not

tttapa, 2020-10-04:
Use the `NoteValue` class, you already have it in your code, just replace `MCU::MUTE_1` with `94`. Then pass that `NoteValue` object to a `NoteBitmapDisplay` (you already  have this in your code as well).

benwadub, 2020-10-04:
is it doable with a cc button instead of a note value?
I send my mute to the daw like this:
{{bank,BankType::CHANGE_CHANNEL}, mux5.pin(0), {94, CHANNEL_1}},
then I must pass it like this?
// Mute
Bankable::NoteValue<2> mute[] = {
  {bank, {94, CHANNEL_1}},
then add bitmap like this?:
NoteBitmapDisplay muteDisp[] = {
  {display, {94, CHANNEL_1}, XBM::mute_10B, {10, 13}, WHITE},
am I right?

tttapa, 2020-10-04:
The `NoteBitmapDisplay` doesn't do any MIDI related things, so `94, CHANNEL_1` is meaningless in that context. 

You need two things: 1. a `NoteValue` object that listens for MIDI messages on the given address, and 2. a `NoteBitmapDisplay` to display the values that are stored by the `NoteValue` object.

```cpp
// Mute
Bankable::NoteValue<2> mute[] = {
  {bank, {94, CHANNEL_1}},
  {bank, {94, CHANNEL_2}},
  // ...
};
NoteBitmapDisplay muteDisp[] = {
  {display, mute[0], XBM::mute_10B, {10, 13}, WHITE},
  {display, mute[1], XBM::mute_10B, {10, 13}, WHITE},
  // ...
};
```
Please see the C++ syntax rules, and the arguments in the Control Surface documentation. Your curly braces were incorrect as well. Just passing random arguments to the constructors until it compiles will never work, please use the documentation.

benwadub, 2020-10-04:
thanks! but impossible for me to know what note Ableton send for mute :-( I think I ll have to use the mcu protocol!
is this the right way to implement the mcu bank?
NoteButton noteButton[]={
  {
    5,                       // Push button on pin 14
    MCU::BANK_LEFT,
  },
  {
    6,                       // Push button on pin 15
    MCU::BANK_RIGHT
  },

MCU::Bankable::NoteValue<8> mute[] = {
  {bank, MCU::MUTE_1},
  {bank, MCU::MUTE_2},
  {bank, MCU::MUTE_3},
  {bank, MCU::MUTE_4},
  {bank, MCU::MUTE_5},
  {bank, MCU::MUTE_6},
  {bank, MCU::MUTE_7},
  {bank, MCU::MUTE_8},

tttapa, 2020-10-04:
> but impossible for me to know what note Ableton send for mute

You can use the [Mackie-Control-Universal-Reverse-Engineering
](https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/d85/Mackie-Control-Universal-Reverse-Engineering_8ino-example.html) example to see what Ableton is sending. Or you can use a MIDI monitor, or pipe the messages from the USBMIDI_Interface to a USBDebugMIDI_Interface.

> is this the right way to implement the mcu bank?

No, your braces don't match, and if you use the MCU banks, there's no point in using the Control Surface Bankable classes. Also, there is no `MCU::Bankable::NoteValue<8>`, you just need `NoteValue`.

benwadub, 2020-10-04:
I can t find the documentation on the mcu note value, where can I find how I have to write it, when I try with only notes values I get CS::NoteValue' is not a template

tttapa, 2020-10-04:
> I can t find the documentation on the mcu note value, where can I find how I have to write it

There is no `MCU::NoteValue`, only `NoteValue`

> when I try with only notes values I get CS::NoteValue' is not a template

The compiler is right: `CS::NoteValue` is not a template, so don't try to pass any template arguments to it (arguments in angled brackets `<...>`).

benwadub, 2020-10-04:
oh my god this mcu totally confuse all I thought I learnt with you!!!
```cpp
// Mute
NoteValue mute = {
  {MCU::MUTE_1},
  {MCU::MUTE_2},
  {MCU::MUTE_3},
  {MCU::MUTE_4},
  {MCU::MUTE_5},
  {MCU::MUTE_6},
  {MCU::MUTE_7},
  {MCU::MUTE_8},
```
I thought that could work but not

tttapa, 2020-10-04:
What doesn't work?

benwadub, 2020-10-04:
this way that didn't t want to compile now I wrote it like this
```cpp
NoteButton noteButton[]={
  {
    14,                       // Push button on pin 14
    MCU::BANK_LEFT,
  },
  {
    15,                       // Push button on pin 15
    MCU::BANK_RIGHT
  },
};
// Mute
NoteValue mute []= {
  {{MCU::MUTE_1}},
  {{MCU::MUTE_2}},
  {{MCU::MUTE_3}},
  {{MCU::MUTE_4}},
  {{MCU::MUTE_5}},
  {{MCU::MUTE_6}},
  {{MCU::MUTE_7}},
  {{MCU::MUTE_8}},
  
};
// Mute
NoteBitmapDisplay muteDisp[] = {
  {display, mute[0], XBM::mute_10B, {10, 13}, WHITE},
  {display, mute[1], XBM::mute_10B, {42, 13}, WHITE},
  {display, mute[2], XBM::mute_10B, {74, 13}, WHITE},
  {display, mute[3], XBM::mute_10B, {106, 13}, WHITE},
  {display, mute[4], XBM::mute_10B, {10, 38}, WHITE},
  {display, mute[5], XBM::mute_10B, {42, 38}, WHITE},
  {display, mute[6], XBM::mute_10B, {74, 38}, WHITE},
  {display, mute[7], XBM::mute_10B, {106, 38}, WHITE},
  

};```
the code compile but now I have no tracks that send the mute icon for Ableton, before that worked for the 8 first tracks and now no tracks work :-(

tttapa, 2020-10-04:
Post your complete code. Did you select the Mackie Control Universal in the Ableton settings?

benwadub, 2020-10-04:
i i selected the Mackie control as output on my daw, rectification tracks 2 3 4 5 6 7 8 work but the 1 9 10...16 doesn't t
```cpp
#include <Control_Surface.h>

#include <Encoder.h> // Include the Encoder library.
#include "Control_Surface.h"
#include <Display/DisplayInterfaces/DisplayInterfaceSSD1306.hpp>

//#include <name.c>
// celà doit etre écrit avant la library control surface
 //USBDebugMIDI_Interface midi = 115200; // enlever les // en début de ligne pour entrer en mode debug usb et voir dans le panneau de control si vos controler envoient bien les infos

//auto &serial = Serial1;// Selectionne le port série à utiliser remplacer par serial pour une arduino
//SerialMIDI_Interface<decltype(serial)> midi = {serial, MIDI_BAUD};// démarre une interface midi serial au midi baud rate par defaut
USBMIDI_Interface usbmidi;// enlever les / en debut de ligne pour activer l'interface usb, penser à désactiver l'interface série(din)
HardwareSerialMIDI_Interface midiser = Serial1;
 
 
MIDI_PipeFactory<8> pipes;
// ----------------------------- Display setup ------------------------------ //
// ========================================================================== //

/*
   Instantiate and initialize the SSD1306 OLED display
*/

constexpr uint8_t SCREEN_WIDTH = 128;
constexpr uint8_t SCREEN_HEIGHT = 64;

constexpr int8_t OLED_DC = 19;    // Data/Command pin of the display
constexpr int8_t OLED_reset = -1; // Use the external RC circuit for reset
constexpr int8_t OLED_CS = 10;   //wired on 13

constexpr uint32_t SPI_Frequency = SPI_MAX_SPEED;

// Instantiate the displays
Adafruit_SSD1306 ssd1306Display = {
  SCREEN_WIDTH, SCREEN_HEIGHT, &SPI,          OLED_DC,
  OLED_reset,   OLED_CS,       SPI_Frequency,
};

// --------------------------- Display interface ---------------------------- //
// ========================================================================== //

// Implement the display interface, specifically, the begin and drawBackground
// methods.
class MySSD1306_DisplayInterface : public SSD1306_DisplayInterface {
 public:
  MySSD1306_DisplayInterface(Adafruit_SSD1306 &display)
    : SSD1306_DisplayInterface(display) {}

  void begin() override {
    // Initialize the Adafruit_SSD1306 display
    if (!disp.begin())
      FATAL_ERROR(F("SSD1306 allocation failed."), 0x1306);

    // If you override the begin method, remember to call the super class method
    SSD1306_DisplayInterface::begin();
  }

  void drawBackground() override { disp.drawLine(1, 8, 126, 8, WHITE); }

} display = ssd1306Display;

/*
CD74HC4067 mux1 = {
  14,       // numéro de broche de l'arduino
  {9, 3, 4, 5} // numéro de pins de l'arduino sur lesquels sont branchés tous les multiplexeurs apellés mux S0, S1, S2
};

[... 270 lines cut ...]
  Control_Surface.begin(); // initialise la library surface de control
  usbmidi >> pipes >> midiser; // all incoming midi from USB is sent to serial
  usbmidi << pipes << midiser; // all incoming midi from Serial is sent to USB
  //usbmidi >> pipes >> usbmidi; // all incoming midi from USB is looped back
  midiser << pipes << midiser;
  Control_Surface >> pipes >> usbmidi;
  Control_Surface << pipes << usbmidi;
  Control_Surface >> pipes >> midiser;
  Control_Surface << pipes << midiser;
  usbmidi.begin();
  midiser.begin();
 // The default SPI MOSI pin (11) is used for I²S, so we need to use the
  // alternative MOSI pin (7)
  SPI.setMOSI(11);
  // Correct relative mode for MCU rotary encoders
  RelativeCCSender::setMode(MACKIE_CONTROL_RELATIVE);
  
}                           
void loop() {
  

  Control_Surface.loop(); // Update the Control Surface
  usbmidi.update();
  midiser.update();
}```

benwadub, 2020-10-04:
https://forum.ableton.com/viewtopic.php?t=190422
hmmm it seems that the Ableton mcu script has problem with mute buttons for others people too! maybe the problem has to be solved there and not in my code!

tttapa, 2020-10-04:
Please see the remarks I made earlier:

> Why are you calling `Control_Surface.begin()` multiple times? Please see https://tttapa.github.io/Control-Surface-doc/Doxygen/d5/da5/Dual-MIDI-Interface_8ino-example.html, you have to connect the [...]

Try removing all things that are not related to the buttons/mute bitmaps from your code. 

Try using a `USBDebugMIDI_Interface` to trigger the display from the Serial Monitor. `90 10 7F` should enable the first mute bitmap, and `80 10 7F` should disable it.

benwadub, 2020-10-04:
oops sorry I made the modification you mentioned on the original code! I copy pasted it to be sure to not make a terrible thing on it!
yes that work for triggering via monitor in debug midi interface mode, that s for the tracks 9 to 16 that I can t have it working.

benwadub, 2020-10-04:
woh damn that work also on track 9 if I use the bank button, the problem is so as I mentioned just before withe the mcu script for Ableton!
i m in parallel making the script for my controller, I ll use the same note number as the mcu and that could work!
again a big thanks for your help and patience!

tttapa, 2020-10-04:
The MCU banks are handled in the DAW, so it's normal that only tracks 1-8 work when using the debug MIDI interface.

benwadub, 2020-10-04:
they don t work with the normal usb interface too, I have a small led indicator in Ableton. that blink when a midi message is output, it blink blue when I switch mute with my mouse on track 1 to 8 but doesn't t blink with the tracks 9 to 16, now I m sure the Ableton script is buggy and my code works fine!

tttapa, 2020-10-04:
They'll only send MIDI when you've selected the second bank (using the BANK_LEFT/BANK_RIGHT buttons), but of course, your DAW/script has to support these MCU banks.

benwadub, 2020-10-04:
oh I found the problem! I have a2 button to manage my banks and added the same button to the mcu bank change but In wraps/wraps mode :-( so my banks displayed was never the same as the bank that was modified by the mute in Ableton! now all is working fine! and you know that push button can manage control surface bank at the same time as mcu bank but the have to be in wrap/clamp mode!
thanks again!

benwadub, 2020-10-21:
hi Pieter, hope you are fine! I made working my screen nicely now I try to understand how to adapt it to my needs, I set Ableton to receive and send cc94 for mute, how can I write the code for my display get state of cc number 94 instead of
```cpp
NoteBitmapDisplay muteDisp[] = {
  {display, mute[0], XBM::mute_10B, {10, 13}, WHITE},```
like in your exemple
I thought writing it like this could work?
 {display, {94, CHANNEL_1}, XBM::mute_10B, {10, 13}, WHITE},

tttapa, 2020-10-21:
Please see my previous reply:

> The `NoteBitmapDisplay` doesn't do any MIDI related things, so `94, CHANNEL_1` is meaningless in that context.
> 
> You need two things: 1. a `NoteValue` object that listens for MIDI messages on the given address, and 2. a `NoteBitmapDisplay` to display the values that are stored by the `NoteValue` object.
> 
> ```c++
> // Mute
> Bankable::NoteValue<2> mute[] = {
>   {bank, {94, CHANNEL_1}},
>   {bank, {94, CHANNEL_2}},
>   // ...
> };
> NoteBitmapDisplay muteDisp[] = {
>   {display, mute[0], XBM::mute_10B, {10, 13}, WHITE},
>   {display, mute[1], XBM::mute_10B, {10, 13}, WHITE},
>   // ...
> };
> ```

If you want to use MIDI control change instead of notes, use `CCValue` instead of `NoteValue`.

benwadub, 2020-10-21:
sorry I didn't t understood all, English is not my native language so sometimes I don t understand all very well! so I send my mute like this:
```cpp
Bankable::CCButtonLatched<2> mute[] = { 
  {{bank,BankType::CHANGE_ADDRESS}, mux5.pin(0), {94, CHANNEL_1}},//numéro de bank correspondant/indique que le changement de bank change le canal midi (+8 car on a 8 adresses par bank)/pin sur laquelle le controleur est branché/numéro de cc/numéro de canal midi
  {{bank,BankType::CHANGE_ADDRESS},mux5.pin(1), {94, CHANNEL_2}},
  {{bank,BankType::CHANGE_ADDRESS},mux5.pin(2), {94, CHANNEL_3}},
  {{bank,BankType::CHANGE_ADDRESS},mux5.pin(3), {94, CHANNEL_4}},
  {{bank,BankType::CHANGE_ADDRESS},mux5.pin(4), {94, CHANNEL_5}},
  {{bank,BankType::CHANGE_ADDRESS},mux5.pin(5), {94, CHANNEL_6}},
  {{bank,BankType::CHANGE_ADDRESS},mux5.pin(6), {94, CHANNEL_7}},
  {{bank,BankType::CHANGE_ADDRESS},mux5.pin(7), {94, CHANNEL_8}},
};```
how does the bitmap display should be written please.

tttapa, 2020-10-21:
First you need a `CCValue` object to receive the MIDI data from Ableton:
```cpp
Bankable::CCValue<2> muteAbltn[] = {
  {{bank, BankType::CHANGE_ADDRESS}, {94, CHANNEL_1}},
  {{bank, BankType::CHANGE_ADDRESS}, {94, CHANNEL_2}},
  // ...
};
```
Then you need something to display the value you received:
```cpp
NoteBitmapDisplay muteDisp[] = {
  {display, muteAbltn[0], XBM::mute_10B, {10, 13}, WHITE},
  {display, muteAbltn[1], XBM::mute_10B, {26, 13}, WHITE},
  // ...
};
```

benwadub, 2020-10-21:
ohhhh yeahhhh thanks!! I finally understand! I was misunderstanding the cc value receiver and the cc I sent from my controller! all is clear now and that compile! now I ll be able to get away from the MCU and be able to use the Ableton script I made without using the MCU script except to receive theme display

benwadub, 2020-10-21:
a last question for tonight :-) why does this don t work for play and reçu button with ccvalue?
```cpp
CCValue play = {103, CHANNEL_1};
CCValue record = {104, CHANNEL_1};```

tttapa, 2020-10-21:
The MIDI address is a single argument, so you have to group it using braces:
```cpp
CCValue play = {{103, CHANNEL_1}};
```

benwadub, 2020-10-22:
thanks!! a lot!
for the time display in MCU do you know how that work? MCU receive midi time code and show it on display? do you know if that s possible to receive it without using MCU?

tttapa, 2020-10-22:
No, it doesn't use MIDI Time Code, the time is sent as text using Control Change messages. The implementation is as follows:

https://github.com/tttapa/Control-Surface/blob/7f0ce0f6f1122fd2b2fe140064268db349be8468/src/MIDI_Inputs/MIDIInputElementMatchers.hpp#L79-L103

https://github.com/tttapa/Control-Surface/blob/7f0ce0f6f1122fd2b2fe140064268db349be8468/src/MIDI_Inputs/MCU/SevenSegmentDisplay.hpp#L43-L55

Also see page 116 of https://images.thomann.de/pics/prod/151261_manual.pdf.

[3 more replies]