tttapa/Control-Surface #287: Master /slave with rs485

radiservi, 2020-09-27:
hello Pieter I would like to design a control surface with several slave modules where a potentiometer button and other would be connected to a master module which would output in USB midi I saw your solution from the uart but I did not understand correctly.  i would like to use rs485 communication but i don't know by where command could you help me?
```

tttapa, 2020-09-28:
I've never used RS485, and I don't know if the half-duplex nature of the protocol will be a problem.

If the RS485 library you're using is fully compatible with the Arduino `Stream` interface, you can just use the [`StreamMIDI_Interface`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/dc3/classStreamMIDI__Interface.html). If it requires more code than just writing and reading, e.g. to start/end a transmission, you'll have to write your own `MIDI_Interface`. This shouldn't be hard at all, you can just start from the `StreamMIDI_Interface` and add the required RS485-specific code:

https://github.com/tttapa/Control-Surface/blob/e5650f8de651c1dbc3b544630fb4afbcb0611fda/src/MIDI_Interfaces/SerialMIDI_Interface.hpp#L16-L95
(Unless you're using an ESP32, you don't have to worry about the mutexes.)

radiservi, 2020-09-28:
ok then is there an easier method than rs485

tttapa, 2020-09-28:
Well, it depends on what you want to achieve.  
Generally, I wouldn't recommend using multiple microcontrollers at all, simply use a microcontroller that is powerful enough to do everything.  
If you need to cover large distances, RS485 is not a bad idea, unless you need full-duplex communication.

radiservi, 2020-09-28:
non-objective and connected several modules with different mounting of button and potentiometer and I can not fit everything on a single controller I would like to add only the modules that interests me to work on my song and if I need other  check I add the necessary module you understand?

tttapa, 2020-09-28:
Why would you use RS485 for the connection between modules? They would at most be a couple of centimeters apart, right?

radiservi, 2020-09-28:
yes it would be separated by 30cm max between each module but I would like to have up to 10 connected to the same master database I work on the basis of the code of fuzzywobble but I have reached the end of the code because it does not also have  complete as yours and i2c and too subject to interference as soon as I connect 2 module with potentiometer if I move them at the same time it bug and stops

tttapa, 2020-09-28:
30cm should not be an issue for I²C. You'll need one I²C master on the bus, so two slaves won't interfere with eachother.

radiservi, 2020-09-28:
1 module is 30 cm so you have to add them and the i2c really poses a problem because in addition I use magnetic plugs and a lot of magnets for the connection

radiservi, 2020-09-28:
if I connect almost all the modules I have more than 3-4 meters of cable

radiservi, 2020-09-28:
Do you have a Facebook Instagram or other so that we can discuss in more detail with photos and other

tttapa, 2020-09-28:
Well I can't really help you with that, you'll have to ask this on the Arduino forum, for example. 

You can use MIDI over almost any transport that can move bytes around, and to make it work with the Control Surface library, you just need to implement the `MIDI_Interface` interface, as explained in my first reply.

radiservi, 2020-09-28:
ok thank you if you want work i can provide you with work and a part time job as a freelance or consultant in my company because i find very few people as experienced as you in midi

tttapa, 2020-09-28:
I'm currently not looking for a job, but I appreciate the offer.

DeltaC6, 2020-10-18:
> yes it would be separated by 30cm max between each module but I would like to have up to 10 connected to the same master database I work on the basis of the code of fuzzywobble but I have reached [...]

If you are separating by just 30 cm, no need to use RS485 at all.... Use RS232 it is applicable upto 50 feets. You then don't have to change the code or code bases at all. RS232 will elevate the signal voltage levels without any change of data and/or adding other code or library.

masydoblig, 2020-10-19:
@DeltaC6 

Let me know how you go with this i too would like to connect several slave modules to a master module
that way i only have one connection to the computer. 

but i definitely have no idea about the programming LOL just ask peter (i am sure he is sick of questions) sorry peter.
the hardware on the other hand i can design quite well.

i was thinking that the slaves could just send and receive the basic states of connected peripherals ie buttons, pots. 
then the master is responsible for all the midi conversion and communication. (over i2c or spi).

Then i could easily make the below array fader bank. (each fader currently uses 1 usb connection) but is very fast in cubase with no glitches. https://www.youtube.com/watch?v=sIzhsGYoD2U video link.
[image]

this is the Eq module which also works flawlessly (just waiting on the face plates to be laser cut).
[image]

radiservi, 2020-11-01:
hello Pieter I found a library compatible with the stream but I can not put it in place I do not understand how streammidi_interface works .. 
do you have an example? 
 i am using the simplecomm.h library from tools40
 if you can show me how i would appreciate it

radiservi, 2020-11-01:
#include <RS485.h>
#include <SimpleComm.h>
#include <Control_Surface.h>

uint8_t slaveAddress = 1;

// Instantiate a MIDI Interface to us
StreamMIDI_Interface (stream);

// Instantiate an analog multiplexer
CD74HC4051 mux = {
  A0,       // Analog input pin
  {3, 4, 5} // Address pins S0, S1, S2
};

// Create an array of potentiometers that send out
// MIDI Control Change messages when you turn the
// potentiometers connected to the eight input pins of
// the multiplexer
CCPotentiometer volumePotentiometers[] = {
  {mux.pin(0), {MIDI_CC::Channel_Volume, CHANNEL_1}},
  {mux.pin(1), {MIDI_CC::Channel_Volume, CHANNEL_2}},
  {mux.pin(2), {MIDI_CC::Channel_Volume, CHANNEL_3}},
  {mux.pin(3), {MIDI_CC::Channel_Volume, CHANNEL_4}},
  {mux.pin(4), {MIDI_CC::Channel_Volume, CHANNEL_5}},
  {mux.pin(5), {MIDI_CC::Channel_Volume, CHANNEL_6}},
  {mux.pin(6), {MIDI_CC::Channel_Volume, CHANNEL_7}},
  {mux.pin(7), {MIDI_CC::Channel_Volume, CHANNEL_8}},
};

// Initialize the Control Surface
void setup() {
  Control_Surface.begin();
  Serial.begin(9600L);

  // Start RS485
  RS485.begin(19200L);
  RS485.setTimeout(20);

  // Start SimpleComm
  SimpleComm.begin(slaveAddress);
}

// Update the Control Surface
void loop() {
  Control_Surface.loop();
}

tttapa, 2020-11-01:
The SimpleComm library doesn't implement the [Arduino `Stream` API](https://www.arduino.cc/reference/en/language/functions/communication/stream/), so you cannot use it with the `StreamMIDI_Interface` class.

If you want to use this library, you'll have to implement your own MIDI interface, you can follow the `StreamMIDI_Interface` very closely as an example:

https://github.com/tttapa/Control-Surface/blob/0efe4a882766894e4e82d5736e4927e92b9d7f41/src/MIDI_Interfaces/SerialMIDI_Interface.hpp#L16-L95

As a side note and for future reference: the internals of the `MIDI_Interface` have changed in version 2.0 (which hasn't been released yet), it makes it easier to implement your own MIDI interfaces. If you want to future-proof your RS485 MIDI interface, I'd recommend using the new API:

https://github.com/tttapa/Control-Surface/blob/f1cdf5ff09c59ab7c17b720a8a3e83e968cce1ba/src/MIDI_Interfaces/SerialMIDI_Interface.hpp#L12-L64

https://github.com/tttapa/Control-Surface/blob/f1cdf5ff09c59ab7c17b720a8a3e83e968cce1ba/src/MIDI_Interfaces/SerialMIDI_Interface.cpp#L8-L35

https://github.com/tttapa/Control-Surface/blob/f1cdf5ff09c59ab7c17b720a8a3e83e968cce1ba/src/MIDI_Parsers/StreamPuller.hpp#L8-L29

radiservi, 2020-12-11:
hello Pieter can you explain to me how the stream works with the wire library because I cannot make it work

radiservi, 2020-12-11:
// Include the library
#include <Wire.h>
 
#include <Control_Surface.h>

// Instantiate a MIDI Interface to use
StreamMIDI_Interface(Stream ) ;

// Instantiate an analog multiplexer
CD74HC4051 mux = {
  A0,       // Analog input pin
  {3, 4, 5} // Address pins S0, S1, S2
};

// Create an array of potentiometers that send out
// MIDI Control Change messages when you turn the
// potentiometers connected to the eight input pins of
// the multiplexer
CCPotentiometer volumePotentiometers[] = {
  {mux.pin(0), {MIDI_CC::Channel_Volume, CHANNEL_1}},
  {mux.pin(1), {MIDI_CC::Channel_Volume, CHANNEL_2}},
  {mux.pin(2), {MIDI_CC::Channel_Volume, CHANNEL_3}},
  {mux.pin(3), {MIDI_CC::Channel_Volume, CHANNEL_4}},
  {mux.pin(4), {MIDI_CC::Channel_Volume, CHANNEL_5}},
  {mux.pin(5), {MIDI_CC::Channel_Volume, CHANNEL_6}},
  {mux.pin(6), {MIDI_CC::Channel_Volume, CHANNEL_7}},
  {mux.pin(7), {MIDI_CC::Channel_Volume, CHANNEL_8}},
};

// Initialize the Control Surface
void setup() {
  Control_Surface.begin();
  Wire.begin(1); 
}

// Update the Control Surface
void loop() {
  Control_Surface.loop();
}

tttapa, 2020-12-15:
> StreamMIDI_Interface(Stream ) ;

This is incorrect syntax. Both StreamMIDI_Interface and Stream are types.

You have to instantiate an object of type StreamMIDI_Interface: `StreamMIDI_Interface myMIDIObject { ... };`  
The `...` is the initialization, the arguments for the StreamMIDI_Interface constructor, see [`StreamMIDI_Interface::StreamMIDI_Interface()`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/dc3/classStreamMIDI__Interface.html#a3760ef5fcb2a7736d80a1248ea9cd107). The only argument is the stream to read from and write to, which is of type [`Stream`](https://www.arduino.cc/reference/en/language/functions/communication/stream/).

As an example, `SoftwareSerial` implements the `Stream` interface:

```cpp
SoftwareSerial mySerial(2, 3);
StreamMIDI_Interface myMIDIObject { mySerial };
```
The Arduino [`Wire`](https://www.arduino.cc/en/Reference/Wire) class also implements the `Stream` interface, but you might need additional logic to handle transactions. It's unclear to me how you plan to use the I²C interface for MIDI.