tttapa/Control-Surface #288: Suggestion for this awesome library

giubu, 2020-09-28:
Hello Pieter,
I wanted to ask you if you know about [XOD](https://xod.io/), the visual programming language for Arduino.
It would be incredible if somehow this library could be ported to be used in it, it would allow the less coding inclined of us to create amazing instruments in a visual manner. 
Just throwing it out there to see what happens...
thank you again for the library.

tttapa, 2020-09-30:
That's an interesting project, I hadn't heard of it before. 

I'm afraid the XOD architecture doesn't really suit Control Surface's structure. Control Surface handles most of its communication internally, while XOD seems to be focussed on communication between nodes. 

I have been thinking about creating a graphical code generator with a beginner-friendly user interface, but I don't have time at the moment to brush up my front-end Javascript skills, so don't expect it any time soon.

giubu, 2020-10-03:
No no, I am not expecting anything, I just wanted to point out the XOD project to you and see what you thought of it.
All the best