tttapa/Control-Surface #291: CCButtons Help

bmarshall91, 2020-09-29:
Hi I am trying to use multiple non latching buttons to send CC but I can't see an example for CCButtons that could help. 
I can do one button no problem, and even multiple buttons with program change but I have a problem with it being a one shot.

I know the code below doesn't work, I don't get past the compiling stage. 

Please help a newbie out

Thanks

Ben

```cpp

// #include <Control_Surface.h> // Include the Control Surface library
#include <MIDI_Interfaces/BluetoothMIDI_Interface.hpp>
// Instantiate a MIDI over Bluetooth BLE interface
BluetoothMIDI_Interface midi;

// Instantiate a CCButton object
CCButton button = {
  // Push button on pin 26:
  26,{MIDI_CC::General_Purpose_Controller_1, CHANNEL_1},
};
CCButton button = {
  // Push button on pin 27:
  27,{MIDI_CC::General_Purpose_Controller_2, CHANNEL_1},
};
void setup() {
  Control_Surface.begin(); // Initialize Control Surface
}

void loop() {
  Control_Surface.loop(); // Update the Control Surface
}

```

alomdj00, 2020-09-29:
Debes incluir la libreria  contros surface, en la primera linea  esta pero comentada,  elimina los //

bmarshall91, 2020-09-29:
Sorry that was just a typo as I've pasted it in. I just don't understand how to use multiple buttons for different CCs.

I know to use CCbuttons instead of CCbutton but I don't know how...?

tttapa, 2020-09-30:
You cannot have two variables with the same name and the same scope. You have two variables with the name `button`.

It's easiest to use an array, please see the [Getting Started](https://tttapa.github.io/Control-Surface-doc/Doxygen/d5/d7d/md_pages_Getting-Started.html) tutorial and the [Multiple-Control-Change-Potentiometers.ino](https://tttapa.github.io/Control-Surface-doc/Doxygen/d3/d8d/Multiple-Control-Change-Potentiometers_8ino-example.html) example. The [MIDIAddress](https://tttapa.github.io/Control-Surface-doc/Doxygen/d6/dc6/md_pages_Basics.html) documentation could be useful as well to understand how the initialization works.

The `CCButtons` class allows you to declare multiple buttons at the same time. The restriction is that their addresses/controller numbers have to have a fixed interval between them. E.g. controller numbers `#5, #10, #15 and #20`. Please see the [documentation](https://tttapa.github.io/Control-Surface-doc/Doxygen/d0/dd3/classCCButtons.html#adf36d36ccf9d119fb357310e6920c72b) for more information.

For example:
```cpp
CCButtons<4> buttons = {
  {26, 27, 28, 29}, // button pins
  {MIDI_CC::General_Purpose_Controller_1, CHANNEL_1}, // address of first button
  {1, 0}, // interval between addresses (1 controller number, 0 channels)
};
```
The address of the first button on pin 26 will be 0x10 (general purpose controller 1) on channel 1. The address of the second button on pin 27 will be controller 0x10 + 1 = 0x11 (general purpose controller 2) on channel 1 + 0 = channel 1. The address of the third button will be controller 0x11 + 1 = 0x12 (general purpose controller 3) on channel 1 + 0 = channel 1. And so on.  
The `4` between angled brackets `<4>` specifies the number of buttons.

bmarshall91, 2020-09-30:
Thanks so much, that's exactly what I was trying to do, but how to lay it out I couldn't quite work out from the docs and examples... 
I'll look through the documentation to make sure I understand it all. But you've made it look so easy. Thanks

closed 2020-09-30 by bmarshall91