tttapa/Control-Surface #292: RGB Button (Launchpad) & MIDI Remote Scripts

milkos86, 2020-09-30:
Hey.
First of all, thank you for your great work. Even if I do not succeed in what I wrote below, I still think that this is one of the coolest projects that I have ever seen on Github! :)
I'm going to build a controller for Ableton live.
Task - Roughly repeat the LaunchPad which receives the colors of the loops when the button is pressed from the live session of Ableton to the RGB LEDs of the controller (that is, there will be a button and next to it an RGB LED). I also want to make for him "MIDI Remote Scripts" in the future, which will move the frame for further work with the live session. Will your code work with such an array of data from Ableton, is it possible to build this kind of controller?
Board - Leonardo
I apologize in advance if I did not ask the question correctly

[image]

4dvn, 2020-10-01:
Is possible, check my project git

tttapa, 2020-10-01:
For reading the buttons, you can use the [`NoteButton`](https://tttapa.github.io/Control-Surface-doc/Doxygen/de/dcc/NoteButton_8ino-example.html) class, for the LEDs, you could use the [`NoteRangeFastLED`](https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/dd2/9_8Note-FastLED_8ino-example.html) class.  
By default, the color scheme used by `NoteRangeFastLED` is the same as the Novation Launchpad (I got it from their manual, I don't own a Launchpad, so I can't test it). You can also define your own color schemes: [Note-FastLED-ColorMapper](https://tttapa.github.io/Control-Surface-doc/Doxygen/d9/d6d/10_8Note-FastLED-ColorMapper_8ino-example.html).

I'd recommend not using clockless addressable LEDs, they have very tight timing constraints, so interrupts are disabled while updating them, which could result in MIDI data being dropped/lost (MIDI is often received in interrupt service routines, these don't work when interrupts are disabled). LEDs with clock lines aren't timing sensitive, so interrupts don't have to be disabled.

milkos86, 2020-10-21:
> For reading the buttons, you can use the [`NoteButton`](https://tttapa.github.io/Control-Surface-doc/Doxygen/de/dcc/NoteButton_8ino-example.html) class, for the LEDs, you could use the [...]

I apologize for disturbing you, but could you tell me what example will be used for the buttons to connect them through the multipexor 4051-4067. I will connect the tape directly. But with the buttons, there is not enough space. Will this option work with NeoPixel RGB Tape?

tttapa, 2020-10-21:
The [Getting Started](https://tttapa.github.io/Control-Surface-doc/Doxygen/d5/d7d/md_pages_Getting-Started.html) guide uses multiplexers for analog inputs, you can use the exact same principles for buttons as well, simply specify the right pin.

```cpp
// Instantiate a multiplexer
CD74HC4051 mux = {
  2,        // Digital input pin
  {3, 4, 5} // Address pins S0, S1, S2
};
 
// Create an array of CCButton objects that send out
// MIDI Control Change messages when you press the buttons
// connected to the eight input pins of
// the multiplexer
CCButton volumePotentiometers[] = {
  {mux.pin(0), {MIDI_CC::General_Purpose_Controller_1, CHANNEL_1}},
  {mux.pin(1), {MIDI_CC::General_Purpose_Controller_1, CHANNEL_2}},
  {mux.pin(2), {MIDI_CC::General_Purpose_Controller_1, CHANNEL_3}},
  {mux.pin(3), {MIDI_CC::General_Purpose_Controller_1, CHANNEL_4}},
  {mux.pin(4), {MIDI_CC::General_Purpose_Controller_1, CHANNEL_5}},
  {mux.pin(5), {MIDI_CC::General_Purpose_Controller_1, CHANNEL_6}},
  {mux.pin(6), {MIDI_CC::General_Purpose_Controller_1, CHANNEL_7}},
  {mux.pin(7), {MIDI_CC::General_Purpose_Controller_1, CHANNEL_8}},
};
```