tttapa/Control-Surface #294: mega 2560

jovabra, 2020-10-05:
Olá amigo ... 

[Control Surface ](https://github.com/tttapa/Control-Surface) é compatível com MEGA 2560?
Seu projeto, tem algum custo, caso eu queira usar?
Obrigado ...

4dvn, 2020-10-05:
yes

tttapa, 2020-10-05:
Yes, you can use Control Surface on an Arduino Mega. However, AVR is not the main target platform, and I wouldn't recommend it at all if you want to use MIDI over USB, because the Arduino Mega doesn't have native USB support.

See the [MIDI over USB](https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/d4a/md_pages_MIDI-over-USB.html) page for a full discussion of compatible boards and their MIDI over USB support status.

The Control Surface library is free and open source software.

jovabra, 2020-10-05:
> Sim, você pode usar a superfície de controle em um Arduino Mega. No entanto, o AVR não é a plataforma de destino principal e eu não o recomendaria de forma alguma se você quiser usar MIDI sobre USB, porque o Arduino Mega não tem suporte nativo para USB.
> 
> Consulte a página [MIDI sobre USB](https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/d4a/md_pages_MIDI-over-USB.html) para uma discussão completa sobre placas compatíveis e seu status de suporte a MIDI sobre USB.
> 
> A biblioteca do Control Surface é um software gratuito e de código aberto, e você pode baixá-lo gratuitamente, consulte as [instruções de instalação](https://tttapa.github.io/Control-Surface-doc/Doxygen/d8/da8/md_pages_Installation.html) .
> Você deve cumprir a licença GPLv3, o que significa que não pode usar o Control Surface como parte de produtos proprietários.

ok meu amigo, obrigado pelo seu tempo .... 

eu realmente gostaria de aproveitar a Placa Compatível com Arduino, Mega 2560, com a finalidade de interface Midi. Como o mega não tem suporte nativo para usb (dei uma lida na pagina sugerida sobre as plataformas de arduino) e pergunto:

minha placa **Mega 2560** é o modelo mais novo com o  chip CH340, nesse caso eu poderia usar a saída para pino Midi e plugar um adaptador conversor midi para usb? 
Exemplo do adaptador: [Robson Martins](https://robsonmartins.com/content/eletr/midiusb/)

**Sim tttpa <-> PieterP, vi sua resposta no forum Arduino.cc. Obrigado**

tttapa, 2020-10-05:
You could do that, but personally, I'd buy a board that does support MIDI over USB.

If you do decide to build a separate MIDI to USB adapter, keep in mind that you don't have to add all the MIDI-specific hardware such as optocouplers and 220Ω resistors. Both the ATmega2560 and the MIDI-to-USB chip will probably have TTL UARTs, so you can connect them directly without creating a current loop.

jovabra, 2020-10-05:
eu vou trabalhar no projeto, [ caixas, botões, adaptador, software, etc ] mostro depois resultado, muito obrigado pela atenção.

jovabra, 2020-10-23:
Olá Peeter! o mapeamento do  Potenciometer A0, usando Hairless e loopmidi, Da certo, a Daw no potenciometro de volume é mapeado, mais nada acontece, não consigo movimentar o controle.

#include <Control_Surface.h> // Include the library
CCPotentiometer potentiometer(A0, MIDI_CC::Channel_Volume, 1);
void setup() {}
void loop() {
    Control_Surface.update();
}

tttapa, 2020-10-23:
You forgot to declare a MIDI interface, and you didn't initialize Control Surface in the setup. 

Please see the [Getting Started](https://tttapa.github.io/Control-Surface-doc/Doxygen/d5/d7d/md_pages_Getting-Started.html) guide.