tttapa/Control-Surface #295: Rotary encoder with ESP32?

shaunchu, 2020-10-05:
Hello,
Sorry for asking noob question, what should I do to get rotary encoder to work with esp32?

I have the esp32 working with buttons via Wi-Fi and BLE on Mac, and I notice the rotary encoder sample doesn't list esp32 as supported board. What should I do? am I missing something?

For any help great thanks in advance.
Best Regards

tttapa, 2020-10-05:
Control Surface uses PaulStoffregen/Encoder to read from rotary encoders. It seems to support the ESP32:

https://github.com/PaulStoffregen/Encoder/blob/006039d4a61fe8f27de9e3fd7fc8362c6909b009/utility/interrupt_pins.h#L211-L244

I haven't tried it, though, but it should work.

shaunchu, 2020-10-06:
thanks for pointing me to the code, I finally found out somehow I had downloaded an old version of PaulStoffregen's library and it's missing the exact part of code you highlighted, thanks!!

closed 2020-10-06 by shaunchu