tttapa/Control-Surface #296: Teensy 2.0

huggre, 2020-10-08:
Hi, why is the Teensy 2.0 supported?

tttapa, 2020-10-08:
I don't own a Teensy 2.0, so I can't test it, but it seems to compile without errors for me.

Are you getting a compilation error? A runtime error? Does it simply not send any MIDI?

huggre, 2020-10-08:
Thanks for your reply. Reason I asked was that it was not listed under supported devices.
Tried anyway and got the following compile error:

C:\Users\hgreg\Documents\Arduino\libraries\Control-Surface-1.1.1\src\MIDI_Interfaces\Teensy-USB-MIDI-name.c:7:10: fatal error: usb_names.h: No such file or directory
 #include "usb_names.h"
Multiple libraries were found for "Control_Surface.h"
          ^~~~~~~~~~~~~
 Used: C:\Users\hgreg\Documents\Arduino\libraries\Control-Surface-1.1.1
compilation terminated.
 Not used: C:\Users\hgreg\Documents\Arduino\libraries\Control-Surface-master
Error compiling for board Teensy 2.0.

Have been testing the Teensy 2.0 with other MIDI libraries today without any issues.

tttapa, 2020-10-08:
You seem to be using an old version. Try at least version 1.2.0 or `master`. Version 1.1.1 didn't support the Teensy 2.0.

tttapa, 2020-10-08:
The `usb_names.h: No such file or directory` error was fixed in back in May:
https://github.com/tttapa/Control-Surface/commit/3f640698fd4675a4d4fdd75d1003c8e0a387360c

huggre, 2020-10-08:
Works after deleting old version of the library.
Looking forward to learning more about this library :-)

Note! You should consider adding the Teensy 2.0 to the list of supported boards.

tttapa, 2020-10-08:
Glad to hear :-) 

Could you let me know if things like MIDI over USB actually work? Teensy 2.0 uses a different MIDI USB implementation than the Teensy 3.x boards, and I haven't been able to test that yet. There could be problems with flushing the MIDI data, for example, which could cause high latency.

huggre, 2020-10-09:
Hi, looks like the MIDI over USB is working based on some very simple tests i did. There is some latency but i'm not sure what to expect.
Will give you you an update as soon as I have more to report.

By the way.., I'm not able to get the "NoteButtonMatrix" function to work. I know the hardware is working correctly but no note on/off events are created when testing your example. Could you please verify that it is working correctly from your side and let me know what i might be missing.

Thanks for your support

tttapa, 2020-10-09:
You could try to add `usbMIDI.send_now()` to your main loop, I'm not sure how long the T2.0 buffers the data before actually sending it.

I've just tried the [NoteButtonMatrix.ino](https://tttapa.github.io/Control-Surface-doc/Doxygen/d4/d53/NoteButtonMatrix_8ino-example.html) example, and it works fine on my Arduino Leonardo (same chip as the T2.0) using Control Surface v1.2.0.

You can easily verify that it works by connecting a wire between a row pin and a column pin. E.g. if you connect pin 2 to pin 8, the example should send note 3, connect pin 5 to pin 8 and it'll send note 10, and so on.

huggre, 2020-10-12:
Just had to switch the rows and column definition and now the matrix works, thanks

4dvn, 2020-10-12:
people can try similar teensy bootloader on arduino leonardo drom mod bootloader: adafruit/TeeOnArdu