tttapa/Control-Surface #297: Midi Learning Mode

kraftZeug, 2020-10-10:
Is there a midi learning mode or could you add one please? 
(or is there a way i could easily add one?)
I have an accelerometer wich sends too many midi info if the controler is not still. This makes midi learning in some apps difficult.
So if there could be a "midi learning mode" in which inputs need a big change (more than 25% for example) to be sent as midi, that would be awesome..

tttapa, 2020-10-11:
I think such a mode would be outside the scope of this library. Since accelerometers aren't supported out of the box, you're probably reading them and sending MIDI yourself in your main `loop`. In that case, it's probably easiest to implement the "learning mode" logic yourself as well.  
If you want concrete help with that, it's probably best to ask your questions on the Arduino forum (https://forum.arduino.cc). If you open a thread over there, feel free to post the link here.

kraftZeug, 2020-10-11:
In fact, even if i put aside my accelerometer, i have the same issue with some potentiometers.

tttapa, 2020-10-11:
What resistance are you using? Ideally, they should be 10kΩ or less. How are they wired? Are you using the analog ground of the Teensy (if available), are you using long wires? If you can't fix this in hardware, you can try to increase the amount of software filtering. Try increasing the analog shift factor to 3 or 4, and/or change the filter type to `uint32_t` instead of `uint16_t`:
https://github.com/tttapa/Control-Surface/blob/21a09383bb3eaa5f1835184a63fdcdb4b8f5799b/src/AH/Settings/Settings.hpp#L55-L71