tttapa/Control-Surface #298: Midi signal not sent.

kraftZeug, 2020-10-11:
**Describe the bug**
I added a portion of code to introduce input from an accelerometer, but then my controler stops sending midi signals.
If i take off the accelerometer part, everything works nicely.

Here is my code:

`#include <Wire.h>                 // Must include Wire library for I2C
#include "SparkFun_MMA8452Q.h"    // Click here to get the library: http://librarymanager/All#SparkFun_MMA8452Q

MMA8452Q accel;                   // create instance of the MMA8452 class

#include <Control_Surface.h>

USBMIDI_Interface usbmidi;

const int speedMultiplier = 1;
int xAxis = 0;
int xBase = 0;
int constrainedXaxis = 0;

// ******************************************************************
// **** Attribution des Potards, Fader & Jogwheel  ******************
// ******************************************************************

// Create a new instance of the class `CCPotentiometer`, called `Pot1`,
// on pin A0, that sends MIDI messages with controller 7 (channel volume)
// on channel 1.
CCPotentiometer Pot1 = {
  A1, {MIDI_CC::Channel_Volume, CHANNEL_1}
};
CCPotentiometer Pot2 = {
  A2, {MIDI_CC::Channel_Volume, CHANNEL_2}
};
CCPotentiometer Pot3 = {
  A3, {MIDI_CC::Channel_Volume, CHANNEL_3}
};
CCPotentiometer Pot4 = {
  A6, {MIDI_CC::Channel_Volume, CHANNEL_4}
};
CCPotentiometer Pot5 = {
  A7, {MIDI_CC::Channel_Volume, CHANNEL_5}
};
CCPotentiometer Pot6 = {
  A8, {MIDI_CC::Channel_Volume, CHANNEL_6}
};
CCPotentiometer Pot7 = {
  A9, {MIDI_CC::Channel_Volume, CHANNEL_7}
};
CCPotentiometer Fader = {
  A0, {MIDI_CC::Channel_Volume, CHANNEL_8}
};
CCPotentiometer Jogwheel = {
  A14, {MIDI_CC::Channel_Volume, CHANNEL_9}
};
CCPotentiometer Axis = {
  xAxis, {MIDI_CC::Channel_Volume, CHANNEL_10}
};
// ******************************************************************
// ******************************************************************

// ******************************************************************
// **** Attribution des Switchs  ************************************
// ******************************************************************

// Instantiate a CCButton object
CCButton Switch1 = {
  // Push button on pin 5:
  5,
  // General Purpose Controller #1 on MIDI channel 1:
  {MIDI_CC::General_Purpose_Controller_1, CHANNEL_1},
};

CCButton Switch2 = {
  6,
  // General Purpose Controller #1 on MIDI channel 1:
  {MIDI_CC::General_Purpose_Controller_1, CHANNEL_2},
};

CCButton Switch3 = {
  7,
  // General Purpose Controller #1 on MIDI channel 1:
  {MIDI_CC::General_Purpose_Controller_1, CHANNEL_3},
};

CCButton Switch4 = {
  8,
  // General Purpose Controller #1 on MIDI channel 1:
  {MIDI_CC::General_Purpose_Controller_1, CHANNEL_4},
};

CCButtonLatching WristSwitch = {
  9,
  // General Purpose Controller #1 on MIDI channel 1:
  {MIDI_CC::General_Purpose_Controller_1, CHANNEL_5},
};

// The filtered value read when potentiometer is at the 0% position
constexpr analog_t minimumValue = 255;
// The filtered value read when potentiometer is at the 100% position
constexpr analog_t maximumValue = 16383 - 255;

// ******************************************************************
// **** Mapping des Potards, Fader & Jogwheel  **********************
// ******************************************************************

// A mapping function to eliminate the dead zones of the potentiometer:
// Some potentiometers don't output a perfect zero signal when you move them to
// the zero position, they will still output a value of 1 or 2, and the same
// goes for the maximum position.

analog_t mapPot1(analog_t raw) {
    // make sure that the analog value is between the minimum and maximum
    raw = constrain(raw, minimumValue, maximumValue);
    // map the value from [minimumValue, maximumValue] to [0, 16383]
    return map(raw, minimumValue, maximumValue, 0, 16383);
    // Note: 16383 = 2¹⁴ - 1 (the maximum value that can be represented by
    // a 14-bit unsigned number
}

analog_t mapPot2(analog_t raw) {
    // make sure that the analog value is between the minimum and maximum
    raw = constrain(raw, minimumValue, maximumValue);
    // map the value from [minimumValue, maximumValue] to [0, 16383]
    return map(raw, minimumValue, maximumValue, 0, 16383);
    // Note: 16383 = 2¹⁴ - 1 (the maximum value that can be represented by
    // a 14-bit unsigned number
}

analog_t mapPot3(analog_t raw) {
    // make sure that the analog value is between the minimum and maximum
    raw = constrain(raw, minimumValue, maximumValue);
    // map the value from [minimumValue, maximumValue] to [0, 16383]
    return map(raw, minimumValue, maximumValue, 0, 16383);
    // Note: 16383 = 2¹⁴ - 1 (the maximum value that can be represented by
    // a 14-bit unsigned number
}

analog_t mapPot4(analog_t raw) {
    // make sure that the analog value is between the minimum and maximum
    raw = constrain(raw, minimumValue, maximumValue);
    // map the value from [minimumValue, maximumValue] to [0, 16383]
    return map(raw, minimumValue, maximumValue, 0, 16383);
    // Note: 16383 = 2¹⁴ - 1 (the maximum value that can be represented by
    // a 14-bit unsigned number
}

analog_t mapPot5(analog_t raw) {
    // make sure that the analog value is between the minimum and maximum
    raw = constrain(raw, minimumValue, maximumValue);
    // map the value from [minimumValue, maximumValue] to [0, 16383]
    return map(raw, minimumValue, maximumValue, 0, 16383);
    // Note: 16383 = 2¹⁴ - 1 (the maximum value that can be represented by
    // a 14-bit unsigned number
}

analog_t mapPot6(analog_t raw) {
    // make sure that the analog value is between the minimum and maximum
    raw = constrain(raw, minimumValue, maximumValue);
    // map the value from [minimumValue, maximumValue] to [0, 16383]
    return map(raw, minimumValue, maximumValue, 0, 16383);
    // Note: 16383 = 2¹⁴ - 1 (the maximum value that can be represented by
    // a 14-bit unsigned number
}

analog_t mapPot7(analog_t raw) {
    // make sure that the analog value is between the minimum and maximum
    raw = constrain(raw, minimumValue, maximumValue);
    // map the value from [minimumValue, maximumValue] to [0, 16383]
    return map(raw, minimumValue, maximumValue, 0, 16383);
    // Note: 16383 = 2¹⁴ - 1 (the maximum value that can be represented by
    // a 14-bit unsigned number
}

analog_t mapFader(analog_t raw) {
    // make sure that the analog value is between the minimum and maximum
    raw = constrain(raw, minimumValue, maximumValue);
    // map the value from [minimumValue, maximumValue] to [0, 16383]
    return map(raw, minimumValue, maximumValue, 0, 16383);
    // Note: 16383 = 2¹⁴ - 1 (the maximum value that can be represented by
    // a 14-bit unsigned number
}

analog_t mapJogwheel(analog_t raw) {
    // make sure that the analog value is between the minimum and maximum
    raw = constrain(raw, minimumValue, maximumValue);
    // map the value from [minimumValue, maximumValue] to [0, 16383]
    return map(raw, minimumValue, maximumValue, 0, 16383);
    // Note: 16383 = 2¹⁴ - 1 (the maximum value that can be represented by
    // a 14-bit unsigned number
}

// ******************************************************************

// ******************************************************************
// ******  SETUP  ***************************************************
// ******************************************************************

void setup() {
    // Add the mapping function to the potentiometer
    Pot1.map(mapPot1);
    Pot2.map(mapPot2);
    Pot3.map(mapPot3);
    Pot4.map(mapPot4);
    Pot5.map(mapPot5);
    Pot6.map(mapPot6);
    Pot7.map(mapPot7);
    Fader.map(mapFader);
    Jogwheel.map(mapJogwheel);

    // Initialize everything
    Control_Surface.begin();
    Serial.begin(115200);

    Serial.begin(9600);
    Wire.begin();

    // Accelerometer 
    Serial.begin(9600);
    while (!Serial) ;

    Serial.println("MMA8452Q Basic Reading Code!");
    Wire.begin();
    if (accel.begin() == false) {
    Serial.println("Not Connected. Please check connections and read the hookup guide.");
    while (1);   
    }

}

void loop() {

  xBase = ((accel.getCalculatedX()+0.75)*127);
  int constrainedXaxis = constrain(xBase, 0, 127);
  xAxis = map(constrainedXaxis, 0, 127, 127, 0);
  xAxis = constrainedXaxis;

  
  Serial.println(xAxis);
  usbMIDI.sendControlChange(7, (xAxis), 11);

  while (usbMIDI.read()) {

    // Update the Control Surface (check whether the potentiometer's
    // input has changed since last time, if so, send the new value over MIDI).
    Control_Surface.loop();
    // Use this to find minimumValue and maximumValue: it prints the raw value
    // of the potentiometer, without the mapping function
    //Serial.println((analogRead(A14))/8);

  }

}`

kraftZeug, 2020-10-11:
To simplify debuging, this is the parts i added to my code:

void setup() {
    Wire.begin();
    Serial.begin(9600);
    while (!Serial) ;
    Serial.println("MMA8452Q Basic Reading Code!");
    Wire.begin();
    if (accel.begin() == false) {
    Serial.println("Not Connected. Please check connections and read the hookup guide.");
    while (1);   
    }
}

void loop() {
  xBase = ((accel.getCalculatedX()+0.75)*127);
  int constrainedXaxis = constrain(xBase, 0, 127);
  xAxis = map(constrainedXaxis, 0, 127, 127, 0);
  xAxis = constrainedXaxis;
  Serial.println(xAxis);
  usbMIDI.sendControlChange(7, (xAxis), 11);
}

tttapa, 2020-10-11:
```cpp
Serial.begin(115200);

    Serial.begin(9600);
    Wire.begin();

    // Accelerometer 
    Serial.begin(9600);
    while (!Serial) ;
```
Why are you initializing the Serial port three times? 9600 is slow, I would recommend at least 115200. The while loop will block your code if you don't open the serial monitor. If you want to be able to send MIDI without having to open the serial monitor open, you have to delete that line.

```cpp
usbMIDI.sendControlChange(7, (xAxis), 11);
```
Don't mix Control Surface and  `usbMIDI`. If you want to send MIDI over USB, use the `USBMIDI_Interface` that you created at the beginning of your sketch:
```cpp
MIDIAddress addr = {7, CHANNEL_11};
usbmidi.sendCC(addr, xAxis);
```
Your code will flood the receiver with MIDI events, this will probably cause issues, it's best to only send a new message if the value has changed. With a noisy accelerometer, you probably need some filtering and hysteresis (if you have a gyroscope, use a complementary filter).

```cpp
while (usbMIDI.read()) {
```
As before, don't use `usbMIDI` when using Control Surface.
```cpp
  while (usbMIDI.read()) {

    // Update the Control Surface (check whether the potentiometer's
    // input has changed since last time, if so, send the new value over MIDI).
    Control_Surface.loop();
    // Use this to find minimumValue and maximumValue: it prints the raw value
    // of the potentiometer, without the mapping function
    //Serial.println((analogRead(A14))/8);

  }
```
This is your main problem: you only update the Control Surface when there is incoming MIDI USB data. If there's no incoming data, your potentiometers are never updated, and Control Surface can never send their MIDI output.

kraftZeug, 2020-10-11:
Woah! Many thanks man!
You did a quick response and found my problem.
I'll check that later but i'm pretty sure this will solve it.
Thanks again!

kraftZeug, 2020-10-11:
It worked like a charm.
Yeah! Long live Tttapa & Control-Surface!

closed 2020-10-11 by kraftZeug

kraftZeug, 2020-10-12:
Hey!
In fact, i encountered another issue on this: it seems i can't add some code in the loop without preventing Control_Surface.loop() to run.
So this does not work:
`void loop() {

    xBase = ((accel.getCalculatedX()+0.75)*127);
    int constrainedXaxis = constrain(xBase, 0, 127);
    xAxis = map(constrainedXaxis, 0, 127, 127, 0);
    xAxis = constrainedXaxis;
    usbMIDI.sendControlChange(7, (xAxis), 11);

    Control_Surface.loop();

}`

Is there a way i can add something in your loop; or can i add another loop or function in a way that doesn't prevent your code from running?

tttapa, 2020-10-12:
As long as the code you add to the `loop` doesn't block (i.e. `delay` or while loops that take longer than a couple of milliseconds), it should be fine. What doesn't work exactly?

Like I mentioned before, using both Control Surface and `usbMIDI` is not recommended. Try using the `USBMIDI_Interface` instance `usbmidi` that you declared at the top of your sketch (see the third snippet in my first reply).

kraftZeug, 2020-10-12:
Ok, i forgot to change:
usbmidi.sendCC(addr, xAxis);
but that is not the issue.

I tried and found that this line is the one stopping your code:
xBase = ((accel.getCalculatedX()+0.75)*127);
I think it's calling a function of the Wire.h library which delays or loops too much.

kraftZeug, 2020-10-13:
I dug a little bit. The conflict comes not from the Wire library but from SparkFun_MMA8452Q library.
Specificaly when i call for this function:
`float MMA8452Q::getCalculatedX()
{
	x = getX();
	return (float)x / (float)(1 << 11) * (float)(scale);
}`
But i did try this with the usbMIDI and it conflicts with it too, so i guess i'll have to use another accelerometer or try to modify sparkfun's library.
Thanks for everything!

tttapa, 2020-10-13:
Does the code work without using MIDI over USB? AFAIK, USB shouldn't interfere with I2C.

Are you sure that the hardware is fine? The `getX()` function just reads some registers over I2C:
https://github.com/sparkfun/SparkFun_MMA8452Q_Arduino_Library/blob/4018c02416b84250a9f68585ff3756b16c80ef0a/src/SparkFun_MMA8452Q.cpp#L431-L449
The Wire library might hang if the device is disconnected or misbehaving. It could also block if you don't call `Wire.begin()` in before using the bus.  
Make sure that you have I2C pull-ups (the ones built-in to the Teensy usually won't do the job) and that the address is correct.

kraftZeug, 2020-10-14:
The code for the accelerometer worked last time i checked.
I'll have to try again to make sure.
Wire.begin() is called at the beginning.
I didn't used pull-ups, as they wasn't required in the tutorial i followed.
Anyway, this is not an issue with your code, so i don't want to bother you.
I ordered a different accelerometer to give it another try.
Just for you to know, the midi controler is inside an electric guitar and i'm using it to control effects on ableton. The gyroscope part of the accelerometer is intented to be used as a potentiometer when you tilt the guitar towards the roof.
I'll let you know how it goes.

kraftZeug, 2020-10-28:
I tried with and MPU-6050 and it worked!
Thanks again! :-)

kraftZeug, 2020-10-30:
Soo.
My accelerometer works (i can read its value on serial monitor) and so does midi.
But i stil have a question:
What would be the correct way to send the Z axis info as a potentiometer?
I know I'll have to constrain the data to 0-127 and to apply some hysteresis, but how do I send this as a midi CC?

tttapa, 2020-10-30:
Once you have the value between 0 and 127, you can just use 

```cpp
MIDIAddress zAddr = {7, CHANNEL_11};
usbmidi.sendCC(zAddr , zAxisValue);
```