tttapa/Control-Surface #299: Is there a maximum of mux that the library can use?

benwadub, 2020-10-13:
Hi Pieter, I try to double my controller and when I put my code for 16 channels nothing work, do you think 10 4067 mux is too mutch ?
[image]

tttapa, 2020-10-13:
Hard to say without seeing your code. Try adding some print statements to figure out where the code hangs. If you have enough memory, try enabling the debug mode:

https://github.com/tttapa/Control-Surface/blob/21a09383bb3eaa5f1835184a63fdcdb4b8f5799b/src/AH/Settings/Settings.hpp#L19-L27

benwadub, 2020-10-13:
my code is this one, it works fine with only 5 mix but as soon as I try using 10 nothing works except few pots on max 2 that seems to send on 4 different channels
```cpp
#include <Control_Surface.h>

//#include <name.c>
// celà doit etre écrit avant la library control surface
 USBDebugMIDI_Interface midi = 115200; // enlever les // en début de ligne pour entrer en mode debug usb et voir dans le panneau de control si vos controler envoient bien les infos

//auto &serial = Serial1;// Selectionne le port série à utiliser remplacer par serial pour une arduino
//SerialMIDI_Interface<decltype(serial)> midi = {serial, MIDI_BAUD};// démarre une interface midi serial au midi baud rate par defaut
//USBMIDI_Interface usbmidi;// enlever les / en debut de ligne pour activer l'interface usb, penser à désactiver l'interface série(din)
//HardwareSerialMIDI_Interface midiser = Serial1;
 
 
//MIDI_PipeFactory<8> pipes;

CD74HC4067 mux1 = {
  14,       // numéro de broche de l'arduino
  {9, 3, 4, 5} // numéro de pins de l'arduino sur lesquels sont branchés tous les multiplexeurs apellés mux S0, S1, S2
};

CD74HC4067 mux2 = {
  15,              
  {9, 3, 4, 5}, 
};

CD74HC4067 mux3 = {
  16,       
  {9, 3, 4, 5} 
};
CD74HC4067 mux4 = {
  17,       
  {9, 3, 4, 5} 
};
CD74HC4067 mux5 = {
  18,      
  {9, 3, 4, 5} 
};
CD74HC4067 mux6 = {
  6,      
  {9, 3, 4, 5} 
};
CD74HC4067 mux7 = {
  7,      
  {9, 3, 4, 5} 
};
CD74HC4067 mux8 = {
  8,      
  {9, 3, 4, 5} 
};
CD74HC4067 mux9 = {
  11,      
  {9, 3, 4, 5} 
};
CD74HC4067 mux10 = {
  10,      
  {9, 3, 4, 5} 
};
//Bank<2> bank = {8}; // active 2 bank avec 8 adresses par bank
/*IncrementSelector<2> selector = {
    bank,       // Bank to manage
    20,     // push button pins (increment, decrement)
    Wrap::Wrap, // Wrap around
};
*/
CCPotentiometer fxparameter [] = {
  {mux1.pin(0), {1, CHANNEL_16}},//delay time
  {mux1.pin(1), {2, CHANNEL_16}},//pingpong
  {mux1.pin(2), {3, CHANNEL_16}},//stereo width
  {mux1.pin(3), {4, CHANNEL_16}},//feedback
[... 189 lines cut ...]
  // General Purpose Controller #1 on MIDI channel 1:
  {121, CHANNEL_16},
};
  
void setup() {
  Control_Surface.begin(); // initialise la library surface de control
  /*usbmidi >> pipes >> midiser; // all incoming midi from USB is sent to serial
  usbmidi << pipes << midiser; // all incoming midi from Serial is sent to USB
  usbmidi >> pipes >> usbmidi; // all incoming midi from USB is looped back
  midiser << pipes << midiser;
  Control_Surface >> pipes >> usbmidi;
  Control_Surface << pipes << usbmidi;
  Control_Surface >> pipes >> midiser;
  Control_Surface << pipes << midiser;
  usbmidi.begin();
  midiser.begin();
 */
}                           
void loop() {
  

  Control_Surface.loop(); // Update the Control Surface

  //midiser.update();
}```

benwadub, 2020-10-13:
i m on a teensy 4.0 I tried the debug to see on the monitor what kind of messages are sent. it s crazy my 2 controllers works fine individually but when I try them together nothings works at all even the one where the teensy is plugged
[image]

tttapa, 2020-10-13:
Have you tried turning on the debug mode as explained in my first reply?

You could try testing the AnalogReadSerial example with all your muxes: https://tttapa.github.io/Control-Surface-doc/Doxygen/dc/d1b/1_8AnalogReadSerial_8ino-example.html

tttapa, 2020-10-13:
Also check if disconnecting the extra muxes changes anything. Maybe the address lines are shorted by one of the muxes, or maybe the capacitance is too high when you have a fan-out of 10 on these lines.

benwadub, 2020-10-13:
with analog read test with the 10 mux I get this on monitor but really don t understand what that mean!
[image]

tttapa, 2020-10-13:
It just prints all values of the analog inputs of the mux: the first 16 columns are the inputs of the first mux, the 16 columns are the inputs of the second mux, and so on.

Try moving some potentiometers to see if the corresponding value changes as expected. You might want to print some separator values between the columns of each mux.

benwadub, 2020-10-13:
all the values are displaying too fast for I can see anychange, I stopped the auto scrolling to check the values each time I turn a knob but on the second controller nothing seems to happend I I didn't t understood how to enter in debug mode
with the debug midi interface I see that controller number one works fine but the second not at all

benwadub, 2020-10-13:
I tried to swap some pins used by mux 6 to 10 and that seems to work! apparently some pins on my teensy don t want to be mux dress pins! don t understand why!!

tttapa, 2020-10-13:
You might have shorted them out previously, and the output drivers could be destroyed. There aren't any limitations on which pins you can for the mux address lines.

If the AnalogReadSerial example doesn't work, the MIDI controller examples won't work either, so it's important to fix that first.

To enter debug mode, you have to edit the file `Control-Surface/src/AH/Settings/Settings.hpp`, comment out the line `#define DEBUG_OUT` and uncomment the line `#define DEBUG_OUT Serial `.